espressif/arduino-esp32 #8029: GDB executable  "arm-none-eabi-gdb.exe" not being found

hezilidemao23, 2023-04-05:
### Board

esp32

### Device Description

euno esp32
https://item.jd.com/10036937330081.html
[image]

### Hardware Configuration

[EUNO 原理图 2021-7-26.pdf](https://github.com/espressif/arduino-esp32/files/11154907/EUNO.2021-7-26.pdf)

### Version

v2.0.7

### IDE Name

Arduino IDE 2.0.4

### Operating System

windows 10

### Flash frequency

240MHZ

### PSRAM enabled

yes

### Upload speed

921600

### Description

when debugging, it warn that :
GDB executable  "arm-none-eabi-gdb.exe" not being found
is taht default configuration is wrong?

### Sketch

```cpp
int LED_BUILTIN = 13;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(10000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(10000);                       // wait for a second
}
```

### Debug Message

```plain
GDB executable "C:\Users\username\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\bin\arm-none-eabi-gdb.exe" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-04-05:
have you selected the debug adapter? The board as-is required an external JTAG adapter and then that to be selected in the board menu. Else necessary files will not be copied and debug will not work.

hezilidemao23, 2023-04-05:
> have you selected the debug adapter? The board as-is required an external JTAG adapter and then that to be selected in the board menu. Else necessary files will not be copied and debug will not work.

Yes, I have tried two existing methods on the board (FTDI adapter and esp usb bridge),but it warn the same.

me-no-dev, 2023-04-06:
If you have selected FTDI, did you recompile and re-upload the project? You need to do so in order for the proper files to get copied to the sketch folder.

hezilidemao23, 2023-04-06:
> If you have selected FTDI, did you recompile and re-upload the project? You need to do so in order for the proper files to get copied to the sketch folder.

thank you so much，i understand it

closed 2023-04-06 by hezilidemao23

reddai75, 2023-04-22:
same issue, no solution

ricard2k, 2023-05-27:
I had this problem: Select board, select debugger... -> gdb not found, Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly

The solution was as stupid as recompile after change the configuration.

frankcohen, 2023-05-28:
I'm having the same problem. On MacOS 13.3.1, Arduino IDE 2.1, board is Adafruit Feather ESP32-S3 No PSRAM. I click Compile, compiles successfully, click Upload, uploads successfully, click Start Debugging, I see the debug window and get the error:

GDB executable "/Users/frankcohen/Library/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823/bin/arm-none-eabi-gdb" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.

I check /Users/frankcohen/Library/Arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/11.2_20220823/bin and see:

frankcohen@FCohenMac2020 bin % ls -l
total 287048
-rwxr-xr-x@ 1 frankcohen  staff  24369216 May 21 20:04 xtensa-esp-elf-gdb-3.10
-rwxr-xr-x@ 1 frankcohen  staff  24361008 May 21 20:04 xtensa-esp-elf-gdb-3.6
-rwxr-xr-x@ 1 frankcohen  staff  24356576 May 21 20:04 xtensa-esp-elf-gdb-3.7
-rwxr-xr-x@ 1 frankcohen  staff  24364992 May 21 20:04 xtensa-esp-elf-gdb-3.8
-rwxr-xr-x@ 1 frankcohen  staff  24369200 May 21 20:04 xtensa-esp-elf-gdb-3.9
-rwxr-xr-x@ 1 frankcohen  staff      4587 May 21 20:04 xtensa-esp-elf-gdb-add-index
-rwxr-xr-x@ 1 frankcohen  staff  23588192 May 21 20:04 xtensa-esp-elf-gdb-no-python
-rwxr-xr-x@ 1 frankcohen  staff   1524400 May 21 20:04 xtensa-esp-elf-gprof
-rwxr-xr-x@ 1 frankcohen  staff      2599 May 21 20:04 xtensa-esp32-elf-gdb
-rwxr-xr-x@ 1 frankcohen  staff      2605 May 21 20:04 xtensa-esp32s2-elf-gdb
-rwxr-xr-x@ 1 frankcohen  staff      2605 May 21 20:04 xtensa-esp32s3-elf-gdb

USB-CDC on boot enabled
USB DFU on boot enabled
Upload UART0/Hardware CDC
USB Mode Hardware CDC and JTAG

I have D+ to GPIO20
D- to GPIO19
+5 and GND

I don't see how to configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath"?

-Frank

me-no-dev, 2023-05-29:
the solution is very basic: select debug adapter in the menu, compile, upload, debug from the green button next to upload.

frankcohen, 2023-05-29:
I got it working. The issue was using the Adafruit Feather ESP32-S3 No PSRAM board. I have an Espressif ESP32-S3 DevKitM board. Using ESP32S3 Dev Module board, gives access to Tools -> JTAG Adaptor. Once I made this change I uploaded the sketch and clicked Start Debugging. Thanks for your help.

frankcohen, 2023-05-29:
Also, Andreas Speiss dropped a new video explaining the best ways to use USB with ESp32. https://www.youtube.com/watch?v=hJSBTFsOnoA