espressif/arduino-esp32 #8033: Memory leakage with WiFi.mode()

muhdRiz07, 2023-04-05:
### Board

ESP-WROOM-32 module

### Device Description

[image]

### Hardware Configuration

No, Nothing else attached to this development board.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Linux

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi, here I am trying WiFi scanning but facing reduction in memory after scanning. Before WiFi.mode(WIFI_STA) the heap size is 311588 but after WiFi.mode(WIFI_STA) the free heap size reduces to 265932, consumes 45656 bytes of memory. May I know why it is consumes this much memory and how to solve it??

Thanks in advance.

### Sketch

```cpp
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    Serial.println("\nFree heap : " + String(ESP.getFreeHeap()));
    Serial.println("Free min heap : " + String(ESP.getMinFreeHeap()));
    WiFi.mode(WIFI_STA);
    Serial.println("\nFree heap : " + String(ESP.getFreeHeap()));
    Serial.println("Free min heap : " + String(ESP.getMinFreeHeap()));
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    Serial.println("\nFree heap : " + String(ESP.getFreeHeap()));
    Serial.println("Free min heap : " + String(ESP.getMinFreeHeap()));
    int n = WiFi.scanNetworks();
    Serial.println("\nFree heap : " + String(ESP.getFreeHeap()));
    Serial.println("Free min heap : " + String(ESP.getMinFreeHeap()));
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");
    Serial.println("\nFree heap : " + String(ESP.getFreeHeap()));
    Serial.println("Free min heap : " + String(ESP.getMinFreeHeap()));
    // Wait a bit before scanning again
    delay(5000);
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4

Free heap : 311588
Free min heap : 305952

Free heap : 265932
Free min heap : 264404

Setup done
scan start
 
Free heap : 265432
Free min heap : 264332

Free heap : 264968
Free min heap : 262392
scan done
2 networks found
1: 1plus (-61)*
2: YOGI (-92)*
 
Free heap : 264968
Free min heap : 262392
 
scan start

Free heap : 264968
Free min heap : 262392
 
Free heap : 264968
Free min heap : 262376
scan done
5 networks found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-05:
Welcome to development on the ESP32.  Most libraries in this repository allocate and hold some global memory as they run.  This memory is necessary for them to operate.  The only way to release the memory would be to completely unload the classes.  You have plenty of heap memory left.  I wouldn't worry about it.

muhdRiz07, 2023-04-06:
@mrengineer7777 thanks for your response. Can you explain how to completely unload the classes?? Also in our Actual implementation we are using more memory. so this memory loss is considerably more.

mrengineer7777, 2023-04-06:
The WiFiClass and many others in this repository are globally declared objects. It's not possible to completely unload them.  The strategy that I use when switching from ETH->WiFi is to save the mode to NVS (using Preferences.h), then reboot and load the new mode.  If you need better control of memory then consider using IDF.