espressif/arduino-esp32 #8036: I2C Wire.beginTrasmission(address) locks up when pull-up resistors are tied to GND

garageeks, 2023-04-05:
### Board

ESP32 Dev Module

### Device Description

ESP32-WROOM-32E module on custom PCB

### Hardware Configuration

GPIO25 = SDA, GPIO26 = SCL

### Version

v2.0.7

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When I flash the board while being unpowered, the I2C pull-up resistors are without 3.3V because it is sourced from a separated step down from the one used to power ESP32. Once the code is loaded, a simple I2C scanner is launched and get immediately stuck at Wire.beginTransmission(x). Nothing happens till watchdog resets the ESP.
Same code but with the board powered externally, pull-up resistors are pulling high and works without issues.
While an I2C bus with pull-up resistors floating is an anomaly, and it shouldn't happen, in practice a minor hardware issue would block the code execution.
It should timeout and let the code do something else.
I tested this on 2.0.4 and on 2.0.7

### Sketch

```cpp
setup() {
      Serial.begin(115200);
      out.println(F("External I2C bus init..."));
      Wire1.begin(25, 26, 100000U);	//External I2C bus
      extI2Cscanner();
}

void extI2Cscanner() {
	byte error, address;
  	int nDevices;
 
  	out.println(F("Ext I2C Scanning..."));
 
  	nDevices = 0;
  	for(address = 1; address < 127; address++ ) {

    	// The i2c_scanner uses the return value of
    	// the Write.endTransmisstion to see if
    	// a device did acknowledge to the address.
    	Wire1.beginTransmission(address);
    	error = Wire1.endTransmission();
 
    	if (error == 0) {
			out.print(F("I2C device found at address 0x"));
			if (address<16) out.print("0");
			out.println(address,HEX);

			nDevices++;

    	} else if (error == 4) {
      		out.print(F("Unknown error at address 0x"));
      		if (address<16) out.print("0");
      		out.println(address,HEX);
    	}    
  	}
  	if (nDevices == 0) {
    	out.println(F("No I2C devices found"));
	} else {
    	out.println(F("done"));
	}
}
```

### Debug Message

```plain
External I2C bus init...
[   117][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=25 scl=26 freq=100000
Ext I2C Scanning...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-05:
I2C slaves can hold the clock line low for "clock stretching".  If a master is clock stretching aware (most are), then it may wait a long time (indefinitely?) for the slave to release the line.  It depends on how Espressif implemented their hardware and the I2C driver at the IDF level.  It's unlikely we can do anything about it in this repository.

closed 2023-04-26 by mrengineer7777

garageeks, 2023-09-14:
@mrengineer7777 Thank you for your reply. Today while testing some boards I had some faulty I2C sensors that locked up code execution again. In my humble opinion this is a critical issue.
Are you sure the ESP-IDF library does not expose some timeout interval or flag to disable clock streching which are not used by the arduino-esp32 wrapper?
In any case I opened up a issue in the ESP-IDF repo and let's see.