espressif/arduino-esp32 #8039: unable to disable wake up source after enable via esp_deep_sleep_enable_gpio_wakeup

our30K, 2023-04-06:
### Board

ESP32C3 devkit

### Device Description

ESP32-C3 devkitM-1

### Hardware Configuration

just standard USB connection to computer via the onboard micro USB port. 

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Mac OS 13.2

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I encounter a rather strange problem, not sure how do I resolve this

I have 2 GPIO pins used to wake up my ESP32C3 in different deep sleep case:
GPIO 2: wake up device when active HIGH. 
GPIO 3: wake up device when active LOW. 

case 1: only wake up with GPIO 3 LOW, 
case 2: wake up with either GPIO 3 LOW or GPIO 2 HIGH

In my code, after ESP boot up. 
device was then asked to sleep in case 1, everything is fine, 
then wake up the device, and ask the device to sleep in case 2, everything is fine again

then wake up the device, and ask to sleep in case 1 again. 
but in this case, GPIO 2 high still wake up the device, even though, it should not. 

to replicate, simply run the sketch code below, 
type espsleep1 to sleep in case 1, and test manually pull high on GPIO 2, and manually pull low on GPIO 3, see device wake up only on GPIO 3 LOW

then type espsleep2 to sleep in case 2, and test manually pull high on GPIO 2 , see device wake up on GPIO 2 HIGH

then type espsleep1 again, and test manually pull high on GPIO 2, and manually pull low on GPIO 3, see device wake up on either GPIO 3 LOW or GPIO 2 HIGH, 

I need to ensure GPIO 2 HIGH wake up is disabled in case 1.  but the only way to do so is to reboot the ESP if I ever run case2 once.  I can't disable GPIO 2 HIGH wakeup without reset the ESP. 

### Sketch

```cpp
const uint64_t WAKEUP_HIGH_PIN_BITMASK = 0b0100;     //GPIO0-GPIO3
const uint64_t WAKEUP_LOW_PIN_BITMASK = 0b1000;     //GPIO0-GPIO3
RTC_DATA_ATTR int bootCount = 0;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("hello");
  bootCount++;
  Serial.println("Boot number: " + String(bootCount));
  print_wakeup_reason();
}

void loop() {

  String output ="";
  while (Serial.available()){
    char c = char(Serial.read());
    if (c != '\n'){
      output = output + c;
    }else{
      Serial.println("<<"+output);
      input_process(output);
    }
  }
  delay(1);
}

void input_process(String output){
  Serial.println(">>"+output);
  if (output == "espsleep1"){
    esp_deep_sleep1();
  }else if  (output == "espsleep2"){
    esp_deep_sleep2();
  }
}

void esp_deep_sleep1(){
  esp_err_t errPD = esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  Serial.print("esp_sleep_pd_config: ");
  switch(errPD) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  

  esp_err_t errGPIO2 = esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
  Serial.print("esp_deep_sleep_disable_gpio_wakeup: all pins from high ");
  switch(errGPIO2) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }

  esp_err_t errGPIO = esp_deep_sleep_enable_gpio_wakeup(WAKEUP_LOW_PIN_BITMASK, ESP_GPIO_WAKEUP_GPIO_LOW);
  Serial.print("esp_deep_sleep_enable_gpio_wakeup: LOW on IO3 ");
  switch(errGPIO) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  Serial.println();
  Serial.println();
  delay(500);
  Serial.println("Sleep Start");
  esp_deep_sleep_start();
}

void esp_deep_sleep2(){
  esp_err_t errPD = esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  Serial.print("esp_sleep_pd_config: ");
  switch(errPD) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  esp_err_t errGPIO = esp_deep_sleep_enable_gpio_wakeup(WAKEUP_LOW_PIN_BITMASK, ESP_GPIO_WAKEUP_GPIO_LOW);
  Serial.print("esp_deep_sleep_enable_gpio_wakeup: LOW on IO3 ");
  switch(errGPIO) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  esp_err_t errGPIO2 = esp_deep_sleep_enable_gpio_wakeup(WAKEUP_HIGH_PIN_BITMASK, ESP_GPIO_WAKEUP_GPIO_HIGH);
  Serial.print("esp_deep_sleep_enable_gpio_wakeup: HIGH on IO2");
  switch(errGPIO2) {
    case ESP_OK: Serial.println("ESP_OK"); break;
    case ESP_ERR_INVALID_ARG: Serial.println("ESP_ERR_INVALID_ARG"); break;
    default: Serial.println("None"); break;
  }
  Serial.println();
  Serial.println();
  delay(500);
  Serial.println("Sleep Start");
  esp_deep_sleep_start();
}

void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();
  switch(wakeup_reason) {
    case ESP_SLEEP_WAKEUP_GPIO : Serial.println("Wakeup caused by external signal using GPIO"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}
```

### Debug Message

```plain
no error message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2023-04-06 by our30K

spilz87, 2023-07-23:
hello @our30K 
did you find your issue ?
I have same issue to wake up from external HIGH level on pin 0,1 and 2
thanks in advance

SuGlider, 2023-08-03:
The issue seems to be related to wrog BIT_MASK when using `esp_deep_sleep_enable_gpio_wakeup()`

our30K, 2023-08-04:
@spilz87 
I am not sure exactly your issue, but after you enable a certain pin for wake up, you will need to use the below function to disable it. 
gpio_wakeup_disable
gpio_deep_sleep_wakeup_disable