espressif/arduino-esp32 #8042: WiFiClient can be disconnected while having unread data, (it shouldn't)

YruamaLairba, 2023-04-06:
### Board

M5StickC-plus

### Device Description

It's a development kit based on ESP32-PICO, for more details see https://shop.m5stack.com/products/m5stickc-plus-esp32-pico-mini-iot-development-kit

### Hardware Configuration

Irrelevant for the issue

### Version

v2.0.6

### IDE Name

Arduino IDE and arduino-cli

### Operating System

Ubuntu 22.04.2 LTS (Jammy Jellyfish)

### Flash frequency

don't know (and probably irrelevant)

### PSRAM enabled

yes

### Upload speed

don't know (and probably irrelevant)

### Description

In a sketch, i get WiFiClient from a WiFiServer and sometimes, I get a disconnected Client that have unread data. According to the Arduino documentation, this can't happen, a client with unread data is considered connected even if the connection is closed. See https://www.arduino.cc/reference/en/libraries/wifi/client.connected/

In my real application, i had data loss because of that. When 2 (or more) clients tried to communicate to the server at same time, i had only data of the first client processed by the server.

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiServer.h>

// Set these to your desired credentials.
const char *ssid = "M5Stack_Ap";
const char *password = "66666666";

WiFiServer server(4321, 1);
WiFiClient client;

void setup() {
  Serial.begin(115200);
  Serial.println("Serial started");
  WiFi.softAP(ssid, password);
  Serial.println("Starting serveur");
  server.begin();
}

void loop() {
  // simulate heavy  work load
  delay(1000);

  // WARNING: WiFiClient behaviour doesn't match Arduino documentation, a
  // client can be considered unconnected while having unread data.
  if (!client.connected() && !client.available()) {
    // current client is finished, trying to get a new one
    if (server.hasClient()) {
      Serial.println("server has client");
      client = server.available(); // listen for incoming clients.
      if (!client.connected() && client.available()) {
        Serial.println("Got a disconnected client with unread data, this "
                       "shouldn't happen");
      }
    }

  } else {
    // current client is not finished, process it 
    while (client.available()) {
      char c = client.read();
      Serial.print(c);
    }
  }
}
```

### Debug Message

```plain
???
```

### Other Steps to Reproduce

To reproduce the issue, you need to have the "nc" command on your computer.

Step to reproduce the issue:
- compile and load the sketch into an esp32
- open a serial monitor to your esp32
- with you computer, connect the WiFi network provided by the esp32 (ssid and password are on top of the sketch)
- open a terminal and run `echo "bla"| nc -v -N -w 0 192.168.4.1 4321 && echo "bli" | nc -v -N -w 0 192.168.4.1 4321`

The serial monitor should display this:
```
server has client
bla
server has client
Got a disconnected client with unread data, this shouldn't happen
bli
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-04-06:
It is really strange why Arduino went that way. Maybe it has to do with the WiFi adapters it is using. Connection state and unread bytes do not have much in common. I suggest you actually check if there is data available, rather than us introducing such "feature". After all you can not communicate further with a client if it has been disconnected, so such "feature" will break the meaning of `connected`

YruamaLairba, 2023-04-06:
Does arduino-esp32 designed to mimic the original arduino API ?
Personally, i assumed the answer was "yes". So, in my opinion, "such feature" is actually "how it is supposed to work" and having a different behavior is a bug. Discussing about poor design choice is out of scope.

Adding to this, i saw many example [like this one](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino) with the following pattern to check if a "client" should be processed
```C++
WiFiClient client = server.available();   // listen for incoming clients
if (client) {                             // if you get a client,
  // some processing
}
```
 `if(client)` is used to know if we "have" client, so i expected same behavior than using `if (server.hasClient()) `.For information,  `if(client)` is currently equivalent to `if(client.disconnected()`. Imagine how i was surprised when i got different result from `if(client)` and `if(server.hasClient())`

me-no-dev, 2023-04-07:
While I do agree that compatibility is good, the ESPs have been working like this for a very long time (since 8266 Arduino exists), so changing that behavior now can have unforeseen consequences. This could get looked into, though I am not sure if it's even possible to asses this on a wide enough scale and prevent breaking other libraries.

closed 2023-04-10 by mrengineer7777

YruamaLairba, 2023-04-10:
I think, this couldn't stay as this. This behavior should be at least mentioned at some very visible place, otherwise many people with spend a lot of time to guess why they silently lose some data because they assume something that is actually not true. Personally, i spent 5 days in total on a issue just because of that.

me-no-dev, 2023-04-10:
@YruamaLairba we are slowly but surely improving our documentation. You are free to add the note, if not, @pedrominatel would you put this as to-do?