espressif/arduino-esp32 #8045: static ip cannot work

davidbcf, 2023-04-07:
### Board

ESP32-C3-S2

### Device Description

ESP32-C3-S2

### Hardware Configuration

no

### Version

other

### IDE Name

Arduino IDE

### Operating System

win 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

In Korea and America, when use Wifi with static ip ,devices cant connet to tcp server.but dynamic ip works.
In China both static ip works.

### Sketch

```cpp
err=read_info(SYS_STATE,M_STATIC_IP,&m_static_ip);
      if(m_static_ip!=""){
        err=read_info(SYS_STATE,M_GATEWAY,&m_gateway);
        err=read_info(SYS_STATE,M_SUBNET,&m_subnet);
        err=read_info(SYS_STATE,M_PRIMARY_DNS,&m_primary_dns);
        err=read_info(SYS_STATE,M_SECOND_DNS,&m_second_dns);
        //IPAddress local_IP=convert_ip(m_static_ip.c_str());
        IPAddress local_IP,gateway,subnet,primaryDNS,secondaryDNS;
        local_IP.fromString(m_static_ip.c_str());
        Serial.println(local_IP);
        gateway.fromString(m_gateway.c_str());
        subnet.fromString(m_subnet.c_str());
        primaryDNS.fromString(m_primary_dns.c_str());
        secondaryDNS.fromString(m_second_dns.c_str());
        if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
          Serial.println("STA Failed to configure");
        }
      }
      WiFi.begin(m_ssid.c_str(), m_pass.c_str());
```

### Debug Message

```plain
when we use wireshark to capture packages,it seems like  device ip was error.for excample,we set static ip :192.168.3.12;wireshark show ip is 0.168.3.12. also we failed to ping address 192.168.3.12 .
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-10:
@davidbcf - 

There is missing information in this issue:
What is the type of the variables `m_static_ip`, `m_gateway`, `m_subnet`, `m_primary_dns` and `m_second_dns`?
What is the code for the function `read_info()`?

>we set static ip :192.168.3.12;wireshark show ip is 0.168.3.12.

Please run the sketch using ArduinoIDE->menu->Tools->`Core Debug Level: "Debug"` and post here the output.

SuGlider, 2023-04-10:
@davidbcf - I have tested this sketch that sets static IP address for WiFi, with success, using an ESP32-S2:

``` cpp
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "SSID";
const char* password = "PWD";

//Your IP address or domain name with URL path
const char* serverURL = "http://www.google.com";

void setup() {
  Serial.begin(115200);

  // testing Static IP Addresses
  String m_static_ip, m_gateway, m_subnet, m_primary_dns, m_second_dns;
  m_static_ip = "192.168.3.12";
  m_gateway = "192.168.3.1";
  m_subnet = "255.255.255.0";
  m_primary_dns = m_gateway;
  m_second_dns = "";
  IPAddress local_IP, gateway, subnet, primaryDNS, secondaryDNS;
  local_IP.fromString(m_static_ip.c_str());
  Serial.printf("==> Setting this static IP Address: %s\n\n", local_IP.toString());
  gateway.fromString(m_gateway.c_str());
  subnet.fromString(m_subnet.c_str());
  primaryDNS.fromString(m_primary_dns.c_str());
  secondaryDNS.fromString(m_second_dns.c_str());

  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
  }

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());
  Serial.print("WiFi network Gateway: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("WiFi network Netmask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("WiFi network DNS: ");
  Serial.print(WiFi.dnsIP(0));
  Serial.print(" -- ");
  Serial.println(WiFi.dnsIP(1));
  Serial.println();
}

void loop() {
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  const long interval = 60 * 1000;  // every minute

  if (currentMillis - previousMillis >= interval) {
    // Check WiFi connection status
    if (WiFi.status() == WL_CONNECTED ) {
        String getResp = httpGETRequest(serverURL);
        Serial.println("HTTP GET RESP:\n" + getResp);
      
      // save the last HTTP GET Request
      previousMillis = currentMillis;
    }
    else {
      Serial.println("WiFi Disconnected");
[... 46 lines cut ...]
.
Connected to WiFi network with IP Address: 192.168.3.12
WiFi network Gateway: 192.168.3.1
WiFi network Netmask: 255.255.255.0
WiFi network DNS: 192.168.3.1 -- 0.0.0.0

HTTP Response code: 200
HTTP GET RESP:
<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="pt-BR"><head><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><meta content="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop="image"><title>Google</title><script nonce="Zuf3gP1FIqS8At0tBb4_BA">(function(){window.google={kEI:'sAM0ZLXEM8vY5OUPpc-74AI',kEXPI:'0,1359409,6058,207,4804,2316,383,246,5,1129120,1197712,689,380090,16114,28684,22431,1361,12317,17582,4998,13228,3847,38444,2872,2891,4139,7615,606,30668,30022,2614,13142,3,346,230,20583,4,1528,2304,29062,12890,174,11444,2215,4437,16786,5830,2527,4094,7596,1,42154,2,14022,2373,23366,5679,1021,31121,4569,6258,23418,1249,5838,14968,4332,7484,445,2,2,1,24626,2006,8155,6680,701,15970,873,6577,13056,8,1921,9779,21391,2302,12461,6305,3142,17056,5797,11,14329,14,82,20206,1622,1779,4976,18960,281,298,3833,991,3030,426,5684,3226,6480,1291,513,5209,3020,1152,1091,1757,1128,421,28,554,4,1487,5160,696,5742,5233,415,5780,669,1,2378,373,1757,43,2322,270,379,13,1635,146,1461,3882,4,1244,1211,314,173,1272,1657,1051,1442,734,395,4943,962,1438,476,815,662,4,167,884,1359,3,346,2,604,412,804,715,258,1873,1128,346,167,467,129,67,538,1861,416,3,129,1177,1,22,689,267,412,329,463,265,916,124,1063,2,729,574,188,49,114,288,420,840,16,104,5,163,750,231,582,3,371,1341,272,1058,48,210,104,968,380,369,275,19,503,485,304,196,151,376,432,571,221,486,2,1987,5206654,97,206,392,68,19,8798364,3311,141,795,19735,1,1,346,4650,153,10,12,1,6,1,7,7,1,23,48,4,10,23945551,4042143,1964,16672,2894,6250,12560,3180,1566,462,1413028,146987,47333',kBL:'x4dB',kOPI:89978449};google.sn='webhp';google.kHL='pt-BR';})();(function(){
var e=this||self;var g,h=[];function k(a){for(var c;a&&(!a.getAttribute||!(c=a.getAttribute("eid")));)a=a.parentNode;return c||g}function l(a){for(var c=null;a&&(!a.getAttribute||!(c=a.getAttribute("leid")));)a=a.parentNode;return c}function m(a){/^http:/i.test(a)&&"https:"===window.location.protocol&&(google.ml&&google.ml(Error("a"),!1,{src:a,glmm:1}),a="");return a}
function p(a,c,b,f){var d="";-1===c.search("&ei=")&&(d="&ei="+k(b),-1===c.search("&lei=")&&(b=l(b))&&(d+="&lei="+b));b="";e._cshid&&-1===c.search("&cshid=")&&"slh"!==a&&(b="&cshid="+e._cshid);return"/"+(f||"gen_204")+"?atyp=i&ct="+String(a)+"&cad="+(c+d)+"&zx="+String(Date.now())+b};g=google.kEI;google.getEI=k;google.getLEI=l;google.ml=function(){return null};google.log=function(a,c,b,f,d){b||(b=p(a,c,f,d));if(b=m(b)){a=new Image;var n=h.length;h[n]=a;a.onerror=a.onload=a.onabort=function(){delete h[n]};a.src=b}};google.logUrl=function(a){return p("",a)};}).call(this);(function(){google.y={};google.sy=[];google.x=function(a,b){if(a)var c=a.id;else{do c=Math.random();while(google.y[c])}google.y[c]=[a,b];return!1};google.sx=function(a){google.sy.push(a)};google.lm=[];google.plm=function(a){google.lm.push.apply(google.lm,a)};google.lq=[];google.load=function(a,b,c){google.lq.push([[a],b,c])};google.loadAll=function(a,b){google.lq.push([a,b])};google.bx=!1;google.lx=function(){};}).call(this);google.f={};(fun [... cut ...] 2Saiba mais\x22,\x22psrc\x22:\x22Esta pesquisa foi removida do seu\\u003Ca href\x3d\\\x22/history\\\x22\\u003EHist⸮rico da web\\u003C/a\\u003E\x22,\x22psrl\x22:\x22Remover\x22,\x22sbit\x22:\x22Pesquisa por imagem\x22,\x22srch\x22:\x22Pesquisa Google\x22},\x22ovr\x22:{},\x22pq\x22:\x22\x22,\x22rfs\x22:[],\x22sbas\x22:\x220 3px 8px 0 rgba(0,0,0,0.2),0 0 0 1px rgba(0,0,0,0.08)\x22,\x22stok\x22:\x225VL5H7y7wDlkn9BS2kjWgD43kNw\x22}}';google.pmc=JSON.parse(pmc);})();</script>       </body></html>

```