espressif/arduino-esp32 #8048: streamFile does not work

pptsk, 2023-04-07:
### Board

MH-ET LIVE ESP32 MiniKIT

### Device Description

W5500 ethernet mini board

### Hardware Configuration

no

### Version

v2.0.7

### IDE Name

Arduino Ide 1.8.19

### Operating System

Windows 10 x64

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

server.streamFile() does not works. Nothing send to client, page is loading forever.
File is open OK

### Sketch

```cpp
#include "FS.h"
#include <LittleFS.h> 
#include <EthernetWebServer.h>
#include <SPI.h>
#include <Ethernet_Generic.h>
#include <soc/spi_pins.h>
#include <WiFi.h>

#define DEBUG_ETHERNET_GENERIC_PORT         Serial
#define DEBUG_ETHERNET_WEBSERVER_PORT       Serial
#define _ETG_LOGLEVEL_                      4
#define _ETHERNET_WEBSERVER_LOGLEVEL_       1
#define USE_W5100                           false  // Default true to use W5100. Must change to false for W5200, W5500, W5100S, W6100 for faster SPI clock
#warning Use ESP32 architecture
#define ETHERNET_USE_ESP32
#define BOARD_TYPE      "ESP32"
#define W5500_RST_PORT     33
#define USE_THIS_SS_PIN     5
#define ETHERNET_LARGE_BUFFERS
#define USE_ETHERNET_GENERIC   true
#define USE_ETHERNET_ESP8266  false
#define USE_ETHERNET_ENC      false
#define USE_CUSTOM_ETHERNET   false
#define USE_NATIVE_ETHERNET false
//--
#define USE_ETHERNET_WRAPPER   false
#define USE_ETHERNET          false
#define BOARD_NAME    BOARD_TYPE
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
EthernetWebServer server(80);
//-------------------------------------------------------------------------------------------------------------------------
void setup() {
  Serial.begin( 115200 );
  delay( 1000 );
  Serial.flush();
  Serial.println(F("----------------------------------------------------"));
  Serial.println( F("BOOT finished! ")) ;
  Serial.println( F("System loading started... ")) ;
  //-----------------------------------------------------------------------------------------------------------------------
  if (!LittleFS.begin( true )) {                                              // format LittleFS if loading fail
    Serial.println(F("An Error has occurred while mounting LittleFS"));
  } else {
    Serial.print( F("LittleFS total: ") ) ;
    Serial.print( LittleFS.totalBytes() ) ;
    Serial.println( F(" bytes") ) ;
    Serial.print( F("LittleFS used:  ") ) ;
    Serial.print( LittleFS.usedBytes() );
    Serial.println( F(" bytes")) ;
    Serial.print( F("Free heap:     ") ) ;
    Serial.print( ESP.getFreeHeap() );
    Serial.println( F(" bytes")) ;
  }
  WiFi.mode( WIFI_AP );
  //-----------------------------------------------------------------------------------------------------------------------
  server.on("/", HandleRoot );
  //-----------------------------------------------------------------------------------------------------------------------
  Serial.println("\nStarting ESP32_FS_EthernetWebServer on " + String(BOARD_TYPE));
  Serial.println( ETHERNET_WEBSERVER_VERSION );
  Serial.println( ETHERNET_GENERIC_VERSION );
  //-----------------------------------------------------------------------------------------------------------------------
  EthernetInit();
  //-----------------------------------------------------------------------------------------------------------------------
  Serial.print( F("System running on CPU core "));
  Serial.println( xPortGetCoreID() );
  //-----------------------------------------------------------------------------------------------------------------------
  Serial.println(F("System ready!"));
  Serial.println(F("----------------------------------------------------"));
  //-----------------------------------------------------------------------------------------------------------------------
  listDir("/", 3 );
  Serial.println(F("----------------------------------------------------"));
}
//-------------------------------------------------------------------------------------------------------------------------
void loop() {
  server.handleClient();
  delay(100);
}
//-------------------------------------------------------------------------------------------------------------------------
void HandleRoot() {
  if ( LittleFS.exists("/tables.html") ) {
    Serial.println( "serving home page" );
    File file = LittleFS.open("/tables.html", "r");
    if ( file ) {
      Serial.print("  FILE: ");
      Serial.print(file.name());
      Serial.print("\tSIZE: ");
      Serial.println(file.size());
    }
    //--
    //    while (file.available()) {                           // working OK, html code printed
    //      Serial.write(file.read());
    //    }
    //--
    //    server.send( 200, "text/plain", "Hello World!" );    // working OK
    //--
    int fsize = file.size();
    server.sendHeader("Content-Length", (String)(fsize));
    server.streamFile(file, "text/html" );                     // does not work, blank page, loading forever
    file.close();
  }
}
//-------------------------------------------------------------------------------------------------------------------------
void EthernetInit() {
  Ethernet.init ( USE_THIS_SS_PIN );
  Ethernet.begin( mac );
  if ( (Ethernet.getChip() == w5500) || (Ethernet.getChip() == w6100) || (Ethernet.getAltChip() == w5100s) ) {
    //--------------------------------------
    if (Ethernet.getChip() == w6100)
      Serial.print(F("W6100 => "));
    else if (Ethernet.getChip() == w5500)
      Serial.print(F("W5500 => "));
    else
      Serial.print(F("W5100S => "));
    //-----
    Serial.print(F("Speed: "));
    Serial.print(Ethernet.speedReport());
    Serial.print(F(", Duplex: "));
    Serial.print(Ethernet.duplexReport());
    Serial.print(F(", Link status: "));
    Serial.println(Ethernet.linkReport());
    //--------------------------------------
    Serial.println(F("Ethernet cable connected OK!"));
    Serial.print(F("Ethernet server runs at IP: "));
    Serial.println(Ethernet.localIP());
  } else {
    Serial.print(F("Ethernet Chip Not Found!"));
  }
}
//-------------------------------------------------------------------------------------------------------------------------
void listDir(const char * dirname, int levels ) {
  Serial.printf("Listing directory: %s\r\n", dirname);
  File root = LittleFS.open(dirname);
  if (!root) {
    Serial.println(F("- failed to open directory"));
    return;
  }
  if (!root.isDirectory()) {
    Serial.println(F(" - not a directory"));
    return;
  }
  File file = root.openNextFile();
  while (file) {
    if (file.isDirectory()) {
      Serial.print(F("  DIR : "));
      Serial.println(file.name());
      if (levels) {
        listDir( file.name(), levels - 1);
      }
    } else {
      Serial.print(F("  FILE: "));
      Serial.print(file.name());
      Serial.print(F("\tSIZE: "));
      Serial.println(file.size());
    }
    file = root.openNextFile();
  }
}
```

### Debug Message

```plain
serial output:

BOOT finished! 
System loading started... 
LittleFS total: 1441792 bytes
LittleFS used:  12288 bytes
Free heap:     308324 bytes

Starting ESP32_FS_EthernetWebServer on ESP32
EthernetWebServer v2.4.1
Ethernet_Generic v2.8.1
W5500 => Speed: 100 MB, Duplex: FULL DUPLEX, Link status: LINK
Ethernet cable connected OK!
Ethernet server runs at IP: 192.168.0.116
System running on CPU core 1
System ready!
----------------------------------------------------
Listing directory: /
  FILE: tables.html	SIZE: 696
----------------------------------------------------
serving home page
  FILE: tables.html	SIZE: 696
```

### Other Steps to Reproduce

alternative functions works fine:

void HandleRoot () {
  FsFileStream( "/tables.html" );
}
//-------------------------------------------------------------------------------------------------------------------------
void FsFileStream( String fileToStream ) {
  File file = LittleFS.open( fileToStream, "r");
  char buf[1024];
  int fileSize = file.size();
  while (fileSize > 0) {
    size_t len = std::min( (int)(sizeof(buf) - 1), fileSize );
    file.read((uint8_t *)buf, len);
    server.client().write((const char*)buf, len);
    fileSize -= len;
  }
  file.close();
  server.client().stop();
}

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pptsk, 2023-04-07:
here is tables.html 

<!DOCTYPE html>
<html>
<body>

<h2>HTML Tables</h2>

<p>HTML tables start with a table tag.</p>
<p>Table rows start with a tr tag.</p>
<p>Table data start with a td tag.</p>

<hr>
<h2>1 Column:</h2>

<table>
  <tr>
    <td>100</td>
  </tr>
</table>

<hr>
<h2>1 Row and 3 Columns:</h2>
<table>
  <tr>
    <td>100</td>
    <td>200</td>
    <td>300</td>
  </tr>
</table>

<hr>
<h2>3 Rows and 3 Columns:</h2>
<table>
  <tr>
    <td>100</td>
    <td>200</td>
    <td>300</td>
  </tr>
  <tr>
    <td>400</td>
    <td>500</td>
    <td>600</td>
  </tr>
  <tr>
    <td>700</td>
    <td>800</td>
    <td>900</td>
  </tr>
</table>

<hr>

</body>
</html>

pptsk, 2023-04-07:
here is tables.html code
```
<!DOCTYPE html>
<html>
<body>

<h2>HTML Tables</h2>

<p>HTML tables start with a table tag.</p>
<p>Table rows start with a tr tag.</p>
<p>Table data start with a td tag.</p>

<hr>
<h2>1 Column:</h2>

<table>
  <tr>
    <td>100</td>
  </tr>
</table>

<hr>
<h2>1 Row and 3 Columns:</h2>
<table>
  <tr>
    <td>100</td>
    <td>200</td>
    <td>300</td>
  </tr>
</table>

<hr>
<h2>3 Rows and 3 Columns:</h2>
<table>
  <tr>
    <td>100</td>
    <td>200</td>
    <td>300</td>
  </tr>
  <tr>
    <td>400</td>
    <td>500</td>
    <td>600</td>
  </tr>
  <tr>
    <td>700</td>
    <td>800</td>
    <td>900</td>
  </tr>
</table>

<hr>

</body>
</html>
```

lbernstone, 2023-04-07:
Does it work if you don't send the content length?  Maybe you are not calculating it correctly?

mrengineer7777, 2023-04-07:
Unfortunately EthernetWebServer.h is not part of this repository.

pptsk, 2023-04-08:
hi, this show there is nothing  sent:
```
void HandleRoot() {
  if ( LittleFS.exists("/tables.html") ) {
    Serial.println( "serving home page" );
    File file = LittleFS.open("/tables.html", "r");
    if ( file ) {
      Serial.print("  FILE: ");
      Serial.print(file.name());
      Serial.print("\tSIZE: ");
      Serial.println(file.size());
    }
    uint16_t dataSentSize = server.streamFile( file, "text/html" );
    if (dataSentSize != file.size()) {
      Serial.printf("\nSent less data (%i) than expected! (%i)", dataSentSize, file.size());
    } else {
      Serial.println("\nPage served!");
      Serial.printf("\nSent data (%i) as expected! (%i)", dataSentSize, file.size());
    }
    file.close();
  }
}
```
output: 
```
----------------------------------------------------
BOOT finished! 
System loading started... 
LittleFS total: 1441792 bytes
LittleFS used:  12288 bytes
Free heap:     345048 bytes

Starting ESP32_FS_EthernetWebServer on ESP32
EthernetWebServer v2.4.1
Ethernet_Generic v2.8.1
W5500 => Speed: 100 MB, Duplex: FULL DUPLEX, Link status: LINK
Ethernet cable connected OK!
Ethernet server runs at IP: 192.168.0.116
System running on CPU core 1
System ready!
----------------------------------------------------
Listing directory: /
  FILE: tables.html	SIZE: 696
----------------------------------------------------
serving home page
  FILE: tables.html	SIZE: 696

Sent less data (0) than expected! (696)
```

pptsk, 2023-04-08:
Can someone help, if this problem is releated to EthernetWebServer library ?

mrengineer7777, 2023-04-08:
Try reading from the file and printing the contents.  If that works then the issue is in EthernetWebServer.

closed 2023-04-09 by pptsk

mrv96, 2024-10-30:
This should fix EthernetWebServer: https://github.com/khoih-prog/EthernetWebServer/commit/b6f3bee14f975ffd03de1852277b74726d7e13f5