espressif/arduino-esp32 #8049: esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure

SreekanthRayavalasa, 2023-04-08:
### Board

ESP32 CAM

### Device Description

plain ESP32CAM connected to its camera.

Please see attached image.

[image]

### Hardware Configuration

ESP32 CAM board.

[image]

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am doing a ESP32 CAM project in Arduino IDE

trying to upload the ESP32 CAM photo to AWS S3 bucket

i followed exactly below tutorial which explains how to upload ESP32 CAM image to AWS S3 bucket

https://www.youtube.com/watch?v=FIPkU-gx_kU

1) Code is compiling in arduino ide
2) Code is uploading to ESP32CAM board
3) Internet getting connected

but getting below error while trying to connect to AWS server. looks like certificte verification problem.

Can anyone please help to resolve the error ? Note: i need use only arduino ide and not ESP-IDF.

WIFI status = 1
Establishing connection to WiFi..
Taking picture...
length is43600
Time:1680319567MAC: 08:B6:1F:3B:F4:60E (11895) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (11895) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (11904) esp-tls: create_ssl_handle failed
E (11908) esp-tls: Failed to open new connection
E (11912) TRANSPORT_BASE: Failed to open a new connection
E (11919) HTTP_CLIENT: Connection failed, sock [ 0
HTTP_EVENT_ERROR
HTTP_EVENT_DISCONNECTED

### Sketch

```cpp
// Full sketch is below:

#include "Arduino.h"
#include "esp_http_client.h"
#include "esp_camera.h"
#include <WiFi.h>
#include "Base64.h"
#include "mbedtls/base64.h"
#include <NTPClient.h>
#include <WiFiUdp.h>

// Change the WIFI name and password
// Change the awswebaddress below

const char* ssid = "******";
const char* password = "*******";
int capture_interval = 60000; // Milli seconds between captures

bool internet_connected = false;
long current_millis;
long last_capture_millis = 0;

// CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void setup()
{
  Serial.begin(115200);

  if (init_wifi()) { // Connected to WiFi
    internet_connected = true;
    Serial.println("Internet connected");
  }

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }
}

bool init_wifi()
{
  int connAttempts = 0;
  Serial.println("\r\nConnecting to: " + String(ssid));
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED ) {
    delay(500);
    Serial.print(".");
    if (connAttempts > 10) return false;
    connAttempts++;
  }
  return true;
}

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
  switch (evt->event_id) {
    case HTTP_EVENT_ERROR:
      Serial.println("HTTP_EVENT_ERROR");
      break;
    case HTTP_EVENT_ON_CONNECTED:
      Serial.println("HTTP_EVENT_ON_CONNECTED");
      break;
    case HTTP_EVENT_HEADER_SENT:
      Serial.println("HTTP_EVENT_HEADER_SENT");
      break;
    case HTTP_EVENT_ON_HEADER:
      Serial.println();
      Serial.printf("HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
      break;
    case HTTP_EVENT_ON_DATA:
      Serial.println();
      Serial.printf("HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
      if (!esp_http_client_is_chunked_response(evt->client)) {
        // Write out data
        // printf("%.*s", evt->data_len, (char*)evt->data);
      }
      break;
    case HTTP_EVENT_ON_FINISH:
      Serial.println("");
      Serial.println("HTTP_EVENT_ON_FINISH");
      break;
    case HTTP_EVENT_DISCONNECTED:
      Serial.println("HTTP_EVENT_DISCONNECTED");
      break;
  }
  return ESP_OK;
}

static esp_err_t take_send_photo()
{
  Serial.println("Taking picture...");
  camera_fb_t * fb = NULL;
  esp_err_t res = ESP_OK;

 
  fb = esp_camera_fb_get();
  if (!fb) {
    Serial.println("Camera capture failed");
    return ESP_FAIL;
  }
  
   
  int image_buf_size = 4000 * 1000;                                                  
 uint8_t *image = (uint8_t *)ps_calloc(image_buf_size, sizeof(char));

 size_t length=fb->len;

 size_t olen;

 Serial.print("length is");

 Serial.println(length);
 int err1 = mbedtls_base64_encode(image, image_buf_size, &olen, fb->buf, length);

 // const char* imageChar = (char*) image;
//  
 // int str_len = imageChar.length() + 1;
//  char image_char[str_len];
//  image.toCharArray(image_char, str_len);

  esp_http_client_handle_t http_client;
  
  esp_http_client_config_t config_client = {0};

   WiFiUDP ntpUDP;
   NTPClient timeClient(ntpUDP);
   timeClient.begin();
   timeClient.update();
   String Time =  String(timeClient.getEpochTime());
   String MAC = String(WiFi.macAddress());
   Serial.print("Time:" );  Serial.print(Time);
   Serial.print("MAC: ");  Serial.print(MAC);

   String post_url2 = "https://7l617yjpuj.execute-api.ap-south-1.amazonaws.com/dev/" + MAC + "/" + Time; // Location where images are POSTED
   char post_url3[post_url2.length() + 1];
   post_url2.toCharArray(post_url3, sizeof(post_url3));
  
  config_client.url = post_url3;
  config_client.event_handler = _http_event_handler;
  config_client.method = HTTP_METHOD_POST;

  http_client = esp_http_client_init(&config_client);

   esp_http_client_set_post_field(http_client, (const char *)fb->buf, fb->len);
  

  esp_http_client_set_header(http_client, "Content-Type", "image/jpg");

  esp_err_t err = esp_http_client_perform(http_client);
  if (err == ESP_OK) {
    Serial.print("esp_http_client_get_status_code: ");
    Serial.println(esp_http_client_get_status_code(http_client));
  }

  esp_http_client_cleanup(http_client);

  esp_camera_fb_return(fb);
}

void loop()
{
  // TODO check Wifi and reconnect if needed
  
  current_millis = millis();
  if (current_millis - last_capture_millis > capture_interval) { // Take another picture
    last_capture_millis = millis();
    take_send_photo();
  }
}
```

### Debug Message

```plain
WIFI status = 1
Establishing connection to WiFi..
Taking picture...
length is43600
Time:1680319567MAC: 08:B6:1F:3B:F4:60E (11895) esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference
E (11895) esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)
E (11904) esp-tls: create_ssl_handle failed
E (11908) esp-tls: Failed to open new connection
E (11912) TRANSPORT_BASE: Failed to open a new connection
E (11919) HTTP_CLIENT: Connection failed, sock [ 0
HTTP_EVENT_ERROR
HTTP_EVENT_DISCONNECTED
```

### Other Steps to Reproduce

i have checked some solutions, but they are related to ESP-IDF menuconfig edit to skip server verification. your help will be apperciated.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

marchingband, 2023-07-11:
I too am running into this, were you able to solve it @SreekanthRayavalasa ?

marchingband, 2023-07-12:
@SreekanthRayavalasa using platformio, I ported this example to esp-idf, and after setting CONFIG_ESP_TLS_INSECURE and CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY, it now works. Possibly you could find those `#define`s inside the Arduino libraries and change them? I don't know. good luck.

SreekanthRayavalasa, 2023-08-15:
@marchingband : glad that it now works for you..I checked in arduino to enable insecure and skip server verification. however i am not successful. getting same error. trying to do the same to port this code to esp-idf. can you please tell me how to port it ? can you please share me any example porting tutorials ? thanks a lot in advance

marchingband, 2023-08-15:
I switched the Arduino functions to the esp-idf equivalent, and reformatted it to work with the esp-idf framework. You will likely have to learn how esp-idf works. This means removing Arduino from the project completely. I don't know of any tutorials on this subject, sorry.

VojtechBartoska, 2023-08-18:
@SreekanthRayavalasa can we close this ticket?

marchingband, 2023-08-18:
@VojtechBartoska IMO Arduino would IDEALLY have an option to emulate the changes to menuconfig that allow low-security mode for testing or low security applications. I know perhaps it's impossible, or difficult, but not having that option is sub-optimal for AWS uploads, and likely many other IOT scenarios.

torntrousers, 2023-08-19:
Isn't it that you just need to tell it the server CA certificate? Something like define that as a string:
```
const char AwsCACertPem[] = R"EOF(
-----BEGIN CERTIFICATE-----
MIIEkjCCA3qgAwIBAgITBn+USionzfP6wq4rAfkI7rnExjANBgkqhkiG9w0BAQsF
ADCBmDELMAkGA1UEBhMCVVMxEDAOBgNVBAgTB0FyaXpvbmExEzARBgNVBAcTClNj
b3R0c2RhbGUxJTAjBgNVBAoTHFN0YXJmaWVsZCBUZWNobm9sb2dpZXMsIEluYy4x
OzA5BgNVBAMTMlN0YXJmaWVsZCBTZXJ2aWNlcyBSb290IENlcnRpZmljYXRlIEF1
dGhvcml0eSAtIEcyMB4XDTE1MDUyNTEyMDAwMFoXDTM3MTIzMTAxMDAwMFowOTEL
MAkGA1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJv
b3QgQ0EgMTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALJ4gHHKeNXj
ca9HgFB0fW7Y14h29Jlo91ghYPl0hAEvrAIthtOgQ3pOsqTQNroBvo3bSMgHFzZM
9O6II8c+6zf1tRn4SWiw3te5djgdYZ6k/oI2peVKVuRF4fn9tBb6dNqcmzU5L/qw
IFAGbHrQgLKm+a/sRxmPUDgH3KKHOVj4utWp+UhnMJbulHheb4mjUcAwhmahRWa6
VOujw5H5SNz/0egwLX0tdHA114gk957EWW67c4cX8jJGKLhD+rcdqsq08p8kDi1L
93FcXmn/6pUCyziKrlA4b9v7LWIbxcceVOF34GfID5yHI9Y/QCB/IIDEgEw+OyQm
jgSubJrIqg0CAwEAAaOCATEwggEtMA8GA1UdEwEB/wQFMAMBAf8wDgYDVR0PAQH/
BAQDAgGGMB0GA1UdDgQWBBSEGMyFNOy8DJSULghZnMeyEE4KCDAfBgNVHSMEGDAW
gBScXwDfqgHXMCs4iKK4bUqc8hGRgzB4BggrBgEFBQcBAQRsMGowLgYIKwYBBQUH
MAGGImh0dHA6Ly9vY3NwLnJvb3RnMi5hbWF6b250cnVzdC5jb20wOAYIKwYBBQUH
MAKGLGh0dHA6Ly9jcnQucm9vdGcyLmFtYXpvbnRydXN0LmNvbS9yb290ZzIuY2Vy
MD0GA1UdHwQ2MDQwMqAwoC6GLGh0dHA6Ly9jcmwucm9vdGcyLmFtYXpvbnRydXN0
LmNvbS9yb290ZzIuY3JsMBEGA1UdIAQKMAgwBgYEVR0gADANBgkqhkiG9w0BAQsF
AAOCAQEAYjdCXLwQtT6LLOkMm2xF4gcAevnFWAu5CIw+7bMlPLVvUOTNNWqnkzSW
MiGpSESrnO09tKpzbeR/FoCJbM8oAxiDR3mjEH4wW6w7sGDgd9QIpuEdfF7Au/ma
eyKdpwAJfqxGF4PcnCZXmTA5YpaP7dreqsXMGz7KQ2hsVxa81Q4gLv7/wmpdLqBK
bRRYh5TmOTFffHPLkIhqhBGWJ6bt2YFGpn6jcgAKUj6DiAdjd4lpFw85hdKrCEVN
0FE6/V1dN2RMfjCyVSRCnTawXZwXgWHxyvkQAiSr6w10kY17RSlQOYiypok1JR4U
akcjMS9cmvqtmg5iUaQqqcT5NJ0hGA==
-----END CERTIFICATE-----)EOF";
```
and then set that on the http client config:
```
config_client.cert_pem = AwsCACertPem;
```
(i've not tried this, just a passing thought as i read read the issue)

VojtechBartoska, 2023-10-23:
@SreekanthRayavalasa Hi, can this be closed?

SreekanthRayavalasa, 2023-10-23:
Hi,

Yes please.  It can be closed.

Regards
Sreekanth

closed 2023-12-12 by VojtechBartoska