espressif/arduino-esp32 #8050: ESP32 package update in using Additional Url isn't responding

rikilshah, 2023-04-08:
### Board

ESP32

### Device Description

ESP32-WROVER

### Hardware Configuration

No

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

NA

### PSRAM enabled

yes

### Upload speed

NA

### Description

I am trying to install Arduino Core for ESP32 in Arduino IDE 2.0.4 on Windows 11 using additional URL method described in official documentation. 

[https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html](url)

In this method, JSON url is given as below: 

[https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json](url)

I pasted this link in File>Preferences>Additional URLs 

When the board index updates, It shows error saying "Some index Can't be updated" . The same error is repeated when I try to install ESP32 core using arduino-cli. 

One thing I tried is to install other core to verify my setup. I successfully installed RP2040 core using JSON URL method. So I can say my steps are right. 

Please let me know solution to this trivial issue. 

### Sketch

```cpp
NA
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-04-09:
Could you maybe have the json links added in a wrong way? I think one per line is how it should be. People here definitely do not have issues installing from Board Manager

rikilshah, 2023-04-10:
Yes, I indeed suspected that. So, I removed all other URLs and only kept ESP32 JSON Link. Still The issue persists. 

Just another observation, I have other two URLs, RP2040 and STM32. 

RP2040 works. When I open JSON URL in browser, I get JSON in response. 

STM32 and ESP32's JSON URLs doesn't give anything back when opened in browser.

me-no-dev, 2023-04-10:
Can you please try with the following link instead: https://espressif.github.io/arduino-esp32/package_esp32_index.json

Please report back the result

rikilshah, 2023-04-10:
Hey, This link works very well! Thanks a lot! 

Wonder why this link is not updated in official documentation!

me-no-dev, 2023-04-10:
I wonder why you are having issues with the other one. In any case, we will update the docs. Both links point to the same file :)

rikilshah, 2023-04-10:
> I wonder why you are having issues with the other one. In any case, we will update the docs. Both links point to the same file :)

Well, I am facing similar issue for stm32duino Arduino core. So, there might be issue on github side. Thanks again!

rikilshah, 2023-04-10:
For future reference, This issue is specific to India's JioFiber connections. Using Google's Public DNS 8.8.8.8 solved the issue immediately.