espressif/arduino-esp32 #8054: Possible isssue on ESP.getChipModel() with D0WDR2

0x0fe, 2023-04-09:
### Board

ESP32D0WR2

### Device Description

custom module with ESP32D0WDR2

### Hardware Configuration

no relevant

### Version

v2.0.7

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

ESP.getChipModel() returns Unknown, but it should return ESP32-D0WDR2-V3
revision is returned correctly by ESP.getChipRevision() (3 in this case).

https://github.com/espressif/arduino-esp32/blob/b98255d8d1be7280e767f15a8c759e13e00f4b4a/cores/esp32/Esp.cpp#L290

### Sketch

```cpp
uint32_t chipId = 0;
void get_chip_id(void){

  for(int i=0; i<17; i=i+8) {
    chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
  }

  printf("ESP32 Chip model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
  printf("%d cores\n", ESP.getChipCores());
  Serial.print("Chip ID: "); Serial.println(chipId);
}
```
```

### Debug Message

```plain
ESP32 Chip model = Unknown Rev 3
2 cores
Chip ID: xxxxxxx
```
```

### Other Steps to Reproduce

It works fine with other variants, like D0WDQ6 for example.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-10:
Likely resolved by pr #7958

0x0fe, 2023-04-10:
this commit was merged into 2.0.7?

me-no-dev, 2023-04-10:
not part of 2.0.7, but will be of 2.0.8

0x0fe, 2023-04-10:
alright, well noted, i will modify my local version for now.

closed 2023-04-10 by 0x0fe