espressif/arduino-esp32 #8055: ESP32-S3 with 32MB Flash File System Error

Oopas, 2023-04-10:
### Board

ESP32-S3R8

### Device Description

This is a circuit board that I designed myself, so it has no name, including an esp32-s3r8 and 32mb flash (MX25L25645GM2I-10G).

### Hardware Configuration

NONE

### Version

latest master (checkout manually)

### IDE Name

Test on PlatformIO and Arduino IDE

### Operating System

Windows 10

### Flash frequency

It seems to be 80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I initially used a 16mb flash (W25Q128JVPIQ). Littlefs runs great without any issues. But now when I replace it with a 32mb flash (MX25L25645GM2I-10G), I find that littlefs cannot be initialized. I have tried re uploading the littlefs image and directly using the `LittleFS.format()` function to reformat the file system, but none of them have been effective.

Then I tried using fatfs (FFat_Test), but still couldn't initialize the file system.

Finally, I tried using spifs (SPIFFS_Test). At first, it tried to format the file system and seemed to be successful. Then, when the sample read file operation was run, the serial port started printing meaningless code, and it seemed that the file system could not be used either.

I think there is no problem with my flash because the program can be burned normally, but why am I unable to use all the file systems 😭 ?

### Sketch

```cpp
Native example, SPIFFS_Test FFAT_Test and LITTLEFS_Test, I just changed the serial port baud rate to 9600.
```

### Debug Message

```plain
Just serial output file system initialization failed.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-04-10:
you did not mention which partition scheme you chose, but anyway we do not currently have one for 32MB flash. Will have it as part of 3.0.0

Oopas, 2023-04-10:
> you did not mention which partition scheme you chose, but anyway we do not currently have one for 32MB flash. Will have it as part of 3.0.0

I chose `large_spiffs_16MB.csv`. I followed the normal layout of 16MB flash without making any modifications. At least I need to make sure this flash is working properly before I can modify the layout file.

I have currently re soldered the flash chip to ensure that there will be no issues with poor release, but the situation has not improved.

mrengineer7777, 2023-04-10:
Does it work with `default.csv` partition definition?

Oopas, 2023-04-10:
> Does it work with `default.csv` partition definition?

Based on your suggestion, I modified the layout file and retested three file systems. Here are my steps:

1. Compile firmware with sample files(SPIFFS_Test LITTLEFS_Test  FFat_Test) and upload.
2. Upload file system image.
3. Reboot the motherboard to run the new firmware.

Litlefs and Fatfs still reported an error and were unable to initialize the file system, but SPIFFS seems to be able to initialize the file system, but it cannot be used properly.

This is serial output, initially everything seemed normal, but when trying to read hello.txt, the serial output was meaningless and garbled.

```
Start !

Total: 1378241 Used: 502
                        Listing directory: /
  FILE: Debug   SIZE: 4
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
▒▒
``` 

After seeing the serial output, I closed the motherboard and used a programmer to read the content of the flash chip. I uploaded it in a compressed package with the firmware source code and file system image, hoping it would be helpful. [SPIFFS TEST.zip](https://anonfiles.com/teyeu9k1z2/SPIFFS_TEST_zip)

mrengineer7777, 2023-04-10:
It looks like the test program read a lot more than 4 bytes from `hello.txt`.  I hate to say this but it sounds like an issue with your hardware.

mrengineer7777, 2023-04-11:
Nice find.  Will be interesting to hear your results.

Oopas, 2023-04-13:
> Nice find. Will be interesting to hear your results.

Sorry for being a bit late.

I have successfully established littlefs on 32MB flash(W25Q256FVEIG) and it runs great. So it should be because MX25L25645GM2I-10G is not supported.

Thanks for your help 😉

closed 2023-04-13 by Oopas

mrengineer7777, 2023-04-14:
That's awesome! Thanks for the update :)

obdevel, 2023-07-06:
Has anybody tested the ESP32-S3-WROOM-2-N32R8V module ? I don't know which flash chip it uses but I guess it must be one of the two supported chips from the doc @Oopas quoted above.

I'm starting a new hardware design and it's for audio projects, so I'd like as much file storage as possible.

Other than selecting OPI mode for flash and PSRAM, are there any other gotchas ?

Thanks.