espressif/arduino-esp32 #8057: Major issue on D0DR2 with A2DP

0x0fe, 2023-04-10:
### Board

ESP32D0WR2

### Device Description

ESP32W0DR2 module 

### Hardware Configuration

MCLK 3
WS 5
SCK 18
BCK 21
MUTE 4

### Version

v2.0.7

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

So the context is a consumer product design which uses an external DAC and provide audio playback via different sources, MMC, wifi and bluetooth (A2DP), initial tests were conducted on ESP32D0WDQ6 and everything worked fine, the A2DP is smooth, no stuttering or anything. 

So for final implementation we switches to D0WDR2 since we use both wifi and bluetooth and a relatively large application logic, additional PSRAM is required. 
Sadly we found that the D0WDR2 cannot work normally for the A2DP, there is a massive stuttering, and the playback is totally abnormal. There is no playback issuewhen playing from MMC.

In order to confirm the problem we performed multiple test, at first changing compile time options such as Flash speed (40M and 80M) CPU speed (160M and 240M) PSRAM enabled/disabled, and even buffer sizes for I2S and A2DP, we found no difference at all, in all cases the stuttering was still present with the same magnitude on the D0WDR2 and no stuttering at all on D0WDQ6. 

Then we  decided to make comparative tests with devkit instead of the product hardware, we changed the DAC for one without master clock (MCLK) as i suspected it could be related to MCLK, so we used : 
-one devkit with D0WDQ6 + MAX98357A
-one devkit with D0WDR2 + MAX98357A
 Identical pinmap, and identical firmware binary flashedin both.

We tested the same way, with an iphone as source and running a simple A2DP firmware.:
-D0WDQ6 : perfectly smooth, no stuttering at all
-D0WDR2 : massive stuttering all over, unusable

With the log in verbose, there is absolutely no error or warning of any kind, both output exactly the same thing. So the problem lies somewhere deep and is not a simple buffer underrun or something (well unless it doesnt gets catched).

Sadly, we can now confirm the problem comes from the D0WDR2 chip itself and not the hardware, the DAC or the firmware.
I have no idea yet of what causes this, nor what exactly changed in the die between D0WDQ6 and D0WDR2, both are rev 3 so i would have expected the D0WDR2 just be a D0WDQ6 with a PSRAM attached, but apaprently it is not the case ,I2S block? wifi stack? who knows.

So please espressif have a look at that.  Attached is a quick audio recording of the problem (from an iphone).

[Dowdr2 a2dp stuttering.zip](https://github.com/espressif/arduino-esp32/files/11191033/Dowdr2.a2dp.stuttering.zip)

### Sketch

```cpp
simple A2DP working perfectly fine on the D0WDQ6
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

can be reproduced with a devkit and any DAC. With or withotu MCLK. Did not try on internal DAC since it is not relevant, we don't use and wont use internal DAC.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-04-10:
Hi @0x0fe I will suggest you post an issue in ESP-IDF's tracker, because we are just a consumer of their A2DP API here. State that you are on ESP-IDF v4.4.4. I hope they will be able to shed some light on why this could be.

closed 2023-04-10 by 0x0fe

0x0fe, 2023-04-17:
aditional question on this issue, is there a way to enable bluedroid log in the arduino core? with a compile flag, i guess.