espressif/arduino-esp32 #8060: BLE - Unable to attach more than 7 characteristics to service

TheDJVG, 2023-04-10:
### Board

ESP32 Dev Module

### Device Description

Plain module connected over micro USB,

### Hardware Configuration

Nothing.

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Manjaro

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When I add more than 7 characteristics to a service only 7 show up when scanned:
[image]

When a characteristic has a descriptor attached only 6 show up.

Important is that when dumping the service everything shows up:
```
[  4933][D][BLEService.cpp:110] dump(): Characteristics:
handle: 0x002a, uuid: 00000000-0000-0000-0000-000000000001
handle: 0x002c, uuid: 00000000-0000-0000-0000-000000000002
handle: 0x002e, uuid: 00000000-0000-0000-0000-000000000003
handle: 0x0030, uuid: 00000000-0000-0000-0000-000000000004
handle: 0x0032, uuid: 00000000-0000-0000-0000-000000000005
handle: 0x0034, uuid: 00000000-0000-0000-0000-000000000006
handle: 0x0036, uuid: 00000000-0000-0000-0000-000000000007
handle: 0x0000, uuid: 00000000-0000-0000-0000-000000000008
handle: 0x0000, uuid: 00000000-0000-0000-0000-00000000000a
handle: 0x0000, uuid: 00000000-0000-0000-0000-00000000000b
```

The sketch provided should work and is based on the example found here:  [libraries/BLE/examples/BLE_server_multiconnect/BLE_server_multiconnect.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server_multiconnect/BLE_server_multiconnect.ino)

### Sketch

```cpp
/*
    Video: https://www.youtube.com/watch?v=oCMOYS71NIU
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updated by chegewara
   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
   And has a characteristic of: beb5483e-36e1-4688-b7f5-ea07361b26a8
   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.
   A connect hander associated with the server starts a background task that performs notification
   every couple of seconds.
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>

BLEServer* pServer = NULL;
BLECharacteristic* pChar1 = NULL;
BLECharacteristic* pChar2 = NULL;
BLECharacteristic* pChar3 = NULL;
BLECharacteristic* pChar4 = NULL;
BLECharacteristic* pChar5 = NULL;
BLECharacteristic* pChar6 = NULL;
BLECharacteristic* pChar7 = NULL;
BLECharacteristic* pChar8 = NULL;
BLECharacteristic* pChar9 = NULL;
BLECharacteristic* pChar10 = NULL;

bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID "00000000-0000-0000-0000-000000000000"
#define CHAR_1 "00000000-0000-0000-0000-000000000001"
#define CHAR_2 "00000000-0000-0000-0000-000000000002"
#define CHAR_3 "00000000-0000-0000-0000-000000000003"
#define CHAR_4 "00000000-0000-0000-0000-000000000004"
#define CHAR_5 "00000000-0000-0000-0000-000000000005"
#define CHAR_6 "00000000-0000-0000-0000-000000000006"
#define CHAR_7 "00000000-0000-0000-0000-000000000007"
#define CHAR_8 "00000000-0000-0000-0000-000000000008"
#define CHAR_9 "00000000-0000-0000-0000-00000000000a"
#define CHAR_10 "00000000-0000-0000-0000-00000000000b"

class MyServerCallbacks : public BLEServerCallbacks {
  void onConnect(BLEServer* pServer) {
    deviceConnected = true;
    BLEDevice::startAdvertising();
  };

  void onDisconnect(BLEServer* pServer) {
    deviceConnected = false;
  }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService* pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pChar1 = pService->createCharacteristic(
    CHAR_1,
    BLECharacteristic::PROPERTY_READ);

  pChar2 = pService->createCharacteristic(
    CHAR_2,
    BLECharacteristic::PROPERTY_READ);

  pChar3 = pService->createCharacteristic(
    CHAR_3,
    BLECharacteristic::PROPERTY_READ);

  pChar4 = pService->createCharacteristic(
    CHAR_4,
    BLECharacteristic::PROPERTY_READ);

  pChar5 = pService->createCharacteristic(
    CHAR_5,
    BLECharacteristic::PROPERTY_READ);

  pChar6 = pService->createCharacteristic(
    CHAR_6,
    BLECharacteristic::PROPERTY_READ);

  pChar7 = pService->createCharacteristic(
    CHAR_7,
    BLECharacteristic::PROPERTY_READ);

  pChar8 = pService->createCharacteristic(
    CHAR_8,
    BLECharacteristic::PROPERTY_READ);

  pChar9 = pService->createCharacteristic(
    CHAR_9,
    BLECharacteristic::PROPERTY_READ);

  pChar10 = pService->createCharacteristic(
    CHAR_10,
    BLECharacteristic::PROPERTY_READ);

  // Start the service
  pService->start();

  // Start advertising
  BLEAdvertising* pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(false);
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter
  BLEDevice::startAdvertising();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {
  // notify changed value
  if (deviceConnected) {
    delay(10);  // bluetooth stack will go into congestion, if too many packets are sent, in 6 hours test i was able to go as low as 3ms
  }
  // disconnecting
  if (!deviceConnected && oldDeviceConnected) {
    delay(500);                   // give the bluetooth stack the chance to get things ready
    pServer->startAdvertising();  // restart advertising
    Serial.println("start advertising");
    oldDeviceConnected = deviceConnected;
  }
  // connecting
  if (deviceConnected && !oldDeviceConnected) {
    // do stuff here on connecting
    oldDeviceConnected = deviceConnected;
  }
}
```

### Debug Message

```plain
[  4810][D][BLEAdvertising.cpp:199] start(): - advertising service: 00000000-0000-0000-0000-000000000000
[  4818][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[  4827][V][BLEAdvertising.cpp:252] start(): << start
[  4837][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  4842][V][BLEDevice.cpp:586] startAdvertising(): << startAdvertising
[  4849][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
Waiting a client connection to notify...
[  4865][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  4877][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[  4886][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  4894][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[  4904][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  4912][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[  4921][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  4929][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[  4939][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[  4947][V][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent
[  4957][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[  4964][V][BLEUtils.cpp:1056] dumpGapEvent(): [status: 0]
[  4968][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  4974][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 0]
[  4981][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[  4990][V][BLEUtils.cpp:1074] dumpGapEvent(): [status: 0]
[  4996][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[  5001][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-11:
@TheDJVG - 
The number of characteristics and descriptors may be limited to the number of handles that the BLE Service has been configured.
By default, a Service is created with 15 handles. 
Each BLE Characteristic takes 2 handles and each BLE Descriptor takes 1 handle.
This is the reason for just seeing 7 Characteristics (7 x 2 = 14 handles - limited to 15).

In order to change this limit, your Sketch must create the Service telling it how many Handles your application may need.
Using your example with 10 BLE Characteristics, its Service may need at least 20 handles.

You can set it by using another constructor:
``` cpp

  // Create the BLE Service
  BLEService* pService = pServer->createService(BLEUUID(SERVICE_UUID), 32); // 32 available handles

```

TheDJVG, 2023-04-11:
Amazing! Not for a single moment did I think to look into the creation of the service itself. I only looked at the Characteristic (Map) creation. This was indeed the case and now I see all the characteristics.

Thanks a lot!

closed 2023-04-11 by TheDJVG