espressif/arduino-esp32 #8067: WiFiSTAClass::disconnect(bool wifioff, bool eraseap) to return false when erase of AP info fails

hemeleerse, 2023-04-12:
### Related area

WiFi

### Hardware specification

ESP32 devkit

### Is your feature request related to a problem?

I'm trying to make the ESP32 connect to the AP with the best signal strength.
Therefore I'm doing a WifiScan, find the BSSID & channel with the strongest signal, do a disconnect(false, true) to erase previous AP info and then do WiFi.begin with the BSSID and channel.
As the disconnect does not return false when the erasure fails, I'm not sure if the erase succeeded

### Describe the solution you'd like

Have the disconnect function return false when erasure of the AP info fails

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-04-27:
@hemeleerse In my opinion disconnect and erase should be separate functions.  Would it be acceptable to introduce a new `eraseAP()` function?

mrengineer7777, 2023-04-27:
Also it appears disconnect will only erase the AP info if the WiFi Mode() has STA enabled.