espressif/arduino-esp32 #8070: An error occurred in the range of the char data type

iBigDane, 2023-04-13:
### Board

ESP32-WROVER-E

### Device Description

development board is ESP-WROVER-KIT_V4.1

### Hardware Configuration

Nothing is connected to the board

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

```
#include <Arduino.h>

void setup()
{
    Serial.begin(115200);
    char a = -25;
    Serial.print("a:");
    Serial.println((int)a);
}

void loop(){

}
```

The output information of the serial port is:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
a:231

```
char cast to int is wrong.
my arduino-esp32 version is 2.0.5
arduion ide is 1.8.13

### Sketch

```cpp
#include <Arduino.h>

void setup()
{
    Serial.begin(115200);
    char a = -25;
    Serial.print("a:");
    Serial.println((int)a);
}

void loop(){

}
```

### Debug Message

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
a:231
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-13:
I've noticed that before.  In ASCII a `char` is 0-127 (no negative).  Perhaps GCC treats `char` the same as `BYTE`?  In that case -25 would indeed wrap around to 256-25 = 231.

What does `Serial.printf("a:%c\r\n", a);` show?

iBigDane, 2023-04-13:
> I've noticed that before. In ASCII a `char` is 0-127 (no negative). Perhaps GCC treats `char` the same as `BYTE`? In that case -25 would indeed wrap around to 256-25 = 231.
> 
> What does `Serial.printf("a:%c\r\n", a);` show?

20:34:28.847 -> ets Jul 29 2019 12:21:46
20:34:28.847 -> 
20:34:28.847 -> rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
20:34:28.847 -> configsip: 0, SPIWP:0xee
20:34:28.847 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:34:28.847 -> mode:DIO, clock div:1
20:34:28.847 -> load:0x3fff0030,len:1344
20:34:28.892 -> load:0x40078000,len:13864
20:34:28.892 -> load:0x40080400,len:3608
20:34:28.892 -> entry 0x400805f0
20:34:28.982 -> [     3][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
20:34:28.982 -> a:⸮

closed 2025-01-16 by Parsaabasi