espressif/arduino-esp32 #8076: SD Card no longer mounts

jbhopestar, 2023-04-14:
### Board

ESP32 Dev Module

### Device Description

plain module on breadboard

### Hardware Configuration

no just the SD card

### Version

v2.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

SDCard mounting fails ungracefully.  This code worked in the past with older versions.

### Sketch

```cpp
#include "FS.h"
#include "SPI.h"
#include "SD.h"

SPIClass gSpi;
//sdCardSetupVSPI( 18, 19, 23, 5 )
//sdCardSetupHSPI( 14, 12, 13, 15 )

void listDir(fs::FS &fs, const char * dirname, uint8_t levels)
	{
	Serial.printf("Listing directory: %s\n", dirname);

	File root = fs.open(dirname);
	if(!root)
		{
		Serial.println("Failed to open directory");
		return;
		}
	if(!root.isDirectory())
		{
		Serial.println("Not a directory");
		return;
		}

	File file = root.openNextFile();
	while(file)
		{
		if(file.isDirectory())
			{
			Serial.print("  DIR : ");
			Serial.println(file.name());
			if(levels)
				{
				listDir( fs, file.name(), levels -1 );
				}
			} 
		else 
			{
			Serial.print("  FILE: ");
			Serial.print(file.name());
			Serial.print("  SIZE: ");
			Serial.println(file.size());
			}
		file = root.openNextFile();
		}
	}

void createDir(fs::FS &fs, const char * path)
	{
	Serial.printf("Creating Dir: %s\n", path);
	if(fs.mkdir(path))
		{
		Serial.println("Dir created");
		} 
	else 
		{
		Serial.println("mkdir failed");
		}
	}

void removeDir(fs::FS &fs, const char * path)
	{
	Serial.printf("Removing Dir: %s\n", path);
	if(fs.rmdir(path))
		{
		Serial.println("Dir removed");
		} 
	else 
		{
		Serial.println("rmdir failed");
		}
	}

void readFile(fs::FS &fs, const char * path)
	{
	Serial.printf("Reading file: %s\n", path);

	File file = fs.open(path);
	if(!file)
		{
		Serial.println("Failed to open file for reading");
		return;
		}

	Serial.print("Read from file: ");
	while(file.available())
		{
		Serial.write(file.read());
		}
	file.close();
	}

void writeFile(fs::FS &fs, const char * path, const char * message)
	{
	Serial.printf("Writing file: %s\n", path);

	File file = fs.open(path, FILE_WRITE);
	if(!file)
		{
		Serial.println("Failed to open file for writing");
		return;
		}
	if(file.print(message))
		{
		Serial.println("File written");
		} 
	else 
		{
		Serial.println("Write failed");
		}
	file.close();
	}

void appendFile(fs::FS &fs, const char * path, const char * message)
	{
	Serial.printf("Appending to file: %s\n", path);

	File file = fs.open(path, FILE_APPEND);
	if(!file)
		{
		Serial.println("Failed to open file for appending");
		return;
		}
	if(file.print(message))
		{
		Serial.println("Message appended");
		}
	else 
		{
		Serial.println("Append failed");
		}
	file.close();
	}

void renameFile(fs::FS &fs, const char * path1, const char * path2)
	{
	Serial.printf("Renaming file %s to %s\n", path1, path2);
	if (fs.rename(path1, path2)) 
		{
		Serial.println("File renamed");
		}
	else 
		{
		Serial.println("Rename failed");
		}
	}

void deleteFile(fs::FS &fs, const char * path)
	{
	Serial.printf("Deleting file: %s\n", path);
	if(fs.remove(path))
		{
		Serial.println("File deleted");
		} 
	else 
		{
		Serial.println("Delete failed");
		}
	}

void testFileIO(fs::FS &fs, const char * path)
	{
	File file = fs.open(path);
	static uint8_t buf[512];
	size_t len = 0;
	uint32_t start = millis();
	uint32_t end = start;
	
	if( file )
		{
		len = file.size();
		size_t flen = len;
		start = millis();
		while(len)
			{
			size_t toRead = len;
			if(toRead > 512)
				{
				toRead = 512;
				}
			file.read(buf, toRead);
			len -= toRead;
			}
		end = millis() - start;
		Serial.printf("%u bytes read for %u ms\n", flen, end );
		file.close();
		} 
	else 
		{
		Serial.println("Failed to open file for reading");
		}

	file = fs.open(path, FILE_WRITE);
	if(!file)
		{
		Serial.println("Failed to open file for writing");
		return;
		}

	size_t i;
	start = millis();
	for(i=0; i<2048; i++)
		{
		file.write(buf, 512);
		}
	end = millis() - start;
	Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
	file.close();
	}

void cardBegin( int8_t pChipSelect )
	{
  Serial.print( "cardBegin chipSelect ");
  Serial.println(pChipSelect);

	if (!SD.begin( pChipSelect, gSpi, 80000000) ) 
		{
		Serial.println("Card Mount Failed");
		return;
		}

 	Serial.print("cardType ");
	uint8_t cardType = SD.cardType();
	Serial.println(cardType);
	
	if(cardType == CARD_NONE)
		{
		Serial.println("No SD card attached");
		return;
		}

	Serial.print("SD Card Type: ");
	if(cardType == CARD_MMC)
		{
		Serial.println("MMC");
		} 
	else if(cardType == CARD_SD)
		{
		Serial.println("SDSC");
		} 
	else if(cardType == CARD_SDHC)
		{
		Serial.println("SDHC");
		} 
	else 
		{
		Serial.println("UNKNOWN");
		}

	uint64_t cardSize = SD.cardSize() / (1024 * 1024);
	Serial.printf("SD Card Size: %lluMB\n", cardSize);

	listDir(SD, "/", 0);
	createDir(SD, "/mydir");
	listDir(SD, "/", 0);
	removeDir(SD, "/mydir");
	listDir(SD, "/", 2);
	writeFile(SD, "/hello.txt", "Hello ");
	appendFile(SD, "/hello.txt", "World!\n");
	readFile(SD, "/hello.txt");
	deleteFile(SD, "/foo.txt");
	renameFile(SD, "/hello.txt", "/foo.txt");
	readFile(SD, "/foo.txt");
	testFileIO(SD, "/test.txt");
	Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
	Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
	}
	
void microSDSetUpVSPI( int8_t pClock, int8_t pMasterInSlaveOut, int8_t pMasterOutSlaveIn, int8_t pChipSelect )
	{
  Serial.println( "microSDSetUpVSPI ");
  Serial.print( "clockPin ");
  Serial.print( pClock );
  Serial.print(" MasterInSlaveOut " );
  Serial.print(pMasterInSlaveOut );
  Serial.print(" MasterOutSlaveIn " );
  Serial.println( pMasterOutSlaveIn );

	gSpi = SPIClass(VSPI);
	gSpi.begin( pClock, pMasterInSlaveOut, pMasterOutSlaveIn, pChipSelect );
	cardBegin(pChipSelect); 	
	}
	
void microSDSetUpHSPI( int8_t pClock, int8_t pMasterInSlaveOut, int8_t pMasterOutSlaveIn, int8_t pChipSelect )
	{
	gSpi = SPIClass(HSPI);	
	gSpi.begin( pClock, pMasterInSlaveOut, pMasterOutSlaveIn, pChipSelect );	
	cardBegin(pChipSelect); 
	}
```

### Debug Message

```plain
it fails without a warning or error message.
```

### Other Steps to Reproduce

tried different chips, different setups

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-14:
What error are you seeing?

P-R-O-C-H-Y, 2023-04-14:
@jbhopestar Can you try to start SD card on lower frequency? 40Mhz or 20Mhz.

jbhopestar, 2023-04-14:
I traced it, it fails silently at the fmount call of the fat file system.
After that the serial port stops working.
it neither reports a failed mounting nor prints this message out.
Serial.print("cardType ");

what i see is this:

clockPin 18 MasterInSlaveOut 19 MasterOutSlaveIn 23

cardBegin chipSelect 5

and then nothing.
I tried the two different frequencies, same behaviour.  These cards used to
mount at that frequency.  I have spent a few days on this one.  My only
other clue is that I spent a couple of days working on the tinyusb library
trying to get the cards to mount on an S2 Mini, which also involves the
same fatfs file system.  I attempted a reinstall of arduino and all the
libraries.  I deleted the tinyUSB libraries in the documents folder as well.

    FRESULT res = f_mount(fs, drv, 1);
it fails on this line
in sd_diskio.cpp
line 800
no Serial calls after that line work properly.

Simon410, 2023-04-15:
Hi, did you format the SD card with Fat32 default and 32 kilo octets.

lbernstone, 2023-04-15:
Does the SD_Test example work?

jbhopestar, 2023-04-15:
yes.  There is little doubt in my mind that this was caused by some change
in the library.  What concerns me is that we are not hearing this from many
many people, and if that is the case, I suppose the solution is to buy some
new sd cards.  They are older cards.

lbernstone, 2023-04-16:
If the test works, then it must have something to do with how you are calling things.  Start from the working Test example, and pull the pieces into there to see what breaks it.  Start with the SD.begin() parameters, then do the SPI.begin.

jbhopestar, 2023-04-17:
I believe I have discovered the source of the problem in the change in the
library.  I regard it as a serious problem that requires an easy fix.  Many
of my old projects do not work on account of this problem, and other
projects by other authors do not work on account of this problem.  I went
back to the original form of the code to mount the card to verify whether
your assertion is correct.  What I got was the watchdog error that claimed
that I had starved the watchdog from any time, so i placed a delay in the
loop and it still did not work.  I then noticed that one of my old projects
did not fail, and it had the code that runs the task for Core 0, so I
placed that in the code, and placed a delay, and then the card mounted
properly.

There are hundreds of old projects that did not deal with the second core,
my guess is they all project this new problem, which means in the old
library you didnt have to deal with the second processor, while in the new
library you do.  I suspect the old library had that delay in the loop while
checking the reset button.  Is there anyone who can fix this problem?

My code runs fine now, now any project that I ran that did not put a delay
in the core 0 loop does not run unless I do it.  If this is the new
requirement of the new libraries, it would be useful to let programmers
know.

I have gone back to the original form of the code to see if your assertion
has merit, and it still didnt work

jbhopestar, 2023-04-17:
sorry that last line was from an earlier revision.

mrengineer7777, 2023-04-17:
@jbhopestar Can you provide more details?  How did you run on core 0?  Where did you place the delay?  Can you show a code example that would fix the library?

jbhopestar, 2023-04-18:
#define cButton1Pin   35
#define cButton2Pin   0
TaskHandle_t              gSensorEventTask;

void processor2Code( void * pvParameters )
  {
  Serial.printf("processor2Code running on core %d\n", xPortGetCoreID() );

  int button1State = 0;
  int button2State = 0;

  for(;;)
    {
    button1State = digitalRead(cButton1Pin);
    button2State = digitalRead(cButton2Pin);
    if (button1State == LOW)
      {
      //buttonOnePressed();
      delay( 10 );
      }

    if (button2State == LOW)
      {
      buttonTwoPressed();
      delay( 10 );
      }

    delay( 10 );
    }

  }

void buttonOnePressed()
  {
  Serial.println("ButtonOnePressed");
  delay(5);
  }

void buttonTwoPressed()
  {
  Serial.println("ButtonTwoPressed");
  delay(5);
  }
void setup()
  {
  Serial.begin(115200);

  xTaskCreatePinnedToCore( processor2Code, "SensingTask", 10000, NULL, 2, &
gSensorEventTask, 0 );
  }

This is what I had to add to fix the project. I think the changes in the
library have to
do with how the watchdog timer is implemented now. I suspect these changes
came in while
doing the update for the S3 chip.
Somewhere in the library there is a loop looking for a change in state of
the reset
button, the delay needs to be added there.

mrengineer7777, 2023-04-18:
Are you saying you had to add a delay to your Task `processor2Code`?  You are running that task with Priority 2, which is higher than `loop` task priority 1.  If there are no delays in `processor2Code` then you will block execution of `loop` functions.

mrengineer7777, 2023-04-26:
Did you resolve the issue?

jbhopestar, 2023-04-26:
Yes, I did.