espressif/arduino-esp32 #8080: ESP32-S2, LOG outputs to UART0 and Serial.print to USB when CDC is selected

N0ury, 2023-04-15:
### Board

ESP32-S2-SOLO module (not kit)

### Device Description

I'm doing tests on a breadboard

### Hardware Configuration

USB cable is connected to D+ and D- (and GND)
The module is powered externally

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

macOS v11.7.5 Big Sur

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I work in CLI mode under platformio.
I have opened an issue at https://community.platformio.org/ and someone pointed me to
- https://github.com/espressif/arduino-esp32/issues/6245#issuecomment-1031540725
and
- https://github.com/espressif/arduino-esp32/issues/7407
I'm not sur it's the same for me.

I have tried under Arduino IDE and it's the same.
That's why I open this issue here.

With a S2-SOLO (chip ESP32-S2R2) I can flash an app using native USB.
I can also print with Serial.print via native USB.

My problem is that when I use log_x (same as esp-idf ESP_LOGx), the output goes to the UART not USB.
``

### Sketch

```cpp
The app is simply a Serial.println and a log_x.

Platformio.ini:
[env]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
monitor_speed = 115200
upload_speed = 921600
board_upload.use_1200bps_touch = true
board_upload.wait_for_upload_port = true
build_flags =
     -DCORE_DEBUG_LEVEL=3
     -DARDUINO_USB_CDC_ON_BOOT=1
     -DBOARD_HAS_PSRAM
     -mfix-esp32-psram-cache-issue

[env:esp-solo]
    build_src_filter = +<main.cpp>
# For debug
platform_packages = platformio/tool-openocd-esp32
debug_tool = esp-prog
debug_init_break = 
  set remote hardware-watchpoint-limit 2
  break base.cpp:16
debug_extra_cmds = break main.cpp:26

# For testing FRAM
[env:fram]
    build_src_filter = +<fram.cpp>
```

### Debug Message

```plain
No errors, no warnings
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-15:
@SuGlider @igrr can you help?

SuGlider, 2023-04-15:
I can investigate it.

SuGlider, 2023-05-25:
@N0ury - I got it to work:

``` cpp
void setup() {
  // Serial is the TinyUSB interface | USBCDC class | -DARDUINO_USB_CDC_ON_BOOT=1
  
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(500);

  Serial.println("\n=========\nTesting...\n=========\n");
  log_e("TESTING...");
}

void loop() {
  log_e("TESTING LOOP...");
  Serial.println("Loop...");
  delay(1000);
}
```

SuGlider, 2023-05-25:
`Serial.setDebugOutput(true);` activates the LOG output to the `Serial` object, which in this case is attached to the USB CDC port.

@N0ury - Let me know if this works for you.

N0ury, 2023-05-25:
@SuGlider I've tried it and it works fine now.
The issue can be closed.
Perhaps this point should be highlighted in the documentation

closed 2023-05-25 by SuGlider