espressif/arduino-esp32 #8083: Core dump flash config is corrupted!

zekageri, 2023-04-16:
### Board

ESP32-Wrover-E

### Device Description

ESP32 Wrover-E, 8MB psram, 16mb flash

### Hardware Configuration

Lan8720
psram
i2c
spi

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

My application worked for more than a year now.
There was occasional updates but nothing serious.

I have no partition for esp_core_dump.
My partition table is the default large_spiffs_16MB.csv.

My PIO ini file looks like this:

```cpp
[env:esp-wrover-kit]
platform = espressif32

board       = esp-wrover-kit
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L
upload_port     = COM3
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = esp32_exception_decoder

board_upload.flash_size = 16MB
board_build.flash_mode = qio
board_build.partitions = large_spiffs_16MB.csv
board_build.f_flash    = 80000000L

build_flags =   -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
                -mfix-esp32-psram-cache-strategy=memw
                -DCORE_DEBUG_LEVEL=0
```

My sketch does not contain any logic to retrive the core dump from the flash, since i don't use it.

I had the following core dump too:

`esp_corf Eյ }   ͡  Incorrect size of core dump image: 1601512059`

My case is similart to this https://www.esp32.com/viewtopic.php?t=13899
And this https://github.com/HomeSpan/HomeSpan/discussions/490

My sketch never recovers from this and just keeps rebooting.

```cpp
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Retrieving maximum program size .pio\build\esp-wrover-kit\firmware.elf
Checking size .pio\build\esp-wrover-kit\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.9% (used 71664 bytes from 327680 bytes)
Flash: [===       ]  30.6% (used 1441777 bytes from 4718592 bytes)
```

### Sketch

```cpp
Can't
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008b028  PS      : 0x00060130  A0      : 0x800dc646  A1      : 0x3ffd9f40  
A2      : 0x3f91567c  A3      : 0x00000000  A4      : 0x000007d0  A5      : 0x3f91567c
A6      : 0x69756372  A7      : 0x0000007d  A8      : 0x00000000  A9      : 0x3ffd9f10  
A10     : 0x00000000  A11     : 0x3f422507  A12     : 0x000000ff  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008b028  LEND    : 0x4008b03e  LCOUNT  : 0x0000007c

Backtrace: 0x4008b025:0x3ffd9f40 0x400dc643:0x3ffd9f50 0x400fae3d:0x3ffd9f90 0x400fe5ee:0x3ffda000 0x400fe946:0x3ffda1a0 0x400fe97b:0x3ffda1c0 0x400fea14:0x3ffda1e0

ELF file SHA256: c6c1a730c8072444

E (9145) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

### Other Steps to Reproduce

Can't

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-17:
Did you upgrade the firmware from an earlier version of arduino-esp32?  I know at some point the partition defaults changed to add `coredump`.  This can cause corruption until the chip is completely erased and re-written.  I recently added partition settings to my project because of that.

mrengineer7777, 2023-04-17:
Apparently it's called ESP Insights. https://github.com/espressif/arduino-esp32/pull/7566/files.  Added Dec 22, 2022.  Release v2.0.6.

zekageri, 2023-04-17:
Well. I can't find any doc how to use it under arduino. Interesting

mrengineer7777, 2023-04-17:
@zekageri How can we help?

zekageri, 2023-04-18:
I wonder how my sketch got here and why it is wanted to read/write to a partition which does not exists and not configured at all, to avoid it in the future, or to be able to initialize this feature so it can not crash next time.

brightproject, 2024-08-08:
It seems I have a [similar](https://github.com/sivar2311/platformio_boards/issues/2) problem, but it doesn't bother me yet, because the ESP32 microcontroller works as it should.

CountBobby, 2024-08-09:
If you use the standard settings (ESP32-WROOM-DA Module) The Flash Frequency is set to 80Mhz - you need to change it to 40MHz and Flash size to 16Mb (or what your board has)
That worked for me

gdragon2789, 2024-11-30:
I have the same issue, I used esptool.exe to flash the firmware successfully. But when I tried to communicate via UART, I got the messaged and the chip is stuck in the rebooting.

Here is my flash setup. Anyone can help is appreciated.
```python
esptool.exe --chip esp32 --port %COMPORT% --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x00 target.bin

``` 

```python
ASSERT_ERR(node->corrupt_check == KE_LIST_PATTERN), in ke_mem.c at line 275
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090818: 00f01d00 00004136 f01d0000
Core  0 register dump:
PC      : 0x4009081f  PS      : 0x00060534  A0      : 0x80085e09  A1      : 0x3ffda9e0  
A2      : 0x3ff9cacf  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ff9cac6  
A6      : 0x00000113  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffda950  
A10     : 0x00000000  A11     : 0x3ffda973  A12     : 0x3ffda91f  A13     : 0x00000035  
rx: b'\r\nA14     : 0x00000000  A15     : 0x3ffda924  SAR     : 0x00000004  EXCCAUSE: 0x00000000  \r\nEXCVADDR: 0x00000000  LBEG    : 0x40085d7d  LEND    : 0x40085d85  LCOUNT  : 0x00000000  \r\n\r\n\r\nBacktrace: 0x4009081c:0x3ffda9e0 0x40085e06:0x3ffdaa00 0x40019f36:0x3ffdaa20 0x4001a1f2:0x3ffdaa50 0x4005184d:0x3ffdaa70 0x40051b49:0x3ffdaa90 0x40052a59:0x3ffdaab0 0x40086f08:0x3ffdaad0 0x40019d11:0x3ffdab00 0x40055b4d:0x3ffdab20 0x4018c8d7:0x3ffdab40 0x4018cf49:0x3ffdab60\r\n\r\n\r\n\r\n\r\nELF file SHA256: c7804d24d41f9759\r\n\r\nE (2738) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0\r\nRebooting...\r\n\xd3R\xf6\xeefgR'
UART response: 
A14     : 0x00000000  A15     : 0x3ffda924  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40085d7d  LEND    : 0x40085d85  LCOUNT  : 0x00000000  

Backtrace: 0x4009081c:0x3ffda9e0 0x40085e06:0x3ffdaa00 0x40019f36:0x3ffdaa20 0x4001a1f2:0x3ffdaa50 0x4005184d:0x3ffdaa70 0x40051b49:0x3ffdaa90 0x40052a59:0x3ffdaab0 0x40086f08:0x3ffdaad0 0x40019d11:0x3ffdab00 0x40055b4d:0x3ffdab20 0x4018c8d7:0x3ffdab40 0x4018cf49:0x3ffdab60

ELF file SHA256: c7804d24d41f9759

E (2738) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

gkostov, 2025-03-15:
As pointed by [mrengineer7777](https://github.com/mrengineer7777) the message is gone after a complete erase of the device (enable the option "Erase All Flash Before Sketch Upload" just once and then disable again). Thanks for that :)