espressif/arduino-esp32 #8085: Built-in usb cdc download issue for esp32-c3

kunc0, 2023-04-17:
### Board

ESP32 C3 DEV Board

### Device Description

It's just a development board with all the pins out.

### Hardware Configuration

GPIO18&19 are connected to usb.

### Version

v2.0.7

### IDE Name

Arduino and platformIO

### Operating System

windows11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

When the com number automatically assigned by windows is too large, the esp32c3 will not be able to use the built-in usb download program. Manually set it to a smaller value (such as com2) in the windows device Manager and the download will work. It seems that the usb cdc register cannot be modified. The same problem occurs with the arduino ide and platformIO. The official esp flash download tool does not have this problem.

### Sketch

```cpp
// Appear in any sketch

uint32_t chipId = 0;
void setup() {
	Serial.begin(115200);
}

void loop() {
	for(int i=0; i<17; i=i+8) {
	  chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
	}

	Serial.printf("ESP32 Chip model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
	Serial.printf("This chip has %d cores\n", ESP.getChipCores());
  Serial.print("Chip ID: "); Serial.println(chipId);
  
	delay(3000);

}
```

### Debug Message

```plain
esptool.py v4.5.1
Serial port COM71
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C3: Wrong boot mode detected (0x8)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
上传失败: 上传错误：exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-04-17:
Hello, thanks for reporting the issue. Maybe it's related to this one https://github.com/espressif/arduino-esp32/issues/8088?

oktkas, 2023-10-06:
I have, unfortunately, been able to reproduce this issue. This post saved me from pulling the other half of my hair out. It seems to happen once you hit double digits.

ESP32-C3FH4
custom board
USB @ GPIO18,19
CDC On Boot Enabled
CPU, flash: 160, 80Mhz
Upload Speed 115200

Windows 10 22H2
Arduino IDE 2.2.1

lbernstone, 2023-10-06:
Solution posted at debian.org :)

closed 2025-01-16 by Parsaabasi