espressif/arduino-esp32 #8086: ESP-ADF Support ?

abdoutech93, 2023-04-17:
### Related area

Audio

### Hardware specification

ESP-Lyrat 4.3

### Is your feature request related to a problem?

no

### Describe the solution you'd like

To use Audio functions like esp-adf using arduino.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-04-18:
better audio support will come in Arduino v3.0.0 where the codec of LyraT will also be supported.

VojtechBartoska, 2024-02-20:
Can we close this in favor of https://github.com/espressif/arduino-esp32/pull/8714? @me-no-dev

me-no-dev, 2024-02-20:
@VojtechBartoska we will not support full ADF, but I2S and BSP will achieve about the same. So maybe after BSP is in? Though it's not very clear what functionality exactly is being requested?

closed 2025-01-16 by Parsaabasi