espressif/arduino-esp32 #8088: ESP32C3 hang after Serial CDC disconnect from specific serial monitor program

adamalfath, 2023-04-17:
### Board

ESP32-C3

### Device Description

ESP32-C3 Custom Board

### Hardware Configuration

```
/* Custom board GPIO mapping */
OUT_LED 3
OUT_BUZZER 2
OUT_LCD_DC 5
OUT_LCD_RST 4
OUT_FET0 10
OUT_FET1 0
OUT_FET2 1
USB_DN 18
USB_DP 19
UART0_TX 21 (NC)
UART0_RX 20 (NC)
I2C_CLK 9
I2C_DATA 8
SPI_CLK 6
SPI_DATA 7
```

### Version

v2.0.7

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

160Hz

### PSRAM enabled

no

### Upload speed

921600

### Description

I found that when using [Microsoft Serial Monitor extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-serial-monitor) on VScode, when its serial connection disconnected it also stops the main program. This issue only found in this software. Trying PIO miniterm, Arduino IDE Serial Monitor, YAT works normally.

### Sketch

```cpp
void setup(void) {
  Serial.begin(115200);
  pinMode(OUT_BUZZER, OUTPUT);
}

void loop(void) {
  Serial.print(F("Hello, world!\r\n"));
  digitalWrite(OUT_BUZZER, HIGH);
  delay(500);
  digitalWrite(OUT_BUZZER, LOW);
  delay(500);
}
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

1. Upload the program via built-in USB port
2. Open connection on microsoft serial monitor
3. **Hello world printed, buzzer beeped**
4. Close connection
5. **Dead**

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-04-17:
@SuGlider Please help with the triage

prazsmarid, 2023-09-18:
I have the same problem in the ESP-IDF. How you solved it?

adamalfath, 2023-09-19:
After further investigation, it might not be related to the ESP hardware nor the framework. I believe it had to do with the way vscode serial monitor DTR/RTS signaling. Upon disconnecting, it sends a signal that makes the board enter download mode, hence it looks like hang. Disabling the DTR/RTS control solved the issue for me.

[image]

Sorry I forgot to update the finding. I'll close the issue.

closed 2023-09-19 by adamalfath