espressif/arduino-esp32 #8090: ESP32 I2S issues

dacattack, 2023-04-17:
### Board

ESP32 DOIT DEVKIT V1

### Device Description

ESP32 DOIT DEVKIT V1

### Hardware Configuration

GPIO 25,32,33 are  WS,BCK,SD pins for the i2s connection with the  INMP441  MEMS microphone

### Version

v2.0.4

### IDE Name

Both Arduino IDE and PlatformIO

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have a simple code to plot the INMP441 audio input into the Serial Plotter. However I find two major issues after testing the code:
-The sample rate does not match with the sample rate I set in the i2s_config structure. In fact, is the half of the frequency i set up in the .sample_rate variable. For instance, in my case, setting up the sample rate to 44.1KHz will mean that sound frequencies above 11025 Hz won’t be captured by the microphone. If I set up to 88.2KHz then it will capture sound waves untill 22.5 KHz.
-INMP441 supports untill 24 bits of resolution. However when I change the bits per sample above 16 bits the data output has no sense(either I  get  zeros only or very high numbers among zeros.). Before you ask, I’ve increased the loop stack size to make room for the 32 bits vector(called buffer in my case) that stores the input from the microphone.
Please take a look at my code in case it’s a firmware error. Otherwise it might be the inmp441 that is malfuctioning.
The code for 16-bit or 32-bit resolution is the same except for the change from 16 to 32 in all the appropriate places.
I've tried the code in PlatformIO and the latest Arduino IDE available and I face the same issues.

### Sketch

```cpp
#include <driver/i2s.h>

#define I2S_BCK_PIN 32
#define I2S_SD_PIN 33
#define I2S_WS_PIN 25
#define I2S_PORT I2S_NUM_0

const uint8_t dma_count=8;
const uint16_t dma_len=256;

void i2s_install(){
 const i2s_config_t i2s_config={
   .mode=i2s_mode_t(I2S_MODE_MASTER | I2S_MODE_RX),
   .sample_rate=44100,
   .bits_per_sample=I2S_BITS_PER_SAMPLE_32BIT,
   .channel_format=I2S_CHANNEL_FMT_ONLY_LEFT,
   .communication_format= i2s_comm_format_t(I2S_COMM_FORMAT_STAND_I2S),
   .intr_alloc_flags=ESP_INTR_FLAG_LEVEL1,
   .dma_buf_count=dma_count,
   .dma_buf_len=dma_len,
   .use_apll=false,
 };
 
 const i2s_pin_config_t pin_config={
   .bck_io_num=I2S_BCK_PIN,
   .ws_io_num=I2S_WS_PIN,
   .data_out_num=I2S_PIN_NO_CHANGE,
   .data_in_num=I2S_SD_PIN
 };

 i2s_driver_install(I2S_PORT,&i2s_config,0,NULL);
 i2s_set_pin(I2S_PORT,&pin_config);
}

void setup(){
   i2s_install();
   Serial.begin(115200);
   Serial.printf("Inicio del programa de comunicación con el micrófono MEMS INMP441 mediante protocolo I2S.\nSe utilizará el Serial Plotter para visualizar los datos de audio.\n ");
   delay(2000);
}
void loop(){
   size_t bytes_read=0;
   int16_t samples=dma_len*dma_count;
   int32_t buffer[samples]; 
   uint16_t samples_read=0,range=1000;
   
   esp_err_t i2s_err= i2s_read(I2S_PORT,&buffer,samples*sizeof(int32_t),&bytes_read,portMAX_DELAY);
   samples_read=bytes_read/sizeof(int32_t);
   if(i2s_err==ESP_OK){
        for(uint16_t i=0; i<samples; i++){
         //Serial.print(range);
         //Serial.print(" ");
         //Serial.print(range * (-1));
         //Serial.print(" ");
         Serial.println(buffer[i]);
        }
    }
}
```

### Debug Message

```plain
There are no error messages.
```

### Other Steps to Reproduce

I've tried the code in both PlatformIO in VS CODE and in the latest Arduino IDE available. I find the same issues.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-18:
@dacattack 
The sketch here posted is based in IDF 4.4 code. 
IDF 4.4 I2S driver has part of the "issues" you describe. The Sample Rate is actually the frequency which may be divided by the number of bit of each sample and number of channels.

But this is not directly related to Arduino Project. 
You may want to post an issue in the IDF repository at
https://github.com/espressif/esp-idf/issues

Still related to IDF and I2S, this driver has changed a lot for the IDF 5.1.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

At this time we are currently working on the Arduino Core 3.0.0 that will be based on the IDF 5.1
I2S will be part of the ESP32 Arduino Library that will be refactored:
https://github.com/espressif/arduino-esp32/tree/master/libraries/I2S

dacattack, 2023-04-19:
Okay so let me get straight with a couple of things:
-If  .sample_rate=20000 and i'm using 16 bits(2 bytes) per sample and 1 channel the real sample rate would be 10KHz  but if  for example .sample_rate=40000 and i'm using 32bits(4bytes) and 2 channels the real sample rate would be 40000/(4*2)=5KHz ?????
-Lastly, I have not been able to make it work beyond or below 16 bits per sample, not 8, not 24, not 32 bits. Is this also part of the IDF 4.4 problem?"
Thanks in advance.

VojtechBartoska, 2023-08-18:
@me-no-dev Please help with triage, thank!

me-no-dev, 2023-08-22:
Is this still valid? Quite old now

VojtechBartoska, 2023-08-22:
@dacattack is this still valid?

dacattack, 2023-09-23:
It is not valid. I wasn't able to make it work. In a matter of fact its a very rare issue. 
The temporal length of the produced audio is consistent with the sampling rate in relation to the number of samples present. For example, with 32,000 samples at a 32 kHz sampling rate, when processed using a player like Audacity, the resulting file will have a duration of 1 second, as expected.
However, when you consider sound quality and the actual Nyquist frequency, you'll realize that sounds with a frequency higher than 8 kHz cannot be captured. This aligns with what you would expect from a sampling rate of 16 kHz rather than 32 kHz.

VojtechBartoska, 2023-10-04:
Closing this issue, there will be new I2S driver in 3.0.0 too.

Ayoubh4, 2025-04-16:
Esp32-S3 & INMP441 Mic

The audio file is distorted, like a distorted radio, and doesn't contain the natural sounds I recorded.

#include <Arduino.h>
#include <driver/i2s.h>
#include <SD.h>
#include <SPI.h>

#define I2S_WS     4   // LRCK
#define I2S_SD    15   // Data 
#define I2S_SCK   14   // BCLK 

#define SD_CS     10
#define SD_MOSI   11
#define SD_MISO   13
#define SD_SCK    12

#define I2S_PORT    I2S_NUM_0
#define I2S_SAMPLE_RATE 16000
#define I2S_SAMPLE_BITS 32
#define I2S_CHANNEL_NUM 1
#define RECORD_TIME 20  // seconds
#define I2S_READ_LEN 1024

File audioFile;
const char* filename = "/recording.wav";
const int headerSize = 44;
int totalSamples = I2S_SAMPLE_RATE * RECORD_TIME;

void setup() {
  Serial.begin(115200);
  delay(1000);

  SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CS)) {
    Serial.println("SD Card initialization failed!");
    while (true);
  }
  Serial.println("SD Card initialized.");

  i2sInit();

  if (SD.exists(filename)) {
    SD.remove(filename);
  }

  audioFile = SD.open(filename, FILE_WRITE);
  if (!audioFile) {
    Serial.println("Failed to open file for writing");
    return;
  }

  byte header[headerSize];
  wavHeader(header, totalSamples * I2S_CHANNEL_NUM * 2);
  audioFile.write(header, headerSize);

  Serial.println("Recording...");
  recordAudio();
  Serial.println("Recording done.");

  audioFile.seek(0);
  wavHeader(header, totalSamples * I2S_CHANNEL_NUM * 2);
  audioFile.write(header, headerSize);
  audioFile.close();

  Serial.println("File saved and closed.");

  if (validateWavHeader(filename)) {
    Serial.println("WAV header valid.");
  } else {
    Serial.println("WAV header invalid.");
  }
}

void loop() {}

void i2sInit() {
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = I2S_SAMPLE_RATE,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 8,
    .dma_buf_len = 1024,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
  };

  i2s_pin_config_t pin_config = {
    .bck_io_num = I2S_SCK,
    .ws_io_num = I2S_WS,
    .data_out_num = -1,
    .data_in_num = I2S_SD
  };

  i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
  i2s_set_pin(I2S_PORT, &pin_config);
  i2s_set_clk(I2S_PORT, I2S_SAMPLE_RATE, I2S_BITS_PER_SAMPLE_32BIT, I2S_CHANNEL_MONO);
}

void recordAudio() {
  size_t bytesRead;
  int32_t samples[I2S_READ_LEN / 4];
  int16_t monoBuffer[I2S_READ_LEN / 4];
  int recordedSamples = 0;
  int lastPercent = -1;

  while (recordedSamples < totalSamples) {
    i2s_read(I2S_PORT, &samples, sizeof(samples), &bytesRead, portMAX_DELAY);

    int sampleCount = bytesRead / 4;
    int32_t sumAbs = 0;

    for (int i = 0; i < sampleCount; i++) {
      int32_t sample24 = samples[i] >> 8;
      int16_t sample16 = sample24 >> 8;
      monoBuffer[i] = sample16;
      sumAbs += abs(sample16);
    }

    for (int i = 0; i < min(sampleCount, 10); i++) {
      Serial.printf("Sample[%d] = %d\n", i, monoBuffer[i]);
    }

    float avgAbs = sumAbs / (float)sampleCount;
    Serial.printf("Avg amplitude: %.1f\n", avgAbs);

    audioFile.write((uint8_t*)monoBuffer, sampleCount * sizeof(int16_t));
    recordedSamples += sampleCount;

    int percent = (recordedSamples * 100) / totalSamples;
    if (percent != lastPercent) {
      Serial.printf("Progress: %d%%\n", percent);
      lastPercent = percent;
    }
  }
}

bool validateWavHeader(const char* file) {
  File f = SD.open(file, FILE_READ);
  if (!f || f.size() < 44) return false;
  byte header[44];
  f.read(header, 44);
  f.close();
  return (memcmp(header, "RIFF", 4) == 0) && (memcmp(header + 8, "WAVE", 4) == 0);
}

void wavHeader(byte* header, int dataSize) {
  uint32_t fileSize = dataSize + 36;
  uint32_t byteRate = I2S_SAMPLE_RATE * I2S_CHANNEL_NUM * 2;
  uint16_t blockAlign = I2S_CHANNEL_NUM * 2;

  memcpy(header, "RIFF", 4);
  header[4] = (fileSize & 0xff);
  header[5] = (fileSize >> 8) & 0xff;
  header[6] = (fileSize >> 16) & 0xff;
  header[7] = (fileSize >> 24) & 0xff;
  memcpy(header + 8, "WAVEfmt ", 8);
  header[16] = 16;
  header[17] = 0;
  header[18] = 0;
  header[19] = 0;
  header[20] = 1;
  header[21] = 0;
  header[22] = I2S_CHANNEL_NUM;
  header[23] = 0;
  header[24] = I2S_SAMPLE_RATE & 0xff;
  header[25] = (I2S_SAMPLE_RATE >> 8) & 0xff;
  header[26] = (I2S_SAMPLE_RATE >> 16) & 0xff;
  header[27] = (I2S_SAMPLE_RATE >> 24) & 0xff;
  header[28] = byteRate & 0xff;
  header[29] = (byteRate >> 8) & 0xff;
  header[30] = (byteRate >> 16) & 0xff;
  header[31] = (byteRate >> 24) & 0xff;
  header[32] = blockAlign;
  header[33] = 0;
  header[34] = 16;
  header[35] = 0;
  memcpy(header + 36, "data", 4);
  header[40] = dataSize & 0xff;
  header[41] = (dataSize >> 8) & 0xff;
  header[42] = (dataSize >> 16) & 0xff;
  header[43] = (dataSize >> 24) & 0xff;
}