espressif/arduino-esp32 #8093: [solution] platformio upload offset

chegewara, 2023-04-18:
### Related area

upload offset

### Hardware specification

any

### Is your feature request related to a problem?

I know this is for platformio environment, and probably the place i am posting this is not the best place, but most esp32 users is reading this issue tracker anyway.
This is simple solution to change upload offset if you want to use custom `partition_table.csv` with non-standard offsets for app partitions.

### Describe the solution you'd like

add to `platformio.ini`
```
extra_scripts = 
	flash_offset.py
```
create `flash_offset.py`
```
Import("env")

env.Replace(
    ESP32_APP_OFFSET=env.get("ESP32_APP_OFFSET").replace(
        "0x10000", "0x210000")
)
```

### Additional context

If anyone has info about uploading additional bin, for example ffat image, it would be nice to share info.

Ive been trying to change `FLASH_EXTRA_IMAGES`, but that does not work, because `UPLOADERFLAGS` is using original values.

Thanks

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

chegewara, 2023-04-18:
Flashing fat partition together with firmware in one upload run is also possible. 
```
Import("env")

__UPLOADERFLAGS=env["UPLOADERFLAGS"]

# file generated with esp-idf master branch https://github.com/espressif/esp-idf/blob/master/components/fatfs/wl_fatfsgen.py
# ./wl_fatfsgen.py --partition_size PARTITION_SIZE --long_name_support path_to_data_folder
# copy fatfs_image.img to project folder
__UPLOADERFLAGS.append(str('0x410000')) # ffat partition start address
__UPLOADERFLAGS.append(str('fatfs_image.img'))

env.Replace(
   UPLOADERFLAGS =__UPLOADERFLAGS
)
```

me-no-dev, 2023-04-18:
@chegewara @pedrominatel @valeros should this go to our docs or to PIO's docs as well?

chegewara, 2023-04-18:
Honestly, no idea, its espressif decision.
I found this and i thought i can share. Maybe with arduino-esp32 v3.x it will be easier to create fatfs image, but this is not the worst case scenario. Also app partition is useful, as i saw some questions about it already.

me-no-dev, 2023-04-18:
@chegewara I agree this is useful. Let's see what PIO and our team think about where to place it.

valeros, 2023-04-18:
@chegewara  Have you tried the `board_upload.offset_address` option in your project config? Something like this:
```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
framework = arduino
board = esp32-s3-devkitc-1
board_upload.offset_address = 0x210000
```

As for an additional image, using an extra script is the right approach.

pedrominatel, 2023-04-18:
IMHO, the right place for this is [here](https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/index.html).

chegewara, 2023-04-18:
@valeros No, i didnt know that option, its also cool.

Thanks

closed 2025-01-06 by rftafas