espressif/arduino-esp32 #8104: How to create a file with SD_MMC.open("/bench.dat, FILE_WRITE) with creation date/time ?

techtoys, 2023-04-21:
### Related area

Date time for any new file with SD_MMC file system 

### Hardware specification

ESP32-PICO-D4

### Is your feature request related to a problem?

All files created with `File file = SD_MMC.open("/bench.dat", FILE_WRITE)` are missing creation date/ time under Windows Explorer. See screen capture attached.
[image]

### Describe the solution you'd like

I wish to see the creation time of file when created and appended under Windows.

### Describe alternatives you've considered

Nil

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-04-22:
Does it have a date created?

techtoys, 2023-04-24:
It has a date created but it is always of Tuesday, 1 January 1980 (see screen capture). Is there any API to set the file creation time?
[image]

serfreeman1337, 2023-04-26:
You need to set system wide clock. Either manualy using `settimeofday`: 
```c
void setup() {
  timeval now;
  now.tv_sec = 1680555600; // 04/04/2023 00:00 UTC
  settimeofday(&now, nullptr);
}
```
or [via NTP](https://github.com/espressif/arduino-esp32/blob/237a3fe96c451e5da88e8cfe9d850961f9bbe036/libraries/SD_MMC/examples/SDMMC_time/SDMMC_time.ino#L167).