espressif/arduino-esp32 #8107: USB stops working after Windows host computer sleep

asafteirobert, 2023-04-22:
### Board

LOLIN S2 mini

### Device Description

LOLIN S2 mini with ESP32-S2FH4 chip

### Hardware Configuration

LOLIN S2 mini with some buttons attached to IO pins, connected to a Windows PC through USB.

### Version

v2.0.8

### IDE Name

Arduino IDE

### Operating System

Windows 10 21H2

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Reproduction steps:
- Setup: LOLIN S2 mini connected to a Windows PC through USB
- Load the examples\USB\Keyboard\KeyboardMessage\
- Pressing the button types the message correctly.
- Put Windows to sleep using Start->Power->Sleep
- Wake up Windows
- Pressing the button no longer types the message.

It appears the whole USB ends up in a broken state as uploading to the port is no longer possible. Trying to access the debug port gives Error 31 "A device attached to the system is not functioning". Disconnecting and reconnecting the USB port (and thus also resetting the device) makes the sketch and upload port work again.

What I tried:
- USB CDC On Boot Enabled or Disabled
- Calling `USB.usbAttributes(TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP);` before `USB.begin();`
- Disabling "Allow the computer to turn off this device to save power" on the HID device.

I can confirm that USB power is not turned off.

### Sketch

```cpp
The official example USB\Keyboard\KeyboardMessage
```

### Debug Message

```plain
No output
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

No-idea-anymore, 2023-05-28:
Good day, I I was wondering if you found a solution to this problem? I'm encountering the same issue with the ESP32-S3-WROOM-1 module.

tobwen, 2023-11-27:
Did you try this?
``` c
usb_attributes(TUSB_DESC_CONFIG_ATT_SELF_POWERED | TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP)
```

source: https://forums.adafruit.com/viewtopic.php?p=937699

wareya, 2024-02-16:
Hi, I'm having this issue too and the fix in the above post doesn't seem to work. I think `TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP` is about the mouse's ability to wake up the computer, not the other way around. I'm using an ESP32-S2-DevKitM-1U.

MRezaNasirloo, 2024-04-13:
I'm having the same issue. It's not an ESP32-* issue, but rather an issue with the TinyUSB library. I tried a different project on an RP2040 board, which also uses TinyUSB and exhibits the same behavior. You should open a bug ticket for that project as well.

wareya, 2024-04-13:
I actually remade my project on an rp2040 and this issue went away, but I'm using the mbed os core, not the arduino-pico core. Does the mbed os core also use TinyUSB?

MRezaNasirloo, 2024-04-17:
I couldn't find any reference to TinyUSB in the mbed project. Do you use USB CDC in your project? Also, do you use USB power or a dedicated 5V voltage?

When the PC goes to sleep, the 5V is not disconnected, leaving the cdc usb stack in a broken state. I found out that if you use a 5V source from the PSU, the MCU will shut off on PC sleep and turn on again on wake up, It's a workaround, not a solution.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

tobwen, 2025-01-16:
Please do NOT close all those tickets as "completed"... They're stale since nobody cared.

me-no-dev, 2025-01-16:
@tobwen if this issue still exists, please open a new issue referencing this one, so we can move it up the queue and try to solve it

admire123hk, 2025-02-03:
i have the same problem with my ESP32-S3 dev board too, i program the ESP32 as USB joystick, but after windows sleep and wake up, the ESP32 joystick not working anymore, i need to reset the ESP32 in order to work

bkaes, 2025-02-06:
> Hello,
> 
> Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.
> 
> Thanks.

Hi, can you please reopen this? This issue still exists.

tobwen, 2025-02-06:
> Please do NOT close all those tickets as "completed"... They're stale since nobody cared.

Just to clarify: With my reply, I wanted to express that it's not appropriate to close all open/stale tickets as "completed," as this implies that the issue has been fixed. However, this is not the case and only makes the statistics look better. This doesn't help anyone... When someone takes the time to report bugs, the ticket should remain open until someone has processed it - OR be marked as "stale" accordingly. But closing tickets as "completed" is absolutely the wrong approach and hits the cmmunity into the face,

Thanks for opening! **Let's try to fix this annoying buy. Maybe here or maybe anywhere else upstream.**