espressif/arduino-esp32 #8108: Compile error under 2.0.8 if library function explicitly returns const __FlashStringHelper * from F() macro

avillacis, 2023-04-22:
### Board

ESP32

### Device Description

YUBOX-Node

### Hardware Configuration

No special hardware configuration needed.

### Version

v2.0.8

### IDE Name

Arduino IDE 2.1.0

### Operating System

Fedora Linux 38 x86_64

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

After updating to Arduino-ESP32 2.0.8, the attached sketch will no longer compile, and displays the following error messages:
```
In file included from /home/alex/.arduino15/packages/esp32-alternatives/2.0.8/hardware/esp32/2.0.8/cores/esp32/Arduino.h:176,
                 from /home/alex/programa/practica/arduino/esp32/test-bug-compile-fail-flashstringhelper/test-bug-compile-fail-flashstringhelper.ino:1:
/home/alex/programa/practica/arduino/esp32/test-bug-compile-fail-flashstringhelper/test-bug-compile-fail-flashstringhelper.ino: In function 'const __FlashStringHelper* testfunc()':
/home/alex/.arduino15/packages/esp32-alternatives/2.0.8/hardware/esp32/2.0.8/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
/home/alex/programa/practica/arduino/esp32/test-bug-compile-fail-flashstringhelper/test-bug-compile-fail-flashstringhelper.ino:5:10: note: in expansion of macro 'F'
   return F("test");
          ^

exit status 1

Compilation error: exit status 1
```

The same sketch compiles correctly under previous 2.0.7.

### Sketch

```cpp
#include <Arduino.h>

const __FlashStringHelper * testfunc(void)
{
  return F("test");
}

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}
```

### Debug Message

```plain
In file included from /home/alex/.arduino15/packages/esp32-alternatives/2.0.8/hardware/esp32/2.0.8/cores/esp32/Arduino.h:176,
                 from /home/alex/programa/practica/arduino/esp32/test-bug-compile-fail-flashstringhelper/test-bug-compile-fail-flashstringhelper.ino:1:
/home/alex/programa/practica/arduino/esp32/test-bug-compile-fail-flashstringhelper/test-bug-compile-fail-flashstringhelper.ino: In function 'const __FlashStringHelper* testfunc()':
/home/alex/.arduino15/packages/esp32-alternatives/2.0.8/hardware/esp32/2.0.8/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
/home/alex/programa/practica/arduino/esp32/test-bug-compile-fail-flashstringhelper/test-bug-compile-fail-flashstringhelper.ino:5:10: note: in expansion of macro 'F'
   return F("test");
          ^

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

avillacis, 2023-04-22:
Forgot to mention: although the sketch shown above does not explicitly need the F() macro, several libraries available through the Arduino IDE library manager do use this pattern internally. I was bitten by this issue when trying to compile a project using the ESPAsyncWebServer library.

avillacis, 2023-04-22:
Apparently the breaking change was introduced in https://github.com/espressif/arduino-esp32/commit/b98255d8d1be7280e767f15a8c759e13e00f4b4a .

me-no-dev, 2023-04-23:
@dok-net any comments? Please take some action or we will need to revert your PR

dok-net, 2023-04-23:
@me-no-dev Now people start calling the fix a bug... #7940 details how there is an inconsistency in this core's libraries, on the other hand. Do we want to add __FlashStringHelper wrappers to all ESP32 libs that don't use, because it is redundant on ESP32, then? It seems like a lot of work for nothing. This has just been left to rot for far to long, it seems?

dok-net, 2023-04-23:
Anyway, here's a PR #8111 to fix what originally got in my way.

tueddy, 2023-04-23:
Same problem compiling this popular RFID lib with 2.0.8:
https://github.com/miguelbalboa/rfid

dok-net, 2023-04-23:
@tueddy RFID lib (I used the example ReadUidMultiReader.ino) also fixed by #8111.

ryancasler, 2023-05-01:
Same problem here with the WS2812FX Library too.

I```
n file included from C:\Users\ryanc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.8\cores\esp32/Arduino.h:176,
                 from c:\Users\ryanc\Documents\Arduino\libraries\Adafruit_NeoPixel/Adafruit_NeoPixel.h:41,
                 from c:\Users\ryanc\Documents\Arduino\libraries\WS2812FX\src\WS2812FX.h:43,
                 from c:\Users\ryanc\Documents\Arduino\libraries\WS2812FX\src\WS2812FX.cpp:55:
c:\Users\ryanc\Documents\Arduino\libraries\WS2812FX\src\WS2812FX.cpp: In member function 'const __FlashStringHelper* WS2812FX::getModeName(uint8_t)':
C:\Users\ryanc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.8\cores\esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
c:\Users\ryanc\Documents\Arduino\libraries\WS2812FX\src\WS2812FX.cpp:404:12: note: in expansion of macro 'F'
     return F("");
            ^
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\ryanc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.8\libraries\WiFi
  Not used: C:\Users\ryanc\Documents\Arduino\libraries\WiFiEspAT
  Not used: C:\Users\ryanc\Documents\Arduino\libraries\WiFiNINA
exit status 1

Compilation error: exit status 1
```

me-no-dev, 2023-05-02:
@P-R-O-C-H-Y can you please add those libs and examples to the CI? Let's see what will fix it for them

zekageri, 2023-05-03:
So how to fix it while we are waiting for the release?

P-R-O-C-H-Y, 2023-05-03:
@zekageri You have to apply fix from PR #8147 manually until we release 2.0.9

zekageri, 2023-05-03:
Thanks, i did copy these two lines

```cpp
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
#define F(string_literal) (FPSTR(PSTR(string_literal)))
```

to my project and it fixed it. ( with a lot of prev def warning ) Thanks!

PR #8143 (2023-05-03): Revert FlashStringHelper Macros
## Description of Change
Revert to previous definition of `FPSTR` and `F` macros.

## Tests scenarios
Tested on arduino-esp32 core v2.0.8 with Adafruit Huzzah32 (Feather ESP32).
Tested on Win11 computer with VSCode+PIO.

```
#include <Arduino.h>

const __FlashStringHelper * testfunc(void)
{
  return F("test");
}

void setup() {
  Serial.begin(115200);
  
}

void loop() {
  String s;

  // put your main code here, to run repeatedly:
  delay(1000);
  s = testfunc();
  Serial.println(s);
}
```

Results:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1 [... cut]
PR #8111 (2023-05-03): Add overloads to support __FlashStringHelper like ESP8266 has them.
Fixes #8108