espressif/arduino-esp32 #8109: WiFi keeping old AP settings after disconnecting

srvmgr, 2023-04-22:
### Board

ESP32

### Device Description

ESP32 Dev Module

### Hardware Configuration

Generic

### Version

v2.0.8

### IDE Name

Arduino IDE 2

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

9216000

### Description

When WiFiMulti::run() start a new scan because the current one had no results, it call the method WiFi.disconnect() without parameters, which should delete old wifi config.
The same happens in the ESP8266WifiMulti library, but on ESP32, the default behavior of WiFi.disconnect() seems to be not deleting old AP settings, as seen on WiFiSTA.h:

ESP32, WiFiSTA.h
bool disconnect(bool wifioff = false, bool eraseap = false);

But on ESP8266, the WiFiMulti does the same, calls WiFi.disconnect() without parameters, but the default behavior is different:

ESP8266, WiFiSTA.h
bool disconnect(bool wifioff = false);

ESP8266, WiFiSTA.cpp
bool ESP8266WiFiSTAClass::disconnect(bool wifioff) {
    // Disconnect with clearing saved credentials.
    return disconnect(wifioff, true);
}

### Sketch

```cpp
// 1: Review WiFiMulti::run(), at the end when last scan had no results, it runs:

// start scan
log_d("[WIFI] delete old wifi config...");
WiFi.disconnect(); // HERE: Default behavrios on ESP32 is not to clear saved credentials

log_d("[WIFI] start scan");
// scan wifi async mode
WiFi.scanNetworks(true);

// 1: Review default behavior of WiFiSTA when disconnect() is called without parameters.
// It calls: bool disconnect(bool wifioff = false, bool eraseap = false);
// But on WiFiMulti::run() is expected to run it with eraseap = true
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-23:
This sounds like it should be solved in the wifimulti library. Changing WiFi.disconnect behavior would be a breaking change for many users, including me.

VojtechBartoska, 2023-08-18:
Closing this issue as a duplicate, please follow mentioned issue above.

closed 2023-08-18 by VojtechBartoska