espressif/arduino-esp32 #8112: Won't upload after 2.0.8 update

neo305, 2023-04-23:
### Board

ESP32-S2FNR2 (revision v0.0)

### Device Description

LilyGo-S3

### Hardware Configuration

No

### Version

v2.0.8

### IDE Name

Arduino 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

after updating to 2.0.8 I get this error and try to go back to 2.0.7 and get more errors

FQBN: esp32:esp32:esp32s3:PSRAM=enabled,FlashSize=16M,PartitionScheme=app2M_nota_spiffs13M_16MB
Using board 'esp32s3' from platform in folder: C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\hardware\esp32
Using core 'esp32' from platform in folder: C:\Users\Laz\AppData\Local\Arduino15\packages\esp32\hardware\esp32

Detecting libraries used...

recipe.preproc.macros pattern is missing

Compilation error: recipe.preproc.macros pattern is missing

### Sketch

```cpp
#include <FS.h>
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include "esp_task_wdt.h"
#include <DNSServer.h>
#include <ESPmDNS.h>
#include <Update.h>
#include <TFT_eSPI.h>
#include "hack.h"
#include "font.h"

TFT_eSPI tft = TFT_eSPI();

#if defined(CONFIG_IDF_TARGET_ESP32S2) | defined(CONFIG_IDF_TARGET_ESP32S3)  // ESP32-S2/S3 BOARDS(usb emulation)
#include "USB.h"
#include "USBMSC.h"
#include "exfathax.h"
#elif defined(CONFIG_IDF_TARGET_ESP32)  // ESP32 BOARDS
#define USBCONTROL false                // set to true if you are using usb control(wired up usb drive)
#define usbPin 4                        // set the pin you want to use for usb control
#else
#error "Selected board not supported"
#endif

// use SD Card [ true / false ]
#define USESD false  // a FAT32 formatted SD Card will be used instead of the onboard flash for the storage. \
                     // this requires a board with a sd card slot or a sd card connected.

// use FatFS not SPIFFS [ true / false ]
#define USEFAT false  // FatFS will be used instead of SPIFFS for the storage filesystem or for larger partitons on boards with more than 4mb flash. \
                      // you must select a partition scheme labeled with "FAT" or "FATFS" with this enabled.

// use LITTLEFS not SPIFFS [ true / false ]
#define USELFS true  // LITTLEFS will be used instead of SPIFFS for the storage filesystem. \
                      // you must select a partition scheme labeled with "SPIFFS" with this enabled and USEFAT must be false.

// enable internal goldhen.h [ true / false ]
#define INTHEN false // goldhen is placed in the app partition to free up space on the storage for other payloads. \
                     // with this enabled you do not upload goldhen to the board, set this to false if you wish to upload goldhen.

// enable autohen [ true / false ]
#define AUTOHEN false  // this will load goldhen instead of the normal index/payload selection page, use this if you only want hen and no other payloads. \
                       // you can update goldhen by uploading the goldhen payload to the board storage with the filename "goldhen.bin".

// enable fan threshold [ true / false ]
#define FANMOD true  // this will include a function to set the consoles fan ramp up temperature in °C \
                     // this will not work if the board is a esp32 and the usb control is disabled.

//-------------------DEFAULT SETTINGS------------------//

// use config.ini [ true / false ]
#define USECONFIG true  // this will allow you to change these settings below via the admin webpage. \
                        // if you want to permanently use the values below then set this to false.

//create access point
boolean startAP = true;
String AP_SSID = "LAZ305_LilyGo-S3";
String AP_PASS = "";
IPAddress Server_IP(10,1,1,1);
IPAddress Subnet_Mask(255,255,255,0);
IPAddress Gateway(192, 168, 1, 1);
IPAddress Ipsta(10,1,1,2);

//connect to wifi
boolean connectWifi = false;
String WIFI_SSID = "Google WiFi";
```

### Debug Message

```plain
recipe.preproc.macros pattern is missing

Compilation error: recipe.preproc.macros pattern is missing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-23:
Hmm. Does it work if you go to 2.0.6?

neo305, 2023-04-24:
No. Now when I downgrade I get same error. I have removed Arduino and did a fresh install with same result.

VojtechBartoska, 2023-04-24:
PTAL @me-no-dev

me-no-dev, 2023-04-24:
@VojtechBartoska @neo305 no idea where this error comes from at all. We do not use such recipe anywhere.

mrengineer7777, 2023-04-24:
Try running an example project.  If that works then the issue is with something in your project.

neo305, 2023-04-24:
I will but the same project I can flash on laptop which I didn’t upgrade and is on 2.0.6 and it works fine.

neo305, 2023-04-24:
Ok where do I look in Local or Documents folder?

sebastiandaneri, 2023-04-24:
I have the same problem. loading fails at a random percentage. 
Board ESP32-S3 

Device Description
T-Display-S3

Hardware Configuration
No

Version
v2.0.8

IDE Name
Arduino 2.1.0

Operating System
Windows 10

Flash frequency
80Mhz

PSRAM enabled
yes

Upload speed
921600

The error message is as follows:

Writing at 0x000fa132... (57 %)

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'El dispositivo no reconoce el comando.', None, 22)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1

Downgrade to framework 2.0.7 dont work.

mrengineer7777, 2023-04-24:
> I have the same problem. loading fails at a random percentage. Board ESP32-S3
>
> Device Description T-Display-S3
>
> Hardware Configuration No
>
> Version v2.0.8
>
> IDE Name Arduino 2.1.0
>
> Operating System [...]

That is completely different issue than `recipe.preproc.macros pattern is missing`.

me-no-dev, 2023-04-25:
Try lower upload baudrate. 460800 works good for me

mrengineer7777, 2023-04-26:
Closing because unable to reproduce.  If you find a difference with the Arduino installs that is related to our repository, then please reply back on this issue.

closed 2023-04-26 by mrengineer7777