espressif/arduino-esp32 #8113: Arduino OTA Update Cause Crash

GodayMax, 2023-04-24:
### Board

ESP32 Dev Module

### Device Description

ESP32-WROOM-32

### Hardware Configuration

N/A

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

macOS12

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

GPIO2 -> LED
GPIO33 -> ADC IN

### Sketch

```cpp
#include <esp32fota.h>
#include <Arduino.h>

esp32FOTA esp32FOTA(PRODUCT_ID, FIRMWARE_VERSION, false, true);

void setup() {
    String ota_url = "https://mz-iot.oss-us-west-1.aliyuncs.com/IoT/MZLFC-1.1.bin";
    Serial.printf("[OTA]start to exec ota from %s \n", ota_url.c_str());
    esp32FOTA.forceUpdate(ota_url, false);
}

void loop() {}

===============================================

/*
   esp32 firmware OTA
   Date: December 2018
   Author: Chris Joyce <https://github.com/chrisjoyce911/esp32FOTA/esp32FOTA>
   Purpose: Perform an OTA update from a bin located on a webserver (HTTP Only)

   Date: 2021-12-21
   Author: Moritz Meintker <https://thinksilicon.de>
   Remarks: Re-written/removed a bunch of functions around HTTPS. The library is
            now URL-agnostic. This means if you provide an https://-URL it will
            use the root_ca.pem (needs to be provided via SPIFFS) to verify the
[... 496 lines cut ...]

~/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-addr2line -pfiaC -e /Users/max/Documents/DEV/codes/IdeaProjects/iot/MZ-LIGHT-MESH-GATEWAY/.pio/build/esp32dev/firmware.elf 0x40090524:0x3ffd0c20 0x400907a1:0x3ffd0c40 0x401fe8dd:0x3ffd0c60 0x4008f686:0x3ffd0c80 0x401fe779:0x3ffd0ca0 0x400f0755:0x3ffd0cc0 0x400e9ff8:0x3ffd0ce0 0x400ea257:0x3ffd0d00 0x400d338a:0x3ffd0d20 0x400d378d:0x3ffd10d0 0x400dd613:0x3ffd10f0 0x400de3ee:0x3ffd1170 0x400f404e:0x3ffd11b0 0x40091816:0x3ffd11d0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13512
ho 0 tail 12 room 4
load:0x40080400,len:3604
entry 0x400805f0
```

### Other Steps to Reproduce

[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-24:
v1.0.6 is no longer supported. Use the exception decoder option in pio to see where it crashed.

GodayMax, 2023-04-24:
[image]

GodayMax, 2023-04-24:
Seems like partition table is missing

SuGlider, 2023-04-24:
I see that the frimware that is binary that this sketch is trying to upload has 1,840,736 bytes.
I think that it may not fit in the OTA Partition.

What is the Partition Table used for this application?

SuGlider, 2023-04-24:
This is a typical OTA Partition File for a ESP32 with 4MB Flash:

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000,  1M,
ota_0,    app,  ota_0,   0x110000, 1M,
ota_1,    app,  ota_1,   0x210000, 1M,
```

As we can see, the maximum size of the supported OTA firmware is 1MB only.

GodayMax, 2023-04-24:
It does fit, we use a partition table like this

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1F0000,
app1,     app,  ota_1,   0x200000,0x1F0000,
spiffs,   data, spiffs,  0x3F0000,0x10000,

GodayMax, 2023-04-24:
almost 5K+ modules updates through ota sucessfully, but several modules occur this issue

GodayMax, 2023-04-24:
this is where the error occurs

[image]

SuGlider, 2023-04-24:
What IDF version are you using?

SuGlider, 2023-04-24:
> almost 5K+ modules updates through ota sucessfully, but several modules occur this issue

It doesn't seem to be an Arduino issue... 
Could it be related to the Flash chip of the failing modules?

SuGlider, 2023-04-24:
This code seems to be from IDF 4.3.

> this is where the error occurs
> 
> [image]

SuGlider, 2023-04-24:
Is it possible that this applications uses IDF 4.3 + Arduino Core 1.0.6?

Is so, this is not standard.... this seems to be not an Arduino issue.

GodayMax, 2023-04-24:
> 

> Is it possible that this applications uses IDF 4.3 + Arduino Core 1.0.6?
> 
> If so, this is not standard.... this seems to be not an Arduino issue.

IDF version is 3.5, arduino version is 1.0.6
[image]

GodayMax, 2023-04-24:
> > almost 5K+ modules updates through ota sucessfully, but several modules occur this issue
> 
> It doesn't seem to be an Arduino issue... Could it be related to the Flash chip of the failing modules?

it seems that the chip itself has nothing to do with this issue
the specific flash chip that occured to  this issue, 
after i flash with pytool from espressif using USB cable, the OTA feature functions like a charm....

mrengineer7777, 2023-04-24:
It's possible the partition info got corrupted on that module.  To prove that you'd have to have a bad module.  Do not reprogram!  Read the entire flash image (using esptool.py) and compare to a good module.  There's a section early on in the flash that describes the partition structure.

SuGlider, 2023-04-24:
The first point in the stack decoding is Line 212 from `nvs_pagemanager.cpp`

[image]

Could the flash of the OTA space (partition) be way too worn?

mrengineer7777, 2023-04-24:
For reference: partitions.bin is written to 0x8000, at least for arduino 2.x/IDF 4.x

GodayMax, 2023-04-24:
> 

> > > It doesn't seem to be an Arduino issue... Could it be related to the Flash chip of the failing modules?
> > 
> > 
> > it seems that the chip itself has nothing to do with this issue the specific flash chip that occured to this issue, after i flash with pytool from espressif using USB cable, the OTA feature functions like a charm....
> 
> When using pytool, it will fash at the main APP_0 partition (OTA_0) at 0x10000 But the OTA Update code will flash it in another region (APP_1 / OTA_1) at 0x200000
> 
> Those are 2 different Flash regions.
> 
> nvs, data, nvs, 0x9000, 0x5000, otadata, data, ota, 0xe000, 0x2000, app0, app, ota_0, 0x10000, 0x1F0000, app1, app, ota_1, 0x200000,0x1F0000, spiffs, data, spiffs, 0x3F0000,0x10000,

It is nearly a brand new module, I tried two times, the first time I found this issue that the OTA Update not work, after I flashed a new firmware using esptool, I tried OTA Update again and it turned out the function worked fine, so as I consider it has very little chance that「 OTA space (partition) be way too worn」

GodayMax, 2023-04-24:
> It's possible the partition info got corrupted on that module. To prove that you'd have to have a bad module. Do not reprogram! Read the entire flash image (using esptool.py) and compare to a good module. There's a section early on in the flash that describes the partition structure.

If 「partition info got corrupted」 , why the module started successfully?  without the partition info ,  how can it finds the right address of the app0/app1?

SuGlider, 2023-04-24:
@GodayMax - we need more information about the execution of this sketch.

Please build it enabling Logging and post here the output.
`build_flags = -DCORE_DEBUG_LEVEL=5` shall go in the `platformio.ini` file.

Jason2866, 2023-04-24:
Flash Chip support (vendor) is maybe a problem. Flash chips from "new" vendors are not native supported in older IDF versions. This can end sometimes! in flash write errors.
Can you see a dependency between failing modules and flash vendor?

SuGlider, 2023-04-24:
> Flash Chip support (vendor) is maybe a problem. Flash chips from "new" vendors are not native supported in older IDF versions. This can end sometimes! in flash write errors. Can you see a dependency between failing modules and flash vendor? My personal experience. Had devices with defect flash from PUYA, XMC and Boya and most of my devices has Flash chips from Winbond. Until today no Winbond chip died...

I was thinking about the same. I know that there is an issue with XMC that has been solved in the IDF 4.4.4
I think that @GodayMax should try to update the Arduino Core and IDF version to the latest 2.0.7, at least.

Could you please guide us on how to do that, @Jason2866 ?
Thanks!

SuGlider, 2023-04-24:
@GodayMax - 
I just tested the sketch using an ESP32 DevKit, Arduino IDE with Arduino Core 2.0.8 (which uses IDF 4.4.4). 
It worked perfectly fine.

The Sketch I used:
``` cpp
#include <esp32fota.h>
#include <Arduino.h>

int FIRMWARE_VERSION = 1;
esp32FOTA esp32FOTA("PRODUCT_ID", FIRMWARE_VERSION, false, true);
static const char *ssid     = "MySSID";  // your network SSID (name of wifi network)
static const char *password = "MyAP_Pwd"; // your network password

void setup() {
    Serial.begin(115200);
    Serial.print("Attempting to connect to SSID: ");
    WiFi.begin(ssid, password);

    // attempt to connect to Wifi network:
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(1000);
    }

    Serial.print("Connected to ");
    Serial.println(ssid);
    
   /* String */ const char *ota_url = "https://mz-iot.oss-us-west-1.aliyuncs.com/IoT/MZLFC-1.1.bin";
    Serial.printf("[OTA]start to exec ota from %s \n", ota_url/*.c_str()*/);
    esp32FOTA.forceUpdate(ota_url, false);
}

void loop() {}
```

The Output of the execution with logging messages:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[     1][D][esp32FOTA.cpp:141] setString(): Assigned value: PRODUCT_ID <= PRODUCT_ID
[     2][I][esp32FOTA.cpp:901] debugSemVer(): Current firmware version: 1.0.0
E (333) esp_⸮⸮⸮⸮e_dump_flash: No core dump partition found!
E (338) esp_core_dump_flash: No core dump partition found!
[    17][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   470][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Attempting to connect to SSID: [   503][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   590][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   589][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   596][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
....[  3645][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: MySSID, BSSID: 90:f6:52:6d:14:54, Reason: 2
[  3646][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  3653][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  3660][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
[  3668][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  3719][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: MySSID, BSSID: 90:f6:52:6d:14:54, Channel: 13, Auth: WPA2_PSK
[  3720][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  3734][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.1.100
[  3735][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  3742][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.100, MASK: 255.255.255.0, GW: 192.168.1.1
Connected to MySSID
[... 99 lines cut ...]
[N]start to scan wifi....
[N]done of scan wifi....
MySSID, -58, 11, 4 
CLARO_2GC356FF, -60, 11, 3 
LIVIA, -61, 6, 3 
CLARO_2G80C97D, -82, 6, 4 
[N]Mesh Key Not Find, Using Default MeshKey 
[N]ESPNOW connected at channel 6 
[BLE]BLE NAME IS MZ_MZLFC_0_78E36DCD0404 
[I][BLEDevice.cpp:569] getAdvertising(): create advertising
ble inited, waiting for connection.....
[Core0] start to execute espnow channel sync job
[N]Save Config.. {"MESHED":0,"PROVISONED":0,"PRODUCT_ID":"MZLFC","LAST_RSSID":"","RSSID":"","RPASS":"","WIFICHANNEL":-1,"GWCHANNEL":-1,"LIGHT_MODE":"adc","ACCOUNT":"0","ACTIVE_SCENE":"","ONOFF":1,"BRIGHT":28.57507376,"LAST_ADC_BRIGHT":28.57507376,"NOW":1609459205}
null
MySSID, -58, 11, 4 
CLARO_2GC356FF, -60, 11, 3 
LIVIA, -61, 6, 3 
CLARO_2G80C97D, -82, 6, 4 
[CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
[CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
[N]Save Config.. {"MESHED":0,"PROVISONED":0,"PRODUCT_ID":"MZLFC","LAST_RSSID":"","RSSID":"","RPASS":"","WIFICHANNEL":-1,"GWCHANNEL":-1,"LIGHT_MODE":"adc","ACCOUNT":"0","ACTIVE_SCENE":"","ONOFF":0,"BRIGHT":0,"LAST_ADC_BRIGHT":0,"NOW":1609459234}
null
[CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
[CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
```

GodayMax, 2023-04-25:
> @GodayMax - we need more information about the execution of this sketch.
> 
> Please build it enabling Logging and post here the output. `build_flags = -DCORE_DEBUG_LEVEL=5` shall go in the `platformio.ini` file.

@mrengineer7777 

It's gonna take sometime,  basiclly we can only recover failed module from our user, we'll look into it.  Thank you guys~ I'll let you know when more infomation comes up~

GodayMax, 2023-04-25:
> @GodayMax - I just tested the sketch using an ESP32 DevKit, Arduino IDE with Arduino Core 2.0.8 (which uses IDF 4.4.4). It worked perfectly fine.
> 
> The Sketch I used:
> 
> ```c++
> #include <esp32fota.h>
> #include <Arduino.h>
> 
> int FIRMWARE_VERSION = 1;
> esp32FOTA esp32FOTA("PRODUCT_ID", FIRMWARE_VERSION, false, true);
> static const char *ssid     = "MySSID";  // your network SSID (name of wifi network)
> static const char *password = "MyAP_Pwd"; // your network password
> 
> 
> 
> void setup() {
>     Serial.begin(115200);
>     Serial.print("Attempting to connect to SSID: ");
>     WiFi.begin(ssid, password);
> 
>     // attempt to connect to Wifi network:
>     while (WiFi.status() != WL_CONNECTED) {
>         Serial.print(".");
>         delay(1000);
>     }
> 
>     Serial.print("Connected to ");
>     Serial.println(ssid);
>     
>    /* String */ const char *ota_url = "https://mz-iot.oss-us-west-1.aliyuncs.com/IoT/MZLFC-1.1.bin";
>     Serial.printf("[OTA]start to exec ota from %s \n", ota_url/*.c_str()*/);
>     esp32FOTA.forceUpdate(ota_url, false);
> }
> 
> void loop() {}
> ```
> 
> The Output of the execution with logging messages:
> 
> ```
> ets Jul 29 2019 12:21:46
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1344
> load:0x40078000,len:13924
> ho 0 tail 12 room 4
> load:0x40080400,len:3600
> entry 0x400805f0
> [     1][D][esp32FOTA.cpp:141] setString(): Assigned value: PRODUCT_ID <= PRODUCT_ID
> [     2][I][esp32FOTA.cpp:901] debugSemVer(): Current firmware version: 1.0.0
> E (333) esp_⸮⸮⸮⸮e_dump_flash: No core dump partition found!
> E (338) esp_core_dump_flash: No core dump partition found!
> [    17][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> [   470][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
> Attempting to connect to SSID: [   503][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [   590][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
> [   589][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
> [   596][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
> ....[  3645][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: MySSID, BSSID: 90:f6:52:6d:14:54, Reason: 2
> [  3646][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
> [  3653][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
> [  3660][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
> [  3668][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
> [  3719][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: MySSID, BSSID: 90:f6:52:6d:14:54, Channel: 13, Auth: WPA2_PSK
> [  3720][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  3734][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.1.100
> [  3735][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[... 105 lines cut ...]
> LIVIA, -61, 6, 3 
> CLARO_2G80C97D, -82, 6, 4 
> [N]Mesh Key Not Find, Using Default MeshKey 
> [N]ESPNOW connected at channel 6 
> [BLE]BLE NAME IS MZ_MZLFC_0_78E36DCD0404 
> [I][BLEDevice.cpp:569] getAdvertising(): create advertising
> ble inited, waiting for connection.....
> [Core0] start to execute espnow channel sync job
> [N]Save Config.. {"MESHED":0,"PROVISONED":0,"PRODUCT_ID":"MZLFC","LAST_RSSID":"","RSSID":"","RPASS":"","WIFICHANNEL":-1,"GWCHANNEL":-1,"LIGHT_MODE":"adc","ACCOUNT":"0","ACTIVE_SCENE":"","ONOFF":1,"BRIGHT":28.57507376,"LAST_ADC_BRIGHT":28.57507376,"NOW":1609459205}
> null
> MySSID, -58, 11, 4 
> CLARO_2GC356FF, -60, 11, 3 
> LIVIA, -61, 6, 3 
> CLARO_2G80C97D, -82, 6, 4 
> [CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
> [CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
> [N]Save Config.. {"MESHED":0,"PROVISONED":0,"PRODUCT_ID":"MZLFC","LAST_RSSID":"","RSSID":"","RPASS":"","WIFICHANNEL":-1,"GWCHANNEL":-1,"LIGHT_MODE":"adc","ACCOUNT":"0","ACTIVE_SCENE":"","ONOFF":0,"BRIGHT":0,"LAST_ADC_BRIGHT":0,"NOW":1609459234}
> null
> [CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
> [CKR]CHECKER RESETTED ~~~~~~~~~~~~~~ 1 
> ```

@SuGlider 

in most cases it works just fine... that is the real tricky problem.  We'll upgrade IDF and Arduino version and see whether the situation changes.

Jason2866, 2023-04-25:
I fear that "just" OTA updating firmware compiled with core 2.0.8 is not enough.
Which flash code routines are called from OTA update? Are the included with the new OTA firmware?

If **yes** the would be called on second OTA update. Fine. Issue fixed.
If **no** the called flash code routines (to write the OTA firmware) are in the bootloader code, compiling the firmware with core 2.0.8 does change nothing here.

In the "no" case it gets complicated. The bootloader code needs to be replaced. This is impossible "by hand" with 5k devices in the field. It needs to be done via OTA. There is no ready made solution for :-(
Good news it is possible :-) 

If you want to do it with Arduino, you have to compile Arduino yourself, since you have to set the sdkconfig option
`CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED=y`  Now you can write (with your firmware) to every flash address you want. Your firmware can check the current bootloader and fetch it form somewhere or ask user to upload a new bootloader and flash it to address 0x1000 (bootloader address for ESP32).
**The bootloader has to match with settings for flash mode / speed. Using a not matching one, ends in bricked device!!**
Sounds complicated? It is. We have the option to do this in Project Tasmota. The written code is NOT reusable since we have implemented with a REPL interpreter (Berry) and wrote this [part](https://github.com/arendst/Tasmota/pull/16362) with Berry. Anyways it shows the way it works.

Our custom Arduino espressif-framework32 (where `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED=y` is set and other changes...) can be used with Platformio with the setting:
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.04.02/platform-espressif32.zip
```

SuGlider, 2023-04-25:
@GodayMax - I have discussed your case here internally.

I have a few questions:
1- There are already deployed units in the field? Are those unit using the Arduino 1.0.6 + IDF 3.3 and its related bootloader?
2- The issue with flashing is happening in the field (with remote units)?

The questions are related to a potential issue with XMC flash chips.
Given that it happen with some units and other units have no issue, it sounds like it may be related to the flash chip of this affected units.

SuGlider, 2023-04-25:
There is a discussion about the XMC failure here:
https://github.com/espressif/esp-idf/issues/7994#issuecomment-1157943173

A potential fix for remote units (with associated risk) is presented here (please read all the content):
https://github.com/espressif/idf-flash-vendor-patches/tree/main/examples/lock_sr

The future firmware fix demands that the application uses IDF 4.4.4 and Arduino Core 2.0.7+
We recommend that the product always uses the latest Arduino and IDF versions.
In order to fix the issue with the already deployed units, it is necessary to fix both, bootloader and application firmware, as @Jason2866 mentioned.

The SR fix solves the issue permanently with no need to replace the firmware.
The alternative solution presented by @Jason2866 also fixes it by a few steps with OTA, including fixing the bottloader and the application firmware. This may a more complex procedure.

We recommend to analyse the possible alternatives and simulate whatever solution in a controlled environment to verify its effectiveness.

closed 2023-05-24 by SuGlider