espressif/arduino-esp32 #8116: Esp32 18650 : not waking up from deepsleep on battery, except if...

MyNameIsRenaud, 2023-04-24:
### Board

esp wroom 32 battery 18650

### Device Description

esp32 battery 18650 board with: 
- 1 potentiometer
- 1 as5600 board (I2C, powered from the vcc of the esp32 board)

### Hardware Configuration

as560 connected on gpio 21 and 22 (I2C)

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

windows

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

Hello, 

So, i use a esp32 1860 battery board (all in one board of this type : https://wiki.geekworm.com/WEMOS_ESP32_Board_with_18650_Battery_Holder). 
[image]
It's used as a remote control to send data to another esp32 board using esp-now. 
On the board is attached a potentiometer and an AS5600 magnetic encoder (used to make a throttle).

So when i use the throttle (ie the as5600) the esp is active and it send the data.
When i release the throttle for more than 3 second, the esp goes to deep sleep.
a touch interrupt on T3 is coded.

So everything works fine when i use the system connected to my computer with usb cable, the code works, it sends the data, wake up with the touch interrupt etc. Perfect !

1. BUT it doesn't wake up from deesleep when it's powered on the battery (fully charged). I don't think (but i'm a noob) it's a battery autopower problem because there is still 3.3v on the VCC pin when in deepsleep for a long time. 
I read some threads here and there about this issue (https://github.com/espressif/esp-idf/issues/6694), and some said you have to attached a RC delay circuit on reset. I tried, but i'm not sure i did it well... i'm confuse on the pin i need to connect. 

2. BUT (n°2) there is a weird behavior, as i tried to turn my esp board to look at some of the connections, it came out of deepsleep. Why ? -> I tested it several time and in different way, if i touch T3, and that one of my fingers touch any GPIO pin (unused), the wake up works. I have no clue about what is the skin contact doing in the circuit, but I would be happy to discover it as it make the board wake up...

If someone can help me, either to show me how to properly make the RC circuit or can explain the skin contact effect, i take it :-)

(if someone wonder, I use this remote throttle thing to control a hacked  hoverboard to make an electric golf trolley :-)  )

### Sketch

```cpp
-
```

### Debug Message

```plain
no error message in deepsleep
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-04-26:
What are you using as a reference for touch sense?

MyNameIsRenaud, 2023-04-26:
If you talk about the threshold, it's set to 15

mrengineer7777, 2023-04-26:
I haven't used touch sensors yet, but as I understand it they need a touch pad and a common ground reference.  Perhaps by touching other pins you are proving that reference.

MyNameIsRenaud, 2023-04-26:
In the different examples i looked at, you just need a wire that you touch, no ground reference to add. What is weird is that it works perfectly when using USB power, and when on battery, it doesn't work properly.

mrengineer7777, 2023-04-26:
The USB will tie the PCB ground to computer ground, and then computer ground ties to earth ground.  Does your battery powered version work if you touch your other hand to a ground point on your PCB?

MyNameIsRenaud, 2023-04-26:
Yes totally, so it means it has to be grounded in some way? But how could I do that ? Adding a resistor from the touch pin (i just tried another one just in case but it doesn't change anything) result in a active pin all the time (I just tried it)

mrengineer7777, 2023-04-26:
Will the user be holding the device in their hand?  Add a chassis gnd.
Will the touch sensor be on a panel?  Add a nearby metal plate the user rests their hand against while pressing the touch sensor.

MyNameIsRenaud, 2023-04-26:
well it's enclosed in a 3d printed case, but i think i can manage to solve it this way.
I would be interested to understand why it does that, and if there is a way to solve the problem "on the baord" rather to tinker, but most important is that it works :-). Thanks a lot for your help

closed 2023-04-26 by mrengineer7777