espressif/arduino-esp32 #8118: How to set coredump file output ways? assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

davidbcf, 2023-04-25:
### Board

ESP32-C3-S2

### Device Description

ESPC3-S2

### Hardware Configuration

nothing

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

win 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

961200

### Description

how to set coredump message print  to serial ? or how can i get coredump file from flash ? 
Enviroment: Win10  + arduino
without coredump info ,how can i use debug message to find bug in my program,for examlpe.
assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

### Sketch

```cpp
NO
```

### Debug Message

```plain
assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)
Core  0 register dump:
MEPC    : 0x403825a4  RA      : 0x40389ef4  SP      : 0x3fca2820  GP      : 0x3fc91400  
TP      : 0x3fc5afa8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fca299a  S1      : 0x0000007d  A0      : 0x3fca2884  A1      : 0x3fc94129  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9d000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca2878  S3      : 0x00000001  
S4      : 0x3fca2878  S5      : 0x4038aea4  S6      : 0x00000000  S7      : 0x003849ba  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fca2820: 0xa5a5a5a5 0xa5a5a5a5 0x3c16d7f4 0x4038f7ee 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3fc94128
3fca2840: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x39343531 0xa5a5a500 0x3fc94a08 0x3c16d7f4 0x3fc94cc4
3fca2860: 0x3c16d891 0x3fc94a18 0x3fca284c 0x3fc94a1c 0x3c16dbd8 0x3fc94128 0x00000000 0x00000000
3fca2880: 0xa5000000 0x65737361 0x66207472 0x656c6961 0x78203a64 0x75657551 0x6d655365 0x6f687061
3fca28a0: 0x61546572 0x7120656b 0x65756575 0x313a632e 0x20393435 0x51787028 0x65756575 0x78753e2d
3fca28c0: 0x6d657449 0x657a6953 0x203d3d20 0xa5002930 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca28e0: 0x4038cc52 0x4038a990 0x3fca2960 0x3fc91400 0x3fc5afa8 0x00000000 0x4038a71a 0x00000000
3fca2900: 0x3fcadbd8 0x00000000 0x00000001 0x00000001 0x00000001 0x00000001 0x3fc9d17c 0x600c2000
3fca2920: 0x00000000 0xfff00000 0x3fca29a8 0x00000000 0x00000000 0x3fcadbe8 0x00000002 0x4038c38a
3fca2940: 0x00000000 0x00000000 0x00000000 0x993467c6 0x4038cc52 0x3fc9d000 0x00ffffff 0xf0de1e1e
3fca2960: 0x3fc9d000 0x3fc9d000 0x3fcb53d8 0x4038aea4 0x3fc9d198 0x00000001 0x00000001 0xffffffff
3fca2980: 0x00000001 0x00000001 0x3fc9d17c 0x600c2000 0x00000000 0xfff00000 0x00000000 0xf0de1e1e
3fca29a0: 0x3fc9d000 0x3fc9d000 0x3fcb46e0 0x420c0f7a 0x00000000 0x00000000 0x00000000 0x00000000
3fca29c0: 0x00000000 0x00000000 0x00000000 0x4038c5d0 0x00000000 0x3fc9d000 0x00ffffff 0xf0de1e1e
3fca29e0: 0x00000000 0x420c0f54 0x3fcb46e0 0x4205abfc 0x00000000 0x00000000 0x00000000 0x00000000
3fca2a00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2a20: 0x00000000 0x00000000 0x00000000 0x4038cbe4 0x00000000 0x00000000 0x00000000 0x00000000
3fca2a40: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234
3fca2a60: 0x00000154 0x3fca2950 0xf1fba55e 0x3fc97380 0x3fc97380 0x3fca2a64 0x3fc97378 0x00000003
3fca2a80: 0x6892c0c1 0x1d2da622 0x3fca2a64 0x00000000 0x00000016 0x3fca1854 0x5f707365 0x656d6974
3fca2aa0: 0x839f0072 0x006bcd84 0x00000000 0x3fca2a50 0x00000016 0x00000000 0x00000000 0x00000000
3fca2ac0: 0x00000000 0x3fca0fc4 0x3fca102c 0x3fca1094 0x00000000 0x00000000 0x00000001 0x00000000
3fca2ae0: 0x00000000 0x00000000 0x42132516 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2b00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2b20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2b40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2b60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2b80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca2ba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x9a000000 0xbaad5678 0x00000014
3fca2bc0: 0xabba1234 0x00000008 0x3fca2be0 0x00000000 0xbaad5678 0x00000018 0xabba1234 0x0000000c
3fca2be0: 0x09c4000a 0x00000000 0x3fca2c14 0xbaad5678 0x00000014 0xabba1234 0x00000008 0x3fca4ce0
3fca2c00: 0x00000000 0xbaad5678 0x0000001c 0xabba1234 0x0000000c 0x0c86000a 0x00000000 0x3fca4ca4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-26:
Please check the information in the GitHub about the tool used for using Coredumps.

https://github.com/espressif/esp-coredump

SuGlider, 2023-04-26:
Documentation can be found in
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html

davidbcf, 2023-04-27:
My IDE is **Arduino**,not vscode +ESP-IDF ，that means i cant analyse coredump ？

SuGlider, 2023-04-27:
From the information about this issue, I see that you have Win10 and Arduino IDE.
There is a tool that can help you:

https://github.com/me-no-dev/EspExceptionDecoder

Installing it will add a Menu option "Arduino IDE Menu -> Tools -> ESP Exception Decoder".
At this time it only works for the IDE 1.8.x

Although this document talks about ESP8266, its concepts also applies to the ESP32 regarding decoding the stack:
https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html

davidbcf, 2023-05-05:
thanks  a lot!

closed 2023-05-05 by davidbcf