espressif/arduino-esp32 #8121: Arduino 2.10 fails to compile

frspp, 2023-04-25:
### Board

Nodemcu esp32

### Device Description

Oled display 
Dallas DS18B20 temperature sensor
TI CC1101 433mhz radio

### Hardware Configuration

-

### Version

other

### IDE Name

Arduino IDE 2.1

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

9600

### Description

GDB executable "C:\Users\CUT\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823\bin\arm-none-eabi-gdb.exe" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.

Did not get this error week or two ago with older Arduino IDE (2.0.5?) and older ESP board information.

### Sketch

```cpp
-
```

### Debug Message

```plain
GDB executable "C:\Users\vy14ju\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\11.2_20220823\bin\arm-none-eabi-gdb.exe" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

frspp, 2023-04-25:
dunno it went away. Did I press "debug" button instead "compile"?

closed 2023-04-25 by frspp

me-no-dev, 2023-04-25:
> dunno it went away. Did I press "debug" button instead "compile"?

yes you did :)