espressif/arduino-esp32 #8123: digitalWrite not working in library

datphys, 2023-04-26:
### Board

Esp32 dev module

### Device Description

Cusom board

### Hardware Configuration

pin 27 control power key of modules sim

### Version

v2.0.8

### IDE Name

Arduino IDE 2.0

### Operating System

Windown 10

### Flash frequency

240 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Module reset frequenly

### Sketch

```cpp
//in sketch file
#include <LYNQ_L506.h>
const int pwrkey = 27;
l506_init(&Serial1, pwrkey);
l506_power_on_off();     // ----> reset here
// Replace with
digitalWrite(pwrkey, HIGH);
delay(2500);
digitalWrite(pwrkey, LOW); // ----> Working

# header file
#include <Arduino.h>
void l506_init(HardwareSerial *serial, int pin);
int8_t l506_power_on_off();

# cpp file
void l506_init(HardwareSerial *serial, int pin)
HardwareSerial *l506;
int PWRKEY;

{
    l506 = serial;
    PWRKEY = pin;
    pinMode(PWRKEY, OUTPUT);
    digitalWrite(PWRKEY, LOW);
}

int8_t l506_power_on_off()
{
    digitalWrite(PWRKEY, HIGH);
    delay(2500);
    digitalWrite(PWRKEY, LOW);
}
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-26:
@datphys - 

This has nothig to do with `digitalWrite()`.
The issue is about declaring a function that shall return a value, but nothing is returned.

You can fix it by:
- making the function `l506_power_on_off()` void like: `void l506_power_on_off()`
- returning a value, for instance zero, at the end of the function `l506_power_on_off()`

SuGlider, 2023-04-26:
You should see a warning message in the compilation output:

```
LYNQ_L506.cpp: In function 'int8_t l506_power_on_off()':
LYNQ_L506.cpp:18:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
```

SuGlider, 2023-04-26:
You can enforce such warning as an error by setting `Compiler warnings:` option to `More` or `All`

[image]

mrengineer7777, 2023-04-26:
Solution noted above

closed 2023-04-26 by mrengineer7777