espressif/arduino-esp32 #8128: possible issue in mbedtls ECDH

anishsheikh, 2023-04-26:
### Board

ESP32-C3-DevKitM-1

### Device Description

official Espressif Board

### Hardware Configuration

Nothing Connected , just bare devboard

### Version

v2.0.8

### IDE Name

Arduino IDE

### Operating System

Windows 11 22h2

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I first wrote mbedtls code in C and tested with corresponding dart code which reproduces the same shared secret. 
In Arduino IDE the main compilation happens on provided gcc compiler so , the code actually works just fine with mbedtls_mpi_write_binary(),
mbedtls_mpi_read_binary(),
It just doesn't produce the intended shared secret .
in RHEL 8 , gcc 8.5.0, i used mbedtls_mpi_write_binary_le(),
mbedtls_mpi_read_binary_le(),
which produced actual intended shared key in both Dart and C program, However in specific case of arduino ide , if we use the

mbedtls_mpi_write_binary_le(),
mbedtls_mpi_read_binary_le(),

both calculating functions produce error which are 
mbedtls_ecdh_calc_secret() and mbedtls_ecdh_compute_shared(),

### Sketch

```cpp
*/Arduino Side Code/*

#include <mbedtls/ecdh.h>
#include <mbedtls/entropy.h>
#include <mbedtls/ctr_drbg.h>
#include "mbedtls/error.h"

  unsigned char my_pubkey[32] = { 0 };
  unsigned char my_privkey[32] = { 0 };
  unsigned char server_pubkey[32] = {0};
  unsigned char server_privkey[32] = {0};
  unsigned char shared_secret[32] = { 0 };
  unsigned char shared_secret2[32] = { 0 };

void printhex(unsigned char* array, int size) {
[... 229 lines cut ...]
load:0x403ce710,len:0x25b0
entry 0x403cc710
PUBKEY: d1cf1e2c0eaea894cd556df1a20ad3bbbb95818b57bce5a90f3786abbc67246c
Privkey: 70625c0deabc8b43fe8cb15e7fd7b7178a28e3e6ceb76f7a5719f32f48bd2944
Server PUBKEY: 7983b2d2c9201f1ac0cfb328d093b25017c706865f0738945ba6c8857eb6893c
Server Privkey: d03a05f065300175edda18d31cebcbe96ee8fcf53bb084284af9f41ff8536364
error in calculating secret
Error message for error code -19584: ECP - Invalid private or public key
Secret buffer :0000000000000000000000000000000000000000000000000000000000000000
Secret :0000000000000000000000000000000000000000000000000000000000000000
Secret :0000000000000000000000000000000000000000000000000000000000000000
error in calculating secret
Error message for error code -19584: ECP - Invalid private or public key
Secret buffer :0000000000000000000000000000000000000000000000000000000000000000
Secret :0000000000000000000000000000000000000000000000000000000000000000
Server Secret :0000000000000000000000000000000000000000000000000000000000000000
```

### Other Steps to Reproduce

it works just fine in x86_64 so i guess it is not a mbedtls specific problem, most probably the problem corresponds to compiler specific endianness issue or linking problems so i opened an issue here, any specific help would be great, the code maybe messy but logic is understandable. can you please help me out?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

anishsheikh, 2023-04-27:
[image]
[image]
[image]

as you can see a bit factored code is working both on dart on aarch64 , c on x86_64 and vice versa , so the problem is esp related.

mrengineer7777, 2023-04-27:
Does your test code work properly under earlier versions of arduino-esp32?

anishsheikh, 2023-04-27:
Yeah, it did . Can you see any irregularities in the code btw ? 
I will try to fix it today.

mrengineer7777, 2023-04-27:
@anishsheikh I haven't used MBED so I can't comment on your code. I can think of two reasons the behavior might have changed between versions:
1) Change in `random` https://github.com/espressif/arduino-esp32/pull/7848
2) The MBED version bundled in IDF might have changed.

It would be useful to know the arduino-esp32 versions your test code works on vs breaks.

anishsheikh, 2023-04-27:
I got it working, somehow it messes up using different contexts irregularly, the mpi bignum functions don't actually read the key properly,
this seems not to affect other architectures, if we use same context things work just fine, but that is not how its supposed to be,
can you consult the compiler related problems, this in not a fix, its just a trick to make it work, an appropriate fix will be on compiler, and how it handles the mpi functions.

mrengineer7777, 2023-04-27:
@anishsheikh That sounds important.  Unfortunately I'm not following you.  Can you spell it out for us?  What lines of code need to change?

anishsheikh, 2023-04-27:
Can you try to reproduce the problem? 

You have to just add some method to give the esp32 the generated public key and esp32 have to convert the hexadecimal into a byte array then input the byte array into  calcsecret function

lbernstone, 2023-04-28:
This really has nothing to do with arduino.  I would recommend you post the issue on [esp-idf](https://github.com/espressif/esp-idf/issues).  They will have much deeper knowledge of the workings of the mbedtls hardware.

anishsheikh, 2023-04-28:
well i used arduino ide, so I posted here. anyway i will try to post there

closed 2023-04-28 by anishsheikh

PavloMcr, 2023-08-01:
@anishsheikh, could you please share your fix of the issue, it would be much appreciated.

PavloMcr, 2023-08-01:
@anishsheikh, I found the issue in your code, when you call sizeof on a pointer, it doesn't give you the size of the data the pointer points to, but rather the size of the pointer itself. This would be 4 bytes for a 32-bit system or 8 bytes for a 64-bit system. Thus, in memcpy you're only porting the first 4 bytes instead of 32. Replacing sizeof with 32 fixes the issue for me.

anishsheikh, 2023-08-02:
I’ve forgot to notify , i fixed the issue long ago. 
giving the size of the array is a fix . But isn’t the compiler supposed to manage that .

anishsheikh, 2023-08-02:
In x86_64 and aarch64 the compiler itself managed it. For code portability it should be manually managed .