espressif/arduino-esp32 #8133: USBHIDMouse.h with Absolute Mouse positioning

gree303, 2023-04-28:
### Related area

ESP32

### Hardware specification

ESP32-S2 / Nano 33 IOT

### Is your feature request related to a problem?

I'm trying to get an absolute mouse working but I'm stuck.
The USBHIDMouse does work great with relative mouse movements.
Is there any way of inserting this feature, technically speaking?
Can you tell me where I'd find the descriptor for it; this way I could fiddle my way through.

### Describe the solution you'd like

Maybe you know a workaround or another Library which might work.

I've tried different attempts with different solutions; MouseTo.h, AbsMouse, HID-Project...

I do have a working CuircutPython with random absolute mouse movements running from @dglaude
It would be a real bummer switching to python this late in my project.

Any tipp or suggestion will help me... Ty

### Describe alternatives you've considered

MouseTo / based on Mouse.h
AbsMouse
AbsoluteMouse5
HID-Project
TFT Touchscreen Simulation 
... all kind of forum suggestions and chatGPT testing failed or resulted in a relative mouse.
All tested on multiple os's (Win/Mac/Linux)
Most of the stuff is 5-10 years old and kinda outdated ;/

### Additional context

I don't mind buying you guys'n'girls a couple of beers or so
Thanks

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

gree303, 2023-04-28:
PS: I thought about the Teensy 3.1 solution by Elmue @ [Teensy 3.1](https://www.codeproject.com/Articles/1001891/A-USB-HID-Keyboard-Mouse-Touchscreen-emulator-with)

But at this point I'm kinda lost in all the research; looking forward to your input,ty.

SuGlider, 2023-04-28:
@gree303 - Something like this?
``` cpp
    AbsoluteMouse.begin();
    AbsoluteMouse.click(50, 50);    //Middle of screen
```

gree303, 2023-04-28:
I have not tested your code; but yes.

SuGlider, 2023-04-28:
There is a HID BLE implementation for Absolute Mouse in
https://github.com/sobrinho/ESP32-BLE-Abs-Mouse

It may help your project.

SuGlider, 2023-04-28:
This other Arduino project may also help:
https://github.com/jonathanedgecombe/absmouse

It is an USB HID report implementation for absolute mouse.

I think that these 2 links are good pointers to implement your project.

gree303, 2023-05-10:
Thanks @SuGlider I will have a look.

wukongxuetang, 2023-11-01:
just this is a sample for absolute mouse
https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/sdkoverride/tusb_absmouse.h

wukongxuetang, 2023-11-01:
and this one https://github.com/hathach/tinyusb/pull/1363/files/8c79b8924eabf441f40a8ce3cab8828688575b6d
esp32 tools have used tinyusb code.

mgi24, 2024-11-27:
UPDATE! I have found way to bypass 8 bit limit on mouse move, if you use arduino mouse, checkout this source where i found the way to do it: https://forum.arduino.cc/t/mouse-move-move-by-more-than-127-units/656825/3

If you using esp_tinyusb from esp-idf (version 1.4.5 with version 0.15 tinyusb dependencies as time of writing this) please edit 
managed_components\espressif__tinyusb\src\class\hid\hid_device.h to this

// Mouse Report Descriptor Template
#define TUD_HID_REPORT_DESC_MOUSE(...) \
  HID_USAGE_PAGE ( HID_USAGE_PAGE_DESKTOP      )                   ,\
  HID_USAGE      ( HID_USAGE_DESKTOP_MOUSE     )                   ,\
  HID_COLLECTION ( HID_COLLECTION_APPLICATION  )                   ,\
    /* Report ID if any */\
    __VA_ARGS__ \
    HID_USAGE      ( HID_USAGE_DESKTOP_POINTER )                   ,\
    HID_COLLECTION ( HID_COLLECTION_PHYSICAL   )                   ,\
      HID_USAGE_PAGE  ( HID_USAGE_PAGE_BUTTON  )                   ,\
        HID_USAGE_MIN   ( 1                                      ) ,\
        HID_USAGE_MAX   ( 5                                      ) ,\
        HID_LOGICAL_MIN ( 0                                      ) ,\
        HID_LOGICAL_MAX ( 1                                      ) ,\
        /* Left, Right, Middle, Backward, Forward buttons */ \
        HID_REPORT_COUNT( 5                                      ) ,\
        HID_REPORT_SIZE ( 1                                      ) ,\
        HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_ABSOLUTE ) ,\
        /* 3 bit padding */ \
        HID_REPORT_COUNT( 1                                      ) ,\
        HID_REPORT_SIZE ( 3                                      ) ,\
        HID_INPUT       ( HID_CONSTANT                           ) ,\
      HID_USAGE_PAGE  ( HID_USAGE_PAGE_DESKTOP )                   ,\
        /* X, Y position [-127, 127] */ \
        HID_USAGE       ( HID_USAGE_DESKTOP_X                    ) ,\
        HID_USAGE       ( HID_USAGE_DESKTOP_Y                    ) ,\
        HID_LOGICAL_MIN_N ( -2047, 2                           ) ,\
        HID_LOGICAL_MAX_N ( 2047, 2                             ) ,\
        HID_REPORT_COUNT( 2                                      ) ,\
        HID_REPORT_SIZE ( 16                                      ) ,\
        HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE ) ,\
        /* Verital wheel scroll [-127, 127] */ \
        HID_USAGE       ( HID_USAGE_DESKTOP_WHEEL                )  ,\
        HID_LOGICAL_MIN ( 0x81                                   )  ,\
        HID_LOGICAL_MAX ( 0x7f                                   )  ,\
        HID_REPORT_COUNT( 1                                      )  ,\
        HID_REPORT_SIZE ( 8                                      )  ,\
        HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE )  ,\
      HID_USAGE_PAGE  ( HID_USAGE_PAGE_CONSUMER ), \
       /* Horizontal wheel scroll [-127, 127] */ \
        HID_USAGE_N     ( HID_USAGE_CONSUMER_AC_PAN, 2           ), \
        HID_LOGICAL_MIN ( 0x81                                   ), \
        HID_LOGICAL_MAX ( 0x7f                                   ), \
        HID_REPORT_COUNT( 1                                      ), \
        HID_REPORT_SIZE ( 8                                      ), \
        HID_INPUT       ( HID_DATA | HID_VARIABLE | HID_RELATIVE ), \
    HID_COLLECTION_END                                            , \
  HID_COLLECTION_END \

it will allow int16_t to be passed on USB buffer with max of -2047 to 2047 (technically you can get up to 32K, but why?)

mgi24, 2024-11-27:
if you're using ESP32 Mouse, change accordingly!

mgi24, 2024-12-10:
> 您好 刚好要用esp32s3实现HID鼠标的绝对移动 找了好多帖子 都没实现，我改了这个hid报告描述 程序写入后鼠标没反应，是缺少什么库吗

if you're using ArduinoIDE, read C:\Users\(yourname)\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\USB\src

you will see function to call ABSELUTE mouse

void USBHIDAbsoluteMouse::move(int16_t x, int16_t y, int8_t wheel, int8_t pan) {
  hid_abs_mouse_report_t report;
  report.buttons = _buttons;
  report.x = _lastx = x;
  report.y = _lasty = y;
  report.wheel = wheel;
  report.pan = pan;
  sendReport(report);
}

void USBHIDAbsoluteMouse::click(uint8_t b) {
  _buttons = b;
  move(_lastx, _lasty);
  _buttons = 0;
  move(_lastx, _lasty);
}

void USBHIDAbsoluteMouse::buttons(uint8_t b) {
  if (b != _buttons) {
    _buttons = b;
    move(_lastx, _lasty);
  }
}

you can use it by editing this example file
(ArduinoIDE > File > Example > USB > Mouse > ButtonMouseControl) 
 
remember, Abselute mouse is used for touchscreen! it takes the resolution of your screen!
for example USBHIDAbsoluteMouse::move(16000,16000,0,0) will move the cursor in the middle of the screen because 16000 is half of 32000. Absolute mouse did't accept negative number, 0,0 is minimal start from top left of your screen

if what you want is Bypass 8bit int limit on each mouse report (-127 to 127 only) use ESP-IDF (NOT ARDUINO IDE), then open example, search for "tusb_hid", compile it! (you need to compile it at least once to satisfy dependencies)

on your project file (use visual studio code for easier access) goto managed_component>espressif_tinyusb>src>class>hid>hid_device.c and hid_device.h <this should be visible after you compile it, no compile = not exist!

edit or change with this file: (ESP IDF version 5.1.4)
[hid_device.zip](https://github.com/user-attachments/files/18077001/hid_device.zip)

916881120, 2024-12-10:
> > 您好 刚好要用esp32s3实现HID鼠标的绝对移动 找了好多帖子 都没实现，我改了这个hid报告描述 程序写入后鼠标没反应，是缺少什么库吗
> 
> 如果您使用的是 ArduinoIDE，请阅读 C：\Users（yourname）\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries\USB\src\USBHIDMouse.cpp
> 
> 您将看到调用 ABSELUTE 鼠标的函数
> 
> void USBHIDAbsoluteMouse：：move（int16_t x， int16_t y， int8_t wheel， int8_t pan） { hid_abs_mouse_report_t report; report.buttons = _buttons; report.x = _lastx = x; report.y = _lasty = y; report.wheel = wheel; report.pan = pan; sendReport（report）; }
> 
> 无效 USBHIDAbsoluteMouse：：click（uint8_t b） { _buttons = b; move（_lastx， _lasty）; _buttons = 0; move（_lastx， _lasty）; }
> 
> 无效 USBHIDAbsoluteMouse：：buttons（uint8_t b） { if （b ！= _buttons） { _buttons = b; move（_lastx， _lasty）; } }
> 
> 您可以通过编辑此示例文件来使用它 （ArduinoIDE > File > Example > USB > Mouse > ButtonMouseControl）
> 
> 请记住，Abselute 鼠标用于触摸屏！它需要您屏幕的分辨率！例如 USBHIDAbsoluteMouse：：move（16000,16000,0,0） 会将光标移动到屏幕中间，因为 16000 是 32000 的一半。绝对鼠标不接受负数，0,0 是从屏幕左上角开始的最小
> 
> 如果你想要的是绕过每个鼠标报告的 8 位 int 限制（仅限 -127 到 127），请使用 ESP-IDF（不是 ARDUINO IDE），然后打开示例，搜索 “tusb_hid”，编译它！（您需要至少编译一次以满足依赖项）
> 
> 在您的项目文件（使用 Visual Studio Code 以便于访问）上，转到 managed_component>espressif_tinyusb>src>class>hid>hid_device.c 和 hid_device.h <这在编译后应该可见，没有编译 = 不存在！
> 
> 使用此文件进行编辑或更改：（ESP IDF 版本 5.1.4）[hid_device.zip](https://github.com/user-attachments/files/18077001/hid_device.zip)

好的 非常感谢   终于有点进展了 非常感谢

Parsaabasi, 2026-10-14:
Hello,  

Closing this issue because it refers to an older version. Please update to the latest version and check if the issue persists. If you still need help, feel free to reopen this issue or create a new one. Thanks!

closed 2026-10-14 by Parsaabasi