espressif/arduino-esp32 #8134: ESP32 MDNS.queryService returns only one instance per service.

JesFo, 2023-04-28:
### Board

wESP32, Olimex POE ISO, ESP32_CAM, ESP32 Devkit V4

### Device Description

I'm trying to get this to work mainly on a wESP32-POE devkit. However, this issue still occurs with any other ESP32 devkit.

### Hardware Configuration

I/O peripherals have no effect on this issue. I've tried multiple different devkits with and without anything attached to the I/O. Omitting any peripheral libraries don't affect the results either.

### Version

v2.0.8

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I've tried getting a list of service instances with this code:
`numServices = MDNS.queryService("_revpipyload", "tcp");`
When using a service browser, I see both of my instances like so:
[image]

Both the MQTT and NodeRED instances are visible.
However, no matter what I do, the line above always only returns "1". Also, whenever i try to look up the entries with `MDNS.hostname(X)`. Only index 0 returns a value.
I've tried this with different ESP32 devkits, Including but not limited to: wESP32, Olimex POE ISO, ESP32_CAM, ESP32 Devkit V4, and so on...
Each one so far has the same problem.

I've also tried changing the service names and protocols using other Avahi/Bonjour capable devices, but I get the same results.
I use the ESP32 Arduino core version 2.0.8 with PlatformIO.

Even if i try manually to print out `MDNS.hostname(1)` I get nothing.
Is this considered a bug?

### Sketch

```cpp
void setup()
{
  Serial.begin(115200); 
  ETH.begin(0, -1, 16, 17, ETH_PHY_RTL8201);
  MDNS.begin("ControlUnit");
  MDNS.addService("revpipyload", "tcp", 123);
  numServices = MDNS.queryService("_revpipyload", "tcp");
  Serial.print("Number of services found: ");
  Serial.println(numServices);
  for (int i = 0; i < numServices; i=i+1) {

    Serial.println("---------------");
      
    Serial.print("Hostname: ");
    Serial.println(MDNS.hostname(i));

    Serial.print("IP address: ");
    Serial.println(MDNS.IP(i));

    Serial.print("Port: ");
    Serial.println(MDNS.port(i));

    Serial.println("---------------");
  }
}
```

### Debug Message

```plain
No relevant output
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-04-28:
@JesFo - Arduino mDNS services are based on the IDF 4.4.4 mDNS API
https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/protocols/mdns.html

I understand that any mDNS operation done in IDF/Arduino is meant only for the internal ESP32 mDNS Server.
It doesn't query the local network searching for external mDNS hosts/services.

Therefore, it will never find an external mDNS host/service.
It finds the mDNS host `ControlUnit` with its own service `revpipyload`. 
`MDNS.queryService("_revpipyload", "tcp");` will return always 1.

me-no-dev, 2023-04-29:
What does this return? `MDNS.queryService("revpipyload", "tcp");`

> I understand that any mDNS operation done in IDF/Arduino is meant only for the internal ESP32 mDNS Server.
It doesn't query the local network searching for external mDNS hosts/services.

Not true. It is a standard implementation of mDNS service and is certified by Apple.

SuGlider, 2023-04-29:
`numServices = MDNS.queryService("_revpipyload", "tcp");` will return the number of hosts that has `revpipyload/TCP` service.
I understand that with an ESP32 serving it, the answer is only 1, itself.

If you want to query the IP of the MQTT service host, you shall execute
`Serial.printf("MQTT IP Address is %s\n",  MDNS.queryHost("MQTT", timeout_ms).toString().c_str());`

This ESP32 mDNS Arduino API is `IPAddress MDNSResponder::queryHost(char *host, uint32_t timeout)`

JesFo, 2023-05-02:
> 

No, that's not it. All of my service instances advertise themselves as TCP based. Even the MQTT instance shows up on my ESP if it happens to be the first one picked up by the mDNS query.

All of the service instances do show up individually on the ESP32 mDNS query.
My issue is that the query only returns one instance of a service at any time, depending on which one was available first.
I can get all of them to show up on the query, but not at the same time, or get a full count of service instances over the network. Unlike the desktop browser i used for testing the services (which finds and displays all the available service instances just fine), I only find a single instance of any given service on my ESP32.

If this is intended behaviour for `MDNS.queryHost()` then why does the function `MDNS.hostname(int)` take an index as an argument?

SuGlider, 2023-05-02:
I see. That line `MDNS.addService("revpipyload", "tcp", 123);` in the snippet was causing some confusion, indeed.

The Arduino mDNS Query API is based on the IDF implementation.
https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/protocols/mdns.html#mdns-query
There is an example using IDF: https://github.com/espressif/esp-idf/tree/v4.4.4/examples/protocols/mdns

Reading the code for the Query API, I'm not sure that it is possible to do wat you are looking for.
https://github.com/espressif/esp-idf/blob/release/v4.4/components/mdns/include/mdns.h#L557-L577

``` cpp
/**
 * @brief  Query mDNS for host or service
 *         All following query methods are derived from this one
 *
 * @param  name         service instance or host name (NULL for PTR queries)
 * @param  service_type service type (_http, _arduino, _ftp etc.) (NULL for host queries)
 * @param  proto        service protocol (_tcp, _udp, etc.) (NULL for host queries)
 * @param  type         type of query (MDNS_TYPE_*)
 * @param  timeout      time in milliseconds to wait for answers.
 * @param  max_results  maximum results to be collected
 * @param  results      pointer to the results of the query
 *                      results must be freed using mdns_query_results_free below
 *
 * @return
 *     - ESP_OK success
 *     - ESP_ERR_INVALID_STATE  mDNS is not running
 *     - ESP_ERR_NO_MEM         memory error
 *     - ESP_ERR_INVALID_ARG    timeout was not given
 */
esp_err_t mdns_query(const char * name, const char * service_type, const char * proto, uint16_t type, uint32_t timeout, size_t max_results, mdns_result_t ** results);

/**
 * @brief  Query mDNS for service
 *
 * @param  service_type service type (_http, _arduino, _ftp etc.)
 * @param  proto        service protocol (_tcp, _udp, etc.)
 * @param  timeout      time in milliseconds to wait for answer.
 * @param  max_results  maximum results to be collected
 * @param  results      pointer to the results of the query
 *
 * @return
 *     - ESP_OK success
 *     - ESP_ERR_INVALID_STATE  mDNS is not running
 *     - ESP_ERR_NO_MEM         memory error
 *     - ESP_ERR_INVALID_ARG    parameter error
 */
esp_err_t mdns_query_ptr(const char * service_type, const char * proto, uint32_t timeout, size_t max_results, mdns_result_t ** results);

```

IDF Code seems to require that both Service Name and Protocol Name are not empty.
### Thus, I can't see a way to just query something like "tell me all the TCP mDNS services in the LAN" 
https://github.com/espressif/esp-idf/blob/release/v4.4/components/mdns/mdns.c#L5538-L5546
``` cpp
esp_err_t mdns_query_ptr(const char * service, const char * proto, uint32_t timeout, size_t max_results, mdns_result_t ** results)
{
    if (_str_null_or_empty(service) || _str_null_or_empty(proto)) {
        return ESP_ERR_INVALID_ARG;
    }

    return mdns_query(NULL, service, proto, MDNS_TYPE_PTR, timeout, max_results, results);
}
```

Looking into the example from the documentation, it sounds like the Query must be specific.
``` cpp
void find_mdns_service(const char * service_name, const char * proto)
{
    ESP_LOGI(TAG, "Query PTR: %s.%s.local", service_name, proto);

    mdns_result_t * results = NULL;
    esp_err_t err = mdns_query_ptr(service_name, proto, 3000, 20,  &results);
    if(err){
        ESP_LOGE(TAG, "Query Failed");
        return;
    }
    if(!results){
        ESP_LOGW(TAG, "No results found!");
        return;
    }

    mdns_print_results(results);
    mdns_query_results_free(results);
}

void my_app_some_method(){
    //search for esp32-mdns.local
    resolve_mdns_host("esp32-mdns");

    //search for HTTP servers
    find_mdns_service("_http", "_tcp");
    //or file servers
    find_mdns_service("_smb", "_tcp"); //windows sharing
    find_mdns_service("_afpovertcp", "_tcp"); //apple sharing
    find_mdns_service("_nfs", "_tcp"); //NFS server
    find_mdns_service("_ftp", "_tcp"); //FTP server
    //or networked printer
    find_mdns_service("_printer", "_tcp");
    find_mdns_service("_ipp", "_tcp");
}
```

SuGlider, 2023-05-02:
I'm not clear about exactly what you are trying to achieve.
At this time, I think that you want to query for a list of all TCP mDNS services available in the LAN, like done by the Service Browser desktop application.

But I think that it is not how ESP32 IDF mDNS does its queries. It seems that for using an IDF query must pass the name of the Service and the name of the Protocol. Or query for a Host name IP. Arduino just does exactly the same that IDF can do.

SuGlider, 2023-05-24:
@JesFo - Closing the issue. Feel free to reopen it, whenever necessary.

closed 2023-05-24 by SuGlider