espressif/arduino-esp32 #8138: How can I set the optimization level> O2 at arduino

Magical-Lin, 2023-04-29:
### Related area

I want to use the Feature such as usb_soft_host

### Hardware specification

esp32c3

### Is your feature request related to a problem?

 I'm facing the issue that I don't know how to set optimization level to O2 at arduino.
I want to use this demo[ usb_soft_host ](https://github.com/sdima1357/esp32_usb_soft_host.git)  that I can use the usb_host at esp32c3

### Describe the solution you'd like

Please tell me how to optimization level to O2 in Adruino, thank you very much !

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-04-29:
You may use an Arduino resource called `platform.local.txt`
https://arduino.github.io/arduino-cli/0.21/platform-specification/#platformlocaltxt
It is used to override properties defined in `platform.txt` file.

It must set a new value for `compiler.optimization_flags=-Os`, changing it to `compiler.optimization_flags=-O2`
https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L97

`platform.local.txt` shall be placed in the same folder where the ESP32 Arduino `platform.txt` has been installed.
This folder is in the `Arduino15/packages/esp32/hardware/esp32/2.0.8` in the case of the Arduino Core 2.0.8.
You can find the location of the folder `Arduino15` for your operating system in the documentation below: 
https://support.arduino.cc/hc/en-us/articles/360018448279-Open-the-Arduino15-folder

The content of the `platform.local.txt` file shall be just one line:
`compiler.optimization_flags=-O2`

SuGlider, 2023-04-29:
A second option would be build this project with "Arduino as IDF component" and set the compiler optimization in `menuconfig`.

Magical-Lin, 2023-04-30:
> A second option would be build this project with "Arduino as IDF component" and set the compiler optimization in `menuconfig`.

Thank you very much, with you help I found it. Dir : C:\Users\myusername\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.8\platform.txt
[image]

closed 2023-04-30 by Magical-Lin