espressif/arduino-esp32 #8141: Code signing for [OTA] Updates

tushev, 2023-05-01:
### Related area

Updater library, Build process, OTA

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

I'm missing the possibility to enforce signed OTA updates with ESP32, which are possible with ESP8266. This is critical to ensure that only authorized code will run on end devices, even if update server gets compromised.

### Describe the solution you'd like

The same procedure as with ESP8266:
https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#advanced-security-signed-updates

```cpp
// <in globals>
BearSSL::PublicKey signPubKey( ... key contents ... );
BearSSL::HashSHA256 hash;
BearSSL::SigningVerifier sign( &signPubKey );
//...
//<in setup()>
Update.installSignature( &hash, &sign );
```

- https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.cpp
- https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.h
- https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater_Signing.h
- https://github.com/esp8266/Arduino/blob/master/tools/signing.py

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

tushev, 2025-01-16:
This is still a very important feature for anyone who needs to perform authenticated OTA updates.

Please consider prioritizing it.