espressif/arduino-esp32 #8142: Wifimulti not connecting if SSID exists twice

s0170071, 2023-05-01:
### Board

Liligo T-Display S3

### Device Description

ESP32S3 with display

### Hardware Configuration

https://www.lilygo.cc/products/t-display-s3

### Version

latest master (checkout manually)

### IDE Name

Adruino IDE 2.1.0 

### Operating System

Linux Mint 21

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

In my home I run two routers "Fritzbox", one 7590 and a 4020. The 7590 is the master and controls the settings of the 4020. 
Both use the same SSID.  

In that case, the wifiMulti does not connect. If I pull the plug on one of the routers, it works.

### Sketch

```cpp
Serial.begin(115200);         // Start the Serial communication to send messages to the computer
  delay(10);
  WiFi.mode(WIFI_STA);   //To Turn on WiFi in Specific Mode like WIFI_STA or WIFI_AP of WIFI_AP_STA or WIFI_OFF
  wifiMulti.addAP("MNET", "vanillasky");   // add Wi-Fi networks you want to connect to
  wifiMulti.addAP("MNET", "vanillasky");   // add Wi-Fi networks you want to connect to
  wifiMulti.addAP("FRONIUS_33690527", "12345678");
  wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");
  

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0) {
      Serial.println("no networks found");
  } 
  else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
      delay(10);
    }
  }

 

  LOG("Connecting ...");
  uint8_t status = WL_DISCONNECTED; 
  while ( status != WL_CONNECTED) { // Wait for the Wi-Fi to connect: scan for Wi-Fi networks, and connect to the strongest of the networks above
    status = wifiMulti.run(5000); 
    delay(250);
    LOG('.');
    LOG (status); 
  }
```

### Debug Message

```plain
networks found
1: MNET (-73)*
2: MNET (-81)*

Connecting ....6.6.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.6.6.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.4.4.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.4.4.1.1.1.1.1.1.1
```

### Other Steps to Reproduce

In the debug message you can see that there actually are two networks with the same SSID and the network states it goes through separated by a dot. The sequence goes on forever until one network is not reachable.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mahboud, 2023-05-16:
The WiFiMulti,run needs to have a more complex test of IP address availability and possibly connectivity, after selecting the strongest network. If no IP address is available, and/or no IP pinging is functional, then the next strongest SSID needs to be selected.

sudheeshsmadhav, 2024-02-19:
Any update on this issue? WiFImulti keeps disconnecting form the network after every 20minutes.

mhaberler, 2024-04-18:
I found that completely powering off the device completely makes it again connect immediately on first scan. 

Looks like leftover state in the hardware persisting a reset to me.

YMMV.

currently using assorted betas
```ini
platform = https://github.com/mhaberler/platform-espressif32.git#Arduino/IDF5_mah
platform_packages = espressif/toolchain-xtensa-esp32s3
  framework-arduinoespressif@ https://github.com/mhaberler/arduino-esp32.git#mah-work

framework = arduino
```

mega-byte, 2024-10-17:
Hi All
I'm experiencing similar problems using ESP32-S2. As I see it the only benefit that wifimulti brings is in the initial connect to the strongest signal. After that you are in a hole if the connection is lost. This is a fundamental problem that has now been reported multiple times over an extended period. Any chance that a fix could be given some priority please?

Parsaabasi, 2025-01-09:
Hello, 

I close it, since this report contains a release no longer supported. Please try newer version. Incase the issue persists, please reopen it. 

Thanks.

closed 2025-01-09 by Parsaabasi