espressif/arduino-esp32 #8144: Manually set local port of AsyncUDP instance

cwilling, 2023-05-02:
### Related area

Async UDP messaging over wifi

### Hardware specification

Using ESP32-E here but don't believe that is significant.

### Is your feature request related to a problem?

udp.print(msg) chooses a local port or uses port from incoming ASyncUDPPacket, depending on who initiates message exchange. Such an exchange of UDP messages therefore consistently uses the same local port. This is good since, in my use case, the remote software is using ipaddr/port to categorize many different message sources.

The problem arises when the local device awakes from deep sleep. In this case there is no "memory" of the last ASyncUDPPacket received; therefore the next message sent by the device will have a new local port assigned. I could save the port number as RTC_DATA_ATTR during sleep but there is no way to apply it after wakeup. Since any new message is now bound to a new local port, the remote software cannot match any new message to the previous messages.

### Describe the solution you'd like

It would be preferred to be able to manually set the local port in the constructor or helper method - similar to WiFiUDP library which has WiFiUDP::begin(uint16_t p) method. A separate bind(uint16_t p) method would also work.

### Describe alternatives you've considered

I have tried using WiFiUDP library instead - it works but misses advantages of async.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-05-02:
I suspect the localPort should stay dynamic (assigned by OS).  You probably want to change the remote port. I see listen(), writeTo(), sendTo(), broadcastTo() specify a port.  Would one of those work?
https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/src/AsyncUDP.cpp#L653

cwilling, 2023-05-02:
No those other methods would not work since they specify only the destination port, not the local port.

It is crucial that the same local port is maintained over multiple message exchanges; the remote system identifies individual correspondents by their ipaddr/port combination. A new message with different port number would be classified as a different correspondent i.e. the same local port must be used. This is not possible after wakeup from sleep because the local port (normally maintained through ASyncUDPPacket exchanges) has been lost. As I already mentioned, I can store the local port and retrieve it after a wakeup but currently have no way of applying it to the next message to be sent.

mrengineer7777, 2023-05-03:
Ok. Which function calls are you using to set the target IP and then send the message?

cwilling, 2023-05-03:
In response to a broadcast requesting service details (udp.broadcastTo(output, discovery_port) where output is the request message), I receive a ASyncUDPPacket containing available services from which I extract the remote port number of the service I want to interact with. It looks like my broadcast message has chosen to use the same local port as the destination port because I receive the response on that same port number - udp.listen(discovery_port) followed by udp.onPacket([](AsyncUDPPacket message) {...}. I'm able to respond with my next message using udp.print(output) (where output is my new message).

I'm now able to send and receive messages this way as long as I like until my device wakes from deep sleep. At this point I need to reestablish a new AsyncUDP object - there's no recall of the previous one I was using. I need a new one that I can feed with a local port number to use, so that the device can send messages to the server and have the server recognize the device as being the same one it was previously communicating with (by virtue of coming from same ipaddr/port number).

(hope that makes sense - I'm in a rush to catch a flight interstate)

closed 2025-01-16 by Parsaabasi