espressif/arduino-esp32 #8153: OBDII unable to connect

hobbit-xD, 2023-05-03:
### Board

ESP32 WROOM 32

### Device Description

Esp32, in arduino using DOIT ESP32 DEVKIT V1

### Hardware Configuration

Nothing connected

### Version

v2.0.8

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Good morning, 
i am trying to connect my esp32 to OBD2 device, but without success. Below you can find the log

It seems that the ESP_SPP_DISCOVERY_COMP_EVT fails
Could you help me please?

Thanks a lot
Amedeo

### Sketch

```cpp
I am using SerialToSerialBTM.ino sketch

// This example code is in the Public Domain (or CC0 licensed, at your option.)
// By Victor Tchistiak - 2019
//
// This example demonstrates master mode Bluetooth connection to a slave BT device using PIN (password)
// defined either by String "slaveName" by default "OBDII" or by MAC address
//
// This example creates a bridge between Serial and Classical Bluetooth (SPP)
// This is an extension of the SerialToSerialBT example by Evandro Copercini - 2018
//
// DO NOT try to connect to phone or laptop - they are master
// devices, same as the ESP using this code - it will NOT work!
//
// You can try to flash a second ESP32 with the example SerialToSerialBT - it should
// automatically pair with ESP32 running this code

#include "BluetoothSerial.h"

//#define USE_NAME           // Comment this to use MAC address instead of a slaveName
const char *pin = "1234";  // Change this to reflect the pin expected by the real slave BT device

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

#ifdef USE_NAME
String slaveName = "OBDII";  // Change this to reflect the real name of your slave BT device
#else
String MACadd = "AA:BB:CC:11:22:33";                          // This only for printing
uint8_t address[6] = { 0xAA, 0xBB, 0xCC, 0x11, 0x22, 0x33 };  // Change this to reflect real MAC address of your slave BT device
#endif

String myName = "ESP32-BT-Master";

void setup() {
  bool connected;
  Serial.begin(115200);

  SerialBT.begin(myName, true);
  Serial.printf("The device \"%s\" started in master mode, make sure slave BT device is on!\n", myName.c_str());

#ifndef USE_NAME
  SerialBT.setPin(pin);
  Serial.println("Using PIN");
#endif

// connect(address) is fast (up to 10 secs max), connect(slaveName) is slow (up to 30 secs max) as it needs
// to resolve slaveName to address first, but it allows to connect to different devices with the same name.
// Set CoreDebugLevel to Info to view devices Bluetooth address and device names
#ifdef USE_NAME
  connected = SerialBT.connect(slaveName);
  Serial.printf("Connecting to slave BT device named \"%s\"\n", slaveName.c_str());
#else
  connected = SerialBT.connect(address);
  Serial.print("Connecting to slave BT device with MAC ");
  Serial.println(MACadd);
#endif

  if (connected) {
    Serial.println("Connected Successfully!");
  } else {
    while (!SerialBT.connected(30000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app.");
    }
  }
  // Disconnect() may take up to 10 secs max
  if (SerialBT.disconnect()) {
    Serial.println("Disconnected Successfully!");
  }
  // This would reconnect to the slaveName(will use address, if resolved) or address used with connect(slaveName/address).
  SerialBT.connect();
  if (connected) {
    Serial.println("Reconnected Successfully!");
  } else {
    while (!SerialBT.connected(10000)) {
      Serial.println("Failed to reconnect. Make sure remote device is available and in range, then restart app.");
    }
  }
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[     5][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   508][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   508][I][BluetoothSerial.cpp:683] _init_bt(): device name set
The device "ESP32-BT-Master" started in master mode, make sure slave BT device is on!
[   511][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
[   523][I][BluetoothSerial.cpp:903] setPin(): pin: 1234
[   531][I][BluetoothSerial.cpp:146] btSetPin(): pin set
Using PIN
[   527][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   546][I][BluetoothSerial.cpp:978] connect(): master : remoteAddress
[  5686][I][BluetoothSerial.cpp:350] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT num=0
[  5687][E][BluetoothSerial.cpp:378] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT failed!, status:1
[ 10551][D][BluetoothSerial.cpp:763] waitForConnect(): timeout
Connecting to slave BT device with MAC AA:BB:CC:11:22:33
[ 40551][D][BluetoothSerial.cpp:763] waitForConnect(): timeout
Failed to connect. Make sure remote device is available and in range, then restart app.
[ 70561][D][BluetoothSerial.cpp:763] waitForConnect(): timeout
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-08-18:
Hello, unfortunately I'm closing this ticket as expired, if needed you can reopen it.

closed 2023-08-18 by VojtechBartoska

bluegizmo83, 2023-10-24:
I too have this exact same issue. I can get two ESP32's to connect to each other via BluetoothSerial using either RemoteName or Address, but I cannot get any of my ESP32's to connect to any other bluetooth serial devices using either name or address. My debug log looks identical to the one posted here, same exact messages and errors.

themelisx, 2023-12-13:
I have the same problem while trying to connect my ESP32 to ELM327

[  7374][E][BluetoothSerial.cpp:378] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT failed!, status:1

hobbit-xD, 2024-07-06:
Hi @JayKapse, No I couldn't solve it.
I switched to a Raspberry Pi Pico W with micropython for this project, because with esp32 not able to connect.

JayKapse, 2024-07-07:
I found the solution Hobbit. You have to downgrade to esp 2.0.17 in the board manager. Just tried my luck and it worked!

PaoloGolinelli, 2024-08-26:
> I too have this exact same issue. I can get two ESP32's to connect to each other via BluetoothSerial using either RemoteName or Address, but I cannot get any of my ESP32's to connect to any other [...]

How did you manage to connect the two ESP32 together? I am using the very simple Bluetooth Classic method, but the slave fails to connect to the master.

gowata, 2024-10-19:
> I found the solution Hobbit. You have to downgrade to esp 2.0.17 in the board manager. Just tried my luck and it worked!

Works for me too ,  thanks @JayKapse !!!!

Finn7D, 2024-12-05:
> I found the solution Hobbit. You have to downgrade to esp 2.0.17 in the board manager. Just tried my luck and it worked!

Thanks for the quick fix, can confirm this makes it work when connecting HC-06 modules.

frankyoualot, 2025-02-04:
> I found the solution Hobbit. You have to downgrade to esp 2.0.17 in the board manager. Just tried my luck and it worked!

This saved me from hours of frustrations.  Thank you so much.
Why on earth do new versions break something this important?