espressif/arduino-esp32 #8158: [CI] merge_packages.py overwriting platform.txt::version value from build

tobozo, 2023-05-04:
### Board

any 

### Device Description

any

### Hardware Configuration

any

### Version

v2.0.9

### IDE Name

Arduino IDE 1.x.x

### Operating System

any

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

According to the last commit on 2.0.9, the platform.txt should come with the following value:

[image]

However it doesnt, the `version` value is empty.

Also according to Arduino standards, the `name` field should also contain the version number, here's an example with the pico boards:

```
name=Raspberry Pi RP2040 Boards(3.1.1)
version=3.1.1
```

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

install the core from Arduino IDE Board Manager

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-02-20:
@me-no-dev I think we can close this and it is not relevant anymore, right?

me-no-dev, 2024-02-20:
right. long fixed

closed 2024-02-20 by me-no-dev