espressif/arduino-esp32 #8159: E (35051) timer_group: timer_set_counter_value(84): TIMER GROUP NUM ERROR

mars000, 2023-05-04:
### Board

LOLIN D32

### Device Description

Just LOLIN D32

### Hardware Configuration

LOLIN D32 development board on its own.

### Version

v2.0.8

### IDE Name

VSC with PlatformIO

### Operating System

Windows 11

### Flash frequency

default set by PlatformIO

### PSRAM enabled

no

### Upload speed

115200

### Description

Using PlatformIO V6.2.0 of ESP32 library.
When compiling using 6.1.0 or 6.2.0 it compiles fine and runs but board then starts to produce following error in Serial console.
Interestingly this does not appear if I compile using PlatformIO V6.0.1
Any ideas what could be causing this ?

[image]

### Sketch

```cpp
My sketch is huge....don't know which part to copy in.
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-05-04:
What is V6.2.0?  Are you using Arduino as IDF component?

Current versions:
VSCode PIO extension 3.1.1
PIO Core 6.1.6
PIO Home 3.4.3
arduino-esp32 2.0.9
idf 4.4, 5.1

mrengineer7777, 2023-05-05:
PIO espressif platform 6.1.0 targets arduino-esp32 v2.07.
https://github.com/platformio/platform-espressif32/releases/tag/v6.2.0 targets arduino-esp32 v2.0.8.

So it might be a change introduced after 2.0.7 (2023.02.20).  I looked through the commits after that date but I don't see anything related to timers.  There may be some other interaction in your code.

mars000, 2023-05-05:
I'm using Ticker,h and also using timerBegin for h/w watchdog using code below.
Could it be that Ticker,h also used timerBegin(0,x,x) from V6.1.0 onwards and therefore the issue ?

```
#ifdef HARDWARE_WATCHDOG

	hardwareTimer = timerBegin(0, 80, true); // used for watchdog   // timer 0, div 80

	timerAttachInterrupt(hardwareTimer, &resetIOTDevice, true); // attach callback

	timerAlarmWrite(hardwareTimer, HARDWARE_WATCHDOG_DEFAULT, false); // 1 second = 1,000,000 ... set time in micro (us)  -- set it at startup to 10s * 3 = 60seconds
	timerAlarmEnable(hardwareTimer);								  // enable interrupt

	Serial.print(F("1196-[main.cpp] HARDWARE_WATCHDOG timerAlarmRead value = "));
	Serial.println(timerAlarmRead(hardwareTimer));
	// timerAlarmDisable(hardwareTimer);
#endif
```

mrengineer7777, 2023-05-05:
It has been over a year(2022.03.31) since Ticker has been modified on any branches.  Not the issue.

This change may affect your code: https://github.com/espressif/arduino-esp32/pull/7809/files

mars000, 2023-05-06:
appreciate your responses.

In the end I managed to bypass TIMER error by not using the above code i.e. :
  ```
hardwareTimer = timerBegin(0, 80, true); // used for watchdog   // timer 0, div 80
  .....
```

....to manage hardware wdt, instead I use below code to manage wdt...

```
#include "esp_system.h"
#include "esp_int_wdt.h"
#include "esp_task_wdt.h"

void setup() {
  // Initialize and enable the hardware WDT with a 5-second timeout
  esp_task_wdt_init(5, true); // Set the timeout period to 5 seconds and enable panic (system reset)
  esp_task_wdt_add(NULL); // Add the current task (loop task) to the WDT

  // ...your other setup code
}

void loop() {
  // ...your loop code

  // Change the WDT timeout to 10 seconds
  esp_task_wdt_init(10, true); // This updates the WDT configuration without disabling it

  // Feed (reset) the hardware WDT
  esp_task_wdt_reset();

  // ...your other loop code
}
```
seems to work and I no longer get the errors, not sure why the original code did not work and caused errors in PlatfomrIO ESP32 librabries from >= V6.1.0   :-)

mrengineer7777, 2023-05-09:
Glad you resolved the issue. I also use esp_task_wdt in my code. Note that Arduino defaults to 15 second timeout. Be careful with http calls that can take 30s. I run my http calls in a separate thread to keep my UI responsive.

closed 2023-05-09 by mrengineer7777

Rob58329, 2024-03-29:
I suddenly started to get "E (1449370) timer_group: timer_set_counter_value(85): HW TIMER NUM ERROR" messages, but intermittently (ie. code would run fine for a bit, then I would get a load of the above messages).  But notwithstanding the above, code appeared to still be running ok.

I THINK (??) the fix was to add more "timerWrite(timer,0);" (watchdog reset) statements, as once I did this the above appears to have stopped happening!?

(I'm using arduino-IDE with github "https://github.com/espressif/arduino-esp32" as at 16Jan23.)