espressif/arduino-esp32 #8160: mkfatfs

tobozo, 2023-05-04:
### Related area

package_index.json

### Hardware specification

any

### Is your feature request related to a problem?

- Official arduino-esp32fs-plugin doesn't support FFat
- In `package_index.json`, the  `toolsDependencies` and `tools` groups only list mklittlefs and mkspiffs.

### Describe the solution you'd like

[lorol's version](https://github.com/lorol/arduino-esp32fs-plugin) supports FFat and the only current solution.

Since it depends on the existence of a "mkfatfs" tool in the package_index.json, the suggested solution is to add mklittlefs to the tools list.

Suggested tool: 
- https://github.com/labplus-cn/mkfatfs

Tested scopes:

- [x] x86_64-pc-linux-gnu
- [ ] aarch64-linux-gnu
- [ ] arm-linux-gnueabihf
- [ ] i686-pc-linux-gnu
- [ ] x86_64-apple-darwin
- [ ] arm64-apple-darwin
- [ ] i686-mingw32
- [ ] x86_64-mingw32

### Describe alternatives you've considered

Alternate tool (in rust but uses different flags):
- https://github.com/software-opal/mkfatfs 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

igrr, 2023-05-04:
Note that there is also [fatfs partition generator tool](https://github.com/espressif/esp-idf/blob/master/components/fatfs/fatfsgen.py) ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/fatfs.html#fatfs-partition-generator)) in IDF. It is written in Python but can be packaged into an executable similar to esptool and others.

tobozo, 2023-05-04:
the esp-idf python tool sounds good, does it needs to be edited or wrapped to match the argument names requirements for the alternate arduino-esp32fs-plugin?

the mkfatfs tool from lab-plus already matches those requirements, however as highlighted by @atanisoft it needs static linking, and I have no idea how to produce that.

```
  mkfatfs  {-c <pack_dir>|-u <dest_dir>|-l|-i} [-d <0-5>] [-b <number>]
            [-p <number>] [-s <number>] [--] [--version] [-t <fstype>] [-h]
            <image_file>

  mklittlefs  {-c <pack_dir>|-u <dest_dir>|-l} [-d <0-5>] [-a] [-b <number>] 
           [-p <number>] [-s <number>] [--] [--version] [-h] 
           <image_file>

  mkspiffs  {-c <pack_dir>|-u <dest_dir>|-l} [-d <0-5>] [-a] [-b <number>] 
           [-p <number>] [-s <number>] [--] [--version] [-h] 
           <image_file>
```
what would be the shortest path, editing the esp-idf tool or produce statically linked binaries?

atanisoft, 2023-05-04:
If it is in python it shouldn't be a problem for Linux / Mac as it can be run as-is natively, only on Windows does an executable need to be provided. If an executable is being provided for mk*fs then it must be statically linked to avoid a number of differences in Linux distributions.

lbernstone, 2023-05-04:
I'm the one that originally compiled the tool for @lorol, and I made it clear that I wasn't going to support it or provide exectuables for other platforms.  I think using what is provided by IDF is in almost all cases the right solution.  If the name needs to be changed, I'm sure lorol will be willing to make a quick PR.

igrr, 2023-05-05:
I am not sure we can completely avoid the knowledge about the underlying tool in arduino-esp32fs-plugin, since each filesystem has some settings specific to it only. For example, when generating SPIFFS we need to specify "page size" and "block size". For FAT, there are no such settings. Instead, we have sector size and allocation unit size (a.k.a. cluster size). I think it is much worse to shoehorn these FS-specific options into option with common names, instead of properly handling this in the FS plugin. It's not difficult to create 3 derived classes for different filesystems with each having knowledge about the underlying FS, and instantiate the right class at run time.

closed 2025-01-16 by Parsaabasi