espressif/arduino-esp32 #8163: Rainmaker confirming Wi-Fi connection

Whiskeytango16, 2023-05-05:
### Board

ESP-WROOM-32S

### Device Description

GPIO BOARD

### Hardware Configuration

See sketch

### Version

v2.0.4

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I uploaded a compiled Rainmaker sketch to my ESP32. I was able to get to the point to scan the QR code in a browser. Using the iOS v2.11.0 I was able to pair to the device, achieve the blue led light on the ESP and complete step one. As for step two, it seems to be stuck on this step. For troubleshooting, I erased the flash via the cmd from windows and received that it completed the task. After reloading the sketch and going through the steps, the rainmaker app is still stuck at step 2. Any help would be appreciated to get this moving forward.

[image]

### Sketch

```cpp
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"
#include <DHT.h>
#include <SimpleTimer.h>
const char *service_name = "my Wi-Fi name";
const char *pop = "my password";
// define the Chip Id
uint32_t espChipId = 5;
// define the Node Name
char nodeName[] = "Hydroponic Rainmaker";
// define the Device Names
char deviceName_1[] = "Switch1";
char deviceName_2[] = "Switch2";
//char deviceName_3[] = "Switch3";
//char deviceName_4[] = "Switch4";
// define the GPIO connected with Relays and switches
static uint8_t Growlights = 19; //D23
static uint8_t Pump = 18; //D22
static uint8_t RelayPin3 = 21; //D21
static uint8_t RelayPin4 = 22; //D19
static uint8_t SwitchPin1 = 13; //D13
static uint8_t SwitchPin2 = 12; //D12
static uint8_t SwitchPin3 = 14; //D14
static uint8_t SwitchPin4 = 27; //D27
[... 261 lines cut ...]
 //Serial.println("WiFi Connected");
 digitalWrite(wifiLed, true);
 if (Timer.isReady()) {
 //Serial.println("Sending Sensor Data");
 sendSensor();
 Timer.reset(); // Reset a second timer
 }
 }
 manual_control();
}
```

### Debug Message

```plain
Rainmaker app is stuck on step two
```

### Other Steps to Reproduce

I tried online tutorials on how to flash the Esp32 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

a-nassif, 2023-09-15:
same issue here, if anyone had any luck figuring it out.

VojtechBartoska, 2023-10-23:
Hi, can you please retest this with version 2.0.14 @Whiskeytango16 & @a-nassif? Thanks

VojtechBartoska, 2023-12-11:
Closing as expired.

closed 2023-12-11 by VojtechBartoska