espressif/arduino-esp32 #8169: EEPROM.commit() blocks interrupts for 30ms

kachurovskiy, 2023-05-07:
### Board

ESP32-S3-WROOM-1U-N16R2

### Device Description

See https://github.com/kachurovskiy/nanoels/tree/main/h4 but I don't think it's relevant.

### Hardware Configuration

My board schematic is available on https://github.com/kachurovskiy/nanoels/tree/main/h4#main-board-schematic but I don't think it's relevant.

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

Espressif Systems 

### PSRAM enabled

no

### Upload speed

115200

### Description

With `commit()` call it looks like this:

[image]

Without `commit()` it's all working well:

[image]

This problem also affects hardware PCNT pulse counter and likely other interrupt logic.

### Sketch

```cpp
#define ENC_A 15
#define OUT 11

#include <EEPROM.h>

void IRAM_ATTR spinEnc() {
  digitalWrite(OUT, digitalRead(ENC_A));
}

void setup() {
  EEPROM.begin(256);
  pinMode(ENC_A, INPUT);
  pinMode(OUT, OUTPUT);
  attachInterrupt(digitalPinToInterrupt(ENC_A), spinEnc, CHANGE);
}

void loop() {
  EEPROM.write(0, digitalRead(ENC_A));
  // EEPROM.commit(); // uncommend this line to get 30ms periods of ignored interrupts
  delay(10);
}
```

### Debug Message

```plain
Not relevant.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-05-07:
Consider the following:
1) `EEPROM.write()` copies a single byte into an array to be written.
2) `EEPROM.commit()` writes buffered data to flash.
3) Writing to flash takes time.
4) Processor is executing code from flash.
5) Processor cannot access flash during a flash write (depends on processor architecture).

What do you expect to happen?

For a further analysis you would have to dig into the NVS libraries in the IDF.

Workarounds:
1) Schedule your commit for a time when interrupts can be blocked.
2) Save to an external EEPROM chip.
3) Save to an external battery backed RAM chip.

Note: the same issue likely occurs when writing files to SPIFFS.

kachurovskiy, 2023-05-07:
I agree with your analysis - the goal of this bug is to ensure this behavior is well documented in EEPROM, Preferences and interrupts documentation - it might not be sufficiently emphasized at the moment.

lbernstone, 2023-05-07:
https://github.com/espressif/arduino-esp32/tree/master/libraries/EEPROM

closed 2025-01-16 by Parsaabasi