espressif/arduino-esp32 #8170: Autoreconnect and failing WiFi.begin

bertmelis, 2023-05-07:
### Board

esp32c3

### Device Description

Wemos C3 mini

### Hardware Configuration

base board powered by USB

### Version

v2.0.9

### IDE Name

Platformio

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

On autoreconnecting to WiFi, the code calls `WiFi.disconnect()` and afterwards `WiFi.begin(ssid, psk)`.

It may happen that the call to `begin` fails. Apparently, in this case, no callback is fired to signal the failed connection and no further attempt is made to reconnect again.

I do have low memory issues because of TLS connections that are closing/retrying after the WiFi loss. I suppose they are only cleaned up after a while by the framework.

I'm using the 2.x release.

### Sketch

```cpp
Simple state machine in repo here: https://github.com/bertmelis/espMqttManager/blob/main/src/espMqttManager.cpp
```

### Debug Message

```plain
[ 50101][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 45 bytes...
[ 50111][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 2 bytes...
[ 50117][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[ 50119][I][espMqttManager.cpp:290] onMqttClientDisconnected(): Disconnected from MQTT (CS-1)
[ 50123][I][espMqttManager.cpp:234] reconnectWaitCleanSession(): Reconnecting to MQTT (CS)
[ 50136][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 211904
[ 50138][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[ 50154][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[ 50157][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
[ 50160][V][ssl_client.cpp:178] start_ssl_client(): Loading CA cert
[ 50178][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[ 50180][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...
[ 50987][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-17040) RSA - The public key operation failed : BIGNUM - Memory allocation failed
[ 50990][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -17040
[ 50997][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[ 51005][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[ 51011][I][espMqttManager.cpp:294] onMqttClientDisconnected(): Disconnected from MQTT (CS-2)
[ 58165][I][espMqttManager.cpp:294] onMqttClientDisconnected(): Disconnected from MQTT (CS-2)
[ 60661][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: VHS100-SH, BSSID: a2:21:b7:9f:89:cf, Reason: 200
[ 60661][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 60668][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[ 60676][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
[ 60683][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
```

After this log, the esp stays silent.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-05-08:
Feature request: Make WiFi.begin() gracefully handle low memory situations.

bertmelis, 2023-05-08:
I must admit I solved my low memory issue so I don't suffer this issue anymore. But underlying, the issue remains of course.

WiFi.begin() gracefully handles low memory situations as it returns `WL_CONNECT_FAILED`. This issue is that afterwards, there is no event anymore to hook into for a next attempt to reconnect.