espressif/arduino-esp32 #8171: Cannot communicate with ESP32S2

NiceLucius, 2023-05-07:
### Board

ESP32-S2

### Device Description

Adafruit QT PY ESP32S2

### Hardware Configuration

Seperate antenna via u.FL connector, USB connected to PC

### Version

latest master (checkout manually)

### IDE Name

-

### Operating System

Windows 11 and 10

### Flash frequency

-

### PSRAM enabled

yes

### Upload speed

-

### Description

I cant upload any new firmware.  The command "esptool -p com3 chip_id" produces the error below. The board disappears from the windows port list and reappears under com4! 

esptool.py v4.5.1
Serial port com3
Connecting...
Failed to get PID of a device on com3, using standard reset sequence.

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Een apparaat dat op het systeem is aangesloten, werkt niet.', None, 31)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.

### Sketch

```cpp
n.a.
```

### Debug Message

```plain
see description
```

### Other Steps to Reproduce

I tried it on a windows 10 and windows 11 machine and got the same result.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-05-07:
The very first time you connect to a esp-xx using usb-cdc, you must hold down the boot/prg button/pin (gpio0 on the s2).  After programming, you will need a hard reset, and then it should behave as expected.

mrengineer7777, 2023-05-08:
@NiceLucius Does lberstone's answer resolve your issue?

NiceLucius, 2023-05-08:
Almost! Pressing the boot button during power up allows to upload new (esphome) firmware. So far so good and it actually performs the task I want it to do. The hard reset I perform is a disconnect and connect to the computer. 

However, after reconnecting it to the computer the problem reappears. I tried some more things and there are now two options to upload firmware but they are both not "pretty" as the problem stays:
- press boot button before and during startup
- ask chip_ip via esptool, The esp32 errors and gets a new com port. On this new com port I can upload new firmware or ask for a chip_id that gives a valid response. Funny enough it then resets to a com port that doesn't work.

So what can I do that after an upload the com port "sticks" and does not give an error? Another hard reset? The board has a  boot button and reset button.

lbernstone, 2023-05-08:
Have you enabled usb-cdc in the tools menu?

NiceLucius, 2023-05-09:
This last part is tricky for me. I enabled the USB-CDC on boot but the problem isn't solved. See below screenshot with also on the left the command window where the **esptool chip_id** command gives altering results after each try. I assume that the error after the sketch upload saying that it can not exit the download mode is not a serious issue.

[image]

mrengineer7777, 2023-05-09:
I don't have an S2 so I can only offer guesses.
The missing chip ID is concerning. I had issues with esptool at 921600 on an older computer. Try a lower baud rate.

VojtechBartoska, 2023-08-18:
Closing this ticket as answered, if needed, you can reopen.

closed 2023-08-18 by VojtechBartoska