espressif/arduino-esp32 #8172: SoftRST Stuck problem

GodayMax, 2023-05-08:
### Board

ESP32 Dev Module

### Device Description

ESP32-WROOM-32

### Hardware Configuration

GPIO2 PWM
GPIO33 ADC

RX empty
TX empty

### Version

v1.0.6

### IDE Name

PlatformIO

### Operating System

MacOS 12

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

with RX empty (or maybe somewhere it set to low)

after first power on, using some way to soft reset the module【using ESP.restart()】，it will not start properly，even can not execute this line in setup

###
Serial.println("=============start============")

### Sketch

```cpp
#include <esp32fota.h>
#include <Arduino.h>

esp32FOTA esp32FOTA(PRODUCT_ID, FIRMWARE_VERSION, false, true);

void setup() {
  Serial.begin(115200);
  Serial.println("=============start============")

}

void loop() {}
```

### Debug Message

```plain
[16:45:06.843]收←◆[I][esp32fota.cpp:63] e
[16:45:06.876]收←◆sp32FOTA(): Current firmware version: 1.1.0"
------

#########
No show of "=============start============"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

GodayMax, 2023-05-08:
it only shows on some modules

if comment out this line 「  Serial.begin(115200);」 it will not happen again;

mrengineer7777, 2023-05-08:
Check the state of your boot strapping pins. A common issue is holding GPIO00 low, which will put your module into bootloader mode. https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html

FYI v1.0.6 has been out of support for a while.  Consider using v2.0.6 or above.

closed 2023-05-09 by mrengineer7777