espressif/arduino-esp32 #8176: esp_bt_controller_mem_release being called in initArduino() when CONFIG_BT_ENABLED=y

sidwarkd, 2023-05-09:
### Board

ESP32 Dev Module

### Device Description

DevKitC plain

### Hardware Configuration

No additional hardware connected.

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

2000000

### Description

The use of `esp_bt_controller_mem_release` does not seem correct in esp32-hal-misc.c. In the function `initArduino()` if you have set BT_ENABLED in sdkconfig, it will call `btInUse` to determine whether to release the BT memory. There are two problems.

First, in using the sample sketch below (a slightly modified version of the provisioning sample included with this library) it does not appear to link correctly. The sample will fail to run properly as the BT memory is released in `initArduino` so trying to enable BT later for provisioning fails. The weak linking is not working as expected.

Secondly, this pattern seems wrong as well. The code in `initArduino()` would imply that you could have CONFIG_BT_ENABLED but have `btInUse()` return false which, per the code, would never be possible. If CONFIG_BT_ENABLED is yes then esp32-hal-bt.c declares `btInUse()` to return true always. So there is no scenario where if the config is enabled `btInUse()` _should_ return false (although it does per the first problem mentioned above).

This makes the following block of code in esp32-hal-misc.c puzzling:
```
#ifdef CONFIG_BT_ENABLED
    if(!btInUse()){
        esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
    }
#endif
```

To get the provisioning example code to work with BLE the call to `esp_bt_controller_mem_release` must NOT be called in `initArduino`.

### Sketch

```cpp
#include "WiFiProv.h"
#include "WiFi.h"
#include "nvs_flash.h"
#include "nvs.h"

bool is_provisioned = false;
void SysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id)
    {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        Serial.print("\nConnected IP address : ");
        Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
        is_provisioned = true;
        break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
        Serial.println("\nDisconnected. Connecting to the AP again... ");
        break;
    case ARDUINO_EVENT_PROV_START:
        Serial.println("\nProvisioning started\nGive Credentials of your access point using \" Android app \"");
        break;
    case ARDUINO_EVENT_PROV_CRED_RECV:
    {
        Serial.println("\nReceived Wi-Fi credentials");
        Serial.print("\tSSID : ");
        Serial.println((const char *)sys_event->event_info.prov_cred_recv.ssid);
        Serial.print("\tPassword : ");
        Serial.println((char const *)sys_event->event_info.prov_cred_recv.password);
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL:
    {
        Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
        if (sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR)
            Serial.println("\nWi-Fi AP password incorrect");
        else
            Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");
        break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
        Serial.println("\nProvisioning Successful");
        break;
    case ARDUINO_EVENT_PROV_END:
        Serial.println("\nProvisioning Ends");
        break;
    default:
        break;
    }
}

void setup()
{
    Serial.begin(115200);
    WiFi.onEvent(SysProvEvent);
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, "abcd1234", "Prov_123");
}

void loop()
{
    if(is_provisioned)
    {
        Serial.println("Connected to Wi-Fi and ready to run main application");
        delay(5000);
    }
    else
    {
        Serial.println("Waiting for Wi-Fi credentials. Open app to get started.");
        delay(5000);
    }
}
```
```

### Debug Message

```plain
E (912) wifi_prov_scheme_ble: bt_mem_release of classic BT failed 259
I (920) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1022) wifi:mode : sta (ac:0b:fb:6c:f4:8c)
I (1023) wifi:enable tsf
I (1024) wifi:Set ps type: 1

E (1027) simple_ble: simple_ble_start enable controller failed 259
E (1028) protocomm_ble: simple_ble_start failed w/ error code 0x103
E (1035) wifi_prov_scheme_ble: Failed to start protocomm BLE service
E (1041) wifi_prov_mgr: Failed to start service
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

MingyaoLiu, 2023-05-17:
Same issue on ESP32-C3 and 2.0.9.

MingyaoLiu, 2023-05-17:
@sidwarkd do you have a temporary fix for this? I tried disable the memory release, however when I try to connect to it through rainmaker app, it always fails and no error in console.

VojtechBartoska, 2023-05-17:
can you please help with triage @PilnyTomas? Thanks

VojtechBartoska, 2023-10-02:
Closing this issue as expired, if needed you can reopen.

sidwarkd, 2023-10-27:
@PilnyTomas I am not able to use the master branch as it requires v5.1 of the IDF. When I check out the 2.0.11 tag I know longer get the error but I also do not get any WiFi networks returned in the provisioning app.  If I roll back to my fixed branch I see them appear again. Then I tried different tags:

2.0.11 - No error but no WiFi networks returned in the app
2.0.12 - Back to getting the error again
2.0.13 - Same error
2.0.14 - Same error

So at least on the 2.x release it seems like this was kind of fixed and then regressed again. Is there some other commit I should try?

sidwarkd, 2023-10-27:
Researching this more and definitely a regression. The following shows 2.0.9 on the left, 2.0.10 in the middle and then 2.0.12 on the right which essentially puts it back to the way it was. So definitely broken again in 2.x. Will this be fixed in 2.x?

[image]

SuGlider, 2023-10-31:
@sidwarkd / @me-no-dev - I have made it work using the sketch presented in this issue.
Important details:
- Upload the Sketch using the `Erase All Flash Before Sketch Upload: "Enabled"` in order to erase previous WiFi/BLE sertup.
- In the `ESP BLE Prov APP` don't forget to change the Prefix to the one the sketch is using with Lower Case `Prov_`

These are the step that I used to make it work:
1- Downloaded the ESP BLE Prov APP in my Android phone.
2- Uploaded the sketch to the ESP32 - **USING the option `Erase All Flash Before Sketch Upload: "Enabled"`**
3- Turn the Android phone Bluetooth ON and open the ESP BLE Prov APP.
4- Click on the "Provision New Device" button
5- Ignore the QR Scan and click on "I don't have a QR code" button. **Change the Prefix to lower case "Prov_"**
6- Android will scan BLE device. It shall list "Prov_123" in the Devices List. If necesary click on "Scan Again"
7- Click the device "Prov_123"
8- Confirm the proof of possession PIN for Prov_123 as "abcd1234" and click the "Next ->" button
9- Click on "Join Other Network" and enter your SSID/Password for your local WiFi network. Attention to Lower/Upper case.
10- Press the "PROVISION" button and wait for the process to complete.

### This is the output:
```
ts Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1416
load:0x40078000,len:14804
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
Waiting for Wi-Fi credentials. Open app to get started.
Provisioning started
Give Credentials of your access point using " Android app "

Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.

Received Wi-Fi credentials
	SSID : MyWiFiSSID
	Password : MyWiFiSSID_PWD
Waiting for Wi-Fi credentials. Open app to get started.

Connected IP address : 192.168.3.128

Provisioning Successful

Provisioning Ends
Connected to Wi-Fi and ready to run main application
Connected to Wi-Fi and ready to run main application
Connected to Wi-Fi and ready to run main application
```

SuGlider, 2023-10-31:
I have used Arduino Core 3.0.0-alpha2 ...

SuGlider, 2023-10-31:
@sidwarkd 
>2.0.11 - No error but no WiFi networks returned in the app

I also get no list of WiFi networks in the APP, but I can set it manually with "Join Other Network" button.

sidwarkd, 2023-10-31:
@SuGlider  On older versions the scan actually returns a list of WiFi networks but the later versions do not. I was able to "Join Other Network" but the flow is supposed to return a list of available WiFi networks. That is a bug, right?

SuGlider, 2023-11-01:
The quesiton would be if this is an APP issue or ESP32 Software issue. Both could supply the list of WiFi SSIDs, given that both shall be close to each other for a BLE connection.

SuGlider, 2023-11-01:
@sidwarkd - I have just run all over again here with Arduino Core 3.0.0-Alpha2 and it is running perfectly fine, including listing the WiFi SSIDs in my Android APP.

Sometime it doesn't list it immediately, but it seems to be running fine with no bugs.

SuGlider, 2023-11-01:
@sidwarkd - Please try it again from scratch. It shall run fine.

sidwarkd, 2023-11-01:
@SuGlider I cannot use the 3.0.0-alpha2 tag as it requires IDF 5.1 or newer. This bug specifically relates to the state of things in 4.x. Is there a way to get the library to work probably in 4.x? I'm am on, what I believe to be, the most current at 4.4.6. I get the following error when I try to build on the 3.0.0-alpha2 tag.

```
.....CMake Error at /home/dev/esp/idf/tools/cmake/build.cmake:463 (message):
  HINT: Please check manifest file of the following component(s): arduino

  ERROR: Because project depends on idf (>=5.1) which doesn't match any
  versions, version solving failed.
```

Is there a commit of this library I can be on where the above sample compiles and works properly on 4.x?

SuGlider, 2023-11-01:
What exactly is the Arduino Core version that your project needs?

Is it using the Arduino IDE or some other building environment?

sidwarkd, 2023-11-01:
@SuGlider I am happy to use any version of Arduino Core that works with the 4.x version of IDF. I am using arduino-esp32 as a component directly in IDF (not Arduino IDE). Currently I am using a fork of the repo with fixes for a [community project](https://github.com/deploythefleet/arduino_ble_provisioning). However, instead of pointing to a fork I would like to point to this official repo and I'm happy to do so at any commit as long as it works with IDF 4.x.

me-no-dev, 2023-11-02:
@sidwarkd then use the `release/v2.x` branch of this repo :) it's for the 4.4 branch of ESP-IDF

sidwarkd, 2023-11-02:
@me-no-dev That has been the history of this issue. Please see my previous comment.
```
2.0.11 - No error but no WiFi networks returned in the app
2.0.12 - Back to getting the error again
2.0.13 - Same error
2.0.14 - Same error

So at least on the 2.x release it seems like this was kind of fixed and then regressed again. Is there some other commit I should try?
```

sidwarkd, 2023-11-03:
@VojtechBartoska I couldn't see how to re-open the issue. Can you do that as it's not resolved. I also re-named the issue to reflect that it applies to the 2.x branch.

SuGlider, 2023-11-05:
@sidwarkd  - I've found out something interesting...

It works fine and lists all WiFi SSIDs only, and just only, if the sketch is built using `Core Debug Level: "Debug"` or higher.
This is valid for Arduino Core 2.0.8 .. 2.0.14.

Whenever Debug level is enabled, it will work fine and display this message:
```
[ 18577][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
```

Maybe this could be the workaround that helps your project.... 2.0.x won't receive new fixes.

sidwarkd, 2023-11-06:
@SuGlider Unfortunately this didn't work for me. I do get all of the debug output but still no WiFi networks and still only in v2.0.11. The other ones are still broken.

```D (30255) wifi_prov_mgr: Scan finished
D (30259) wifi_prov_mgr: Scan results :
D (30263) wifi_prov_mgr:        S.N. SSID                             BSSID        RSSI AUTH
D (30442) proto_wifi_scan: Response packet size : 4
D (30443) protocomm_ble: Response from  write:
D (30444) protocomm_ble: a6 e4 0a 22 
D (30628) protocomm_ble: Inside read w/ session - 0 on param 42 0
```

However, since you mentioned the 2.0.x branch will not receive any new fixes we can close this as "Won't Fix" and I will port my community example to IDF 5.1. I appreciate your time in helping troubleshoot.

sidwarkd, 2023-11-06:
@SuGlider @me-no-dev In my case I'm fine to move to the 5.1 version of IDF. However, from [a recent blog post](https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4) announcing the 3.0.0 release it states:
 > "The expected stable release of the latest version is December 2023 and the 2.0.x will be under support maintenance until July 2024 then will be discontinued."

If that is true why wouldn't the 2.0.x branch receive any new fixes?

SuGlider, 2023-11-07:
The branch may receive one final set of fixes by the end of Q1/24.
It is possible that a final 2.0.15 version is released by that time.

SuGlider, 2023-11-07:
I'll keep this issue in the list of issues to be verified for 2.0.x.

Phetdanai, 2024-01-06:
> ### Board
> ESP32 Dev Module
> 
> ### Device Description
> DevKitC plain
> 
> ### Hardware Configuration
> No additional hardware connected.
> 
> ### Version
> latest master (checkout manually)
> 
> ### IDE Name
> VSCode
> 
> ### Operating System
> Linux
> 
> ### Flash frequency
> 40MHz
> 
> ### PSRAM enabled
> no
> 
> ### Upload speed
> 2000000
> 
> ### Description
> The use of `esp_bt_controller_mem_release` does not seem correct in esp32-hal-misc.c. In the function `initArduino()` if you have set BT_ENABLED in sdkconfig, it will call `btInUse` to determine whether to release the BT memory. There are two problems.
> 
> First, in using the sample sketch below (a slightly modified version of the provisioning sample included with this library) it does not appear to link correctly. The sample will fail to run properly as the BT memory is released in `initArduino` so trying to enable BT later for provisioning fails. The weak linking is not working as expected.
> 
> Secondly, this pattern seems wrong as well. The code in `initArduino()` would imply that you could have CONFIG_BT_ENABLED but have `btInUse()` return false which, per the code, would never be possible. If CONFIG_BT_ENABLED is yes then esp32-hal-bt.c declares `btInUse()` to return true always. So there is no scenario where if the config is enabled `btInUse()` _should_ return false (although it does per the first problem mentioned above).
> 
> This makes the following block of code in esp32-hal-misc.c puzzling:
> 
> ```c++
> #ifdef CONFIG_BT_ENABLED
>     if(!btInUse()){
>         esp_bt_controller_mem_release(ESP_BT_MODE_BTDM);
>     }
> #endif
> ```
> 
> To get the provisioning example code to work with BLE the call to `esp_bt_controller_mem_release` must NOT be called in `initArduino`.
> 
> ### Sketch
> ```c++
> #include "WiFiProv.h"
> #include "WiFi.h"
> #include "nvs_flash.h"
> #include "nvs.h"
> 
> bool is_provisioned = false;
> void SysProvEvent(arduino_event_t *sys_event)
> {
>     switch (sys_event->event_id)
>     {
>     case ARDUINO_EVENT_WIFI_STA_GOT_IP:
>         Serial.print("\nConnected IP address : ");
>         Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
>         is_provisioned = true;
>         break;
>     case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
>         Serial.println("\nDisconnected. Connecting to the AP again... ");
>         break;
>     case ARDUINO_EVENT_PROV_START:
>         Serial.println("\nProvisioning started\nGive Credentials of your access point using \" Android app \"");
>         break;
>     case ARDUINO_EVENT_PROV_CRED_RECV:
>     {
>         Serial.println("\nReceived Wi-Fi credentials");
>         Serial.print("\tSSID : ");
>         Serial.println((const char *)sys_event->event_info.prov_cred_recv.ssid);
>         Serial.print("\tPassword : ");
>         Serial.println((char const *)sys_event->event_info.prov_cred_recv.password);
>         break;
>     }
>     case ARDUINO_EVENT_PROV_CRED_FAIL:
>     {
>         Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
>         if (sys_event->event_info.prov_fail_reason == WIFI_PROV_STA_AUTH_ERROR)
>             Serial.println("\nWi-Fi AP password incorrect");
>         else
>             Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");
>         break;
>     }
>     case ARDUINO_EVENT_PROV_CRED_SUCCESS:
>         Serial.println("\nProvisioning Successful");
>         break;
>     case ARDUINO_EVENT_PROV_END:
>         Serial.println("\nProvisioning Ends");
>         break;
>     default:
>         break;
>     }
> }
> 
> void setup()
> {
>     Serial.begin(115200);
>     WiFi.onEvent(SysProvEvent);
>     WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, "abcd1234", "Prov_123");
> }
> 
> void loop()
> {
>     if(is_provisioned)
>     {
>         Serial.println("Connected to Wi-Fi and ready to run main application");
>         delay(5000);
>     }
>     else
>     {
>         Serial.println("Waiting for Wi-Fi credentials. Open app to get started.");
>         delay(5000);
>     }
> }
> ```
> 
> ```
> 
> 
> ### Debug Message
> 
> ```
> E (912) wifi_prov_scheme_ble: bt_mem_release of classic BT failed 259
> I (920) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
> I (1022) wifi:mode : sta (ac:0b:fb:6c:f4:8c)
> I (1023) wifi:enable tsf
> I (1024) wifi:Set ps type: 1
> 
> E (1027) simple_ble: simple_ble_start enable controller failed 259
> E (1028) protocomm_ble: simple_ble_start failed w/ error code 0x103
> E (1035) wifi_prov_scheme_ble: Failed to start protocomm BLE service
> E (1041) wifi_prov_mgr: Failed to start service
> ```
> 
> ### Other Steps to Reproduce
> _No response_
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-02-20:
@lucasssvaz Can you please test this? thanks

dzungpv, 2024-02-25:
I have the same problem with 2.0.15 as IDF components. Here is the logs:
```
I (765) boot: Disabling RNG early entropy source...
W (888) esp_claim: Generating the private key. This may take time.
E (2795) wifi_prov_scheme_ble: bt_mem_release of classic BT failed 259
W (2798) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
E (2880) simple_ble: simple_ble_start enable controller failed 259
E (2881) protocomm_ble: simple_ble_start failed w/ error code 0x103
E (2883) wifi_prov_scheme_ble: Failed to start protocomm BLE service
E (2890) wifi_prov_mgr: Failed to start service
[  2062][E][WiFiProv.cpp:151] beginProvision(): wifi_prov_mgr_start_provisioning failed!
```
The sketch here: https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/examples/RMakerCustomAirCooler/RMakerCustomAirCooler.ino

lucasssvaz, 2024-02-26:
Just tested as @SuGlider on the latest 2.x branch using ESP32-S3 and everything works as expected:
The WiFi list also appears as expected.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   266][I][WiFiProv.cpp:139] beginProvision(): Starting AP using BLE. service_name : Prov_123, pop : abcd1234
Waiting for Wi-Fi credentials. Open app to get started.

Provisioning started
Give Credentials of your access point using " Android app "
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.
Waiting for Wi-Fi credentials. Open app to get started.

Received Wi-Fi credentials
	SSID : xxxxxxxxxxx
	Password : xxxxxxxxxxxxx

Connected IP address : 192.168.15.59

Provisioning Successful
Connected to Wi-Fi and ready to run main application
```

VojtechBartoska, 2024-02-26:
@shahpiyushv Can you please take a look on this? Any ideas what should be wrong here? thanks

olup, 2024-08-16:
@dzungpv I am faced with the same error, did you find a way to resolve it ?

dzungpv, 2024-08-17:
> @dzungpv I am faced with the same error, did you find a way to resolve it ?

There is a bug with BLE, I make it work by un comment  the line: https://github.com/espressif/arduino-esp32/blob/54c4b0c619458c2a1a830d8518bdc0fdaee435bf/cores/esp32/esp32-hal-misc.c#L268

VojtechBartoska, 2024-10-24:
@azmi-plavaga can you please retest on version 3.1.0-RC1?

lucasssvaz, 2024-11-14:
@VojtechBartoska The sketches from this issue and the examples both with ESP32 and ESP32-C3. Everything worked as expected.

slhddn, 2025-01-12:
I am having the exact same problem. It was working fine on Arduino IDE. I migrated the project to PlatformIO. After migration, WiFi BLE Provisioning stopped working. Please see the log below:

E (10920) simple_ble: simple_ble_start enable controller failed 259
E (10921) protocomm_ble: simple_ble_start failed w/ error code 0x103
E (10922) wifi_prov_scheme_ble: Failed to start protocomm BLE service
E (10928) wifi_prov_mgr: Failed to start service
[  5284][E][WiFiProv.cpp:151] beginProvision(): wifi_prov_mgr_start_provisioning failed!

Thanks

lucasssvaz, 2025-01-15:
@Jason2866 Have you ever seen this issue ?

Jason2866, 2025-01-15:
Without an example sketch and the used platformio.ini can't say anything.
Can't imagine there is an issue, since Arduino code is same as when using ArduinoIDE.

rftafas, 2025-02-18:
Closing this. Seems to be a migration issue or something else, using old release. 

If tested against current/updated software and problem persists, open new issue.

kumardeo, 2025-03-21:
Facing same issue (runtime error) when using `PlatformIO IDE (platform: pioarduino)` and `ESP32 (nodemcu-32s)` board.
The same code works as expected when using `Arduino IDE`.

`src/main.cpp`
From https://github.com/espressif/arduino-esp32/tree/master/libraries/RainMaker/examples/RMakerSwitch

`platformio.ini`
```ini
[env:nodemcu-32s]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = nodemcu-32s
framework = arduino
board_build.partitions = rainmaker_4MB_no_ota.csv
```

soile1991, 2025-03-22:
Hey @kumardeo ,

I had the same problem when I was working with ESP32.

I added this to my platformio.ini and it worked.

```

build_flags = 
	-D SOC_WIFI_SUPPORTED=1
	-D BLUEDROID_ENABLED=1
	-D CONFIG_BT_ENABLED=1
	-D CONFIG_BT_BLE_ENABLED=1

```

Hope it helps!

kumardeo, 2025-03-22:
> I had the same problem when I was working with ESP32.
> 
> I added this to my platformio.ini and it worked.

Didn't work for me :(

Jason2866, 2025-03-23:
Issue can be reproduced with this example https://github.com/pioarduino/test-rainmaker/tree/wifiprov

Fix as suggested here is working https://github.com/espressif/arduino-esp32/issues/8176#issuecomment-1551592102 to solve the issue

Admsher, 2025-03-29:
Can you check if this works, an issue similar solved a couple of days ago https://github.com/espressif/arduino-esp32/issues/11181

Jason2866, 2025-04-07:
Fix for Platformio/pioarduino is to add `lib_archive = no` to platformio.ini

closed 2025-04-07 by Jason2866